tube0013/tube_gateways #1: Support for Tasmota32

s-hadinger, 2021-03-11:
Hi, great devices.

I'm interested in adding this device to the Tasmota compatibility list. It should be straightforward, and would make a great coordinator device working in standalone.

I have ordered an EFR32 based device, but would be interested in the CC too.

https://tasmota.github.io/docs/Zigbee/

tube0013, 2021-03-11:
Thanks @s-hadinger

Hedda, 2021-04-27:
@tube0013 Congrats on Tube's Zigbee Gateways cc2652p2 variant now being officially supported by Tasmota for ESP32!

s-hadinger wrote "**_The Tube's cc2652p2 is now officially supported. The EFR32 will be shortly._**" in https://github.com/arendst/Tasmota/issues/9081

Hedda, 2021-04-27:
@s-hadinger Is there a "zbbridge" specific version of Tasmota32 that should be used on ESP32 based Zigbee bridges/gateways?

closed 2021-06-23 by tube0013

Hedda, 2021-06-23:
FYI, posted a question about Zigbee2Tasmota (Z2T) compatibility/support for the CC2652P varient here -> https://github.com/arendst/Tasmota/discussions/12457