tube0013/tube_gateways #2: [REQUEST] Texas Instruments CC1352P based gateway/bridge or USB-adapter/stick/dongle

Hedda, 2021-03-19:
Any chance would consider making a Texas Instruments CC1352P based gateway/bridge or USB-stick/dongle hardware adapter?

The reason is that CC1352 supports all Zigbee SE frequencies.

CC1352 is more expensive but enable Sub-1 GHz frequencies used by Zigbee Smart Energy (ZSE) profile which isn't part of Zigbee 3.0

https://www.ebyte.com/en/product-view-news.aspx?id=766

https://www.szrfstar.com/product/235-en.html

https://www.szrfstar.com/product/188-en.html

Zigbee Smart Energy is the world standard used by electricity, water, and gas meters that all utility companies install in your house.

As far as I know niether zigpy or zigbee-herdsman (zigbee2mqtt and iobroker) support the Zigbee SE profile as of yet.

Having CC1352 hardware available would allow developers to add support to zigpy and zigbee-herdsman (zigbee2mqtt/iobroker).

https://zigbeealliance.org/solution/smart-energy/

https://zigbeealliance.org/wp-content/uploads/2019/11/docs-07-5356-19-0zse-zigbee-smart-energy-profile-specification.pdf

https://training.ti.com/overview-zigbee-smart-energy-internet-things

https://www.ti.com/lit/slaa467

https://www.ti.com/lit/pdf/tidu213

https://www.ti.com/lit/pdf/swra467

https://www.nxp.com/pages/jn516x-zigbee-smart-energy:ZIGBEE-SMART-ENERGY

https://www.nxp.com/docs/en/supporting-information/MAXSECZBNETART.pdf

"Zigbee Smart Energy (Zigbee SE) is the world's leading standard for interoperable wireless products that monitor, control and automate the delivery and use of energy (and other resources, such as water). A wireless network is implemented at the consumer's premises using the ZigBee PRO protocol with the Smart Energy application profile. This Home Area Network (HAN) contains devices such as a Metering Device, In-Premise Display (IPD) and Load Control Device, and is connected to the energy provider using an Energy Service Interface (ESI) via a backhaul network."

PS: ZOE2 shield/hat by Electrorama will use CC1352P radio from Texas Instruments, using the E79-400DM2005S module by EByte:

https://electrolama.com/projects/zoe2/

https://github.com/electrolama/zoe2

tube0013, 2021-03-19:
Not sure how Low Freq Band Zigbee fits into the Home Automation ecosystem right now, I see it more in use in industry and that is not where I am targeting devices.

Hedda, 2021-03-19:
Should let you get metrics from modern/smart electricity meter, water meter, and gas meter if they are using unencrypted Zigbee SE.

It provides data about energy/water/gas usage, etc. for your utilities, which is something many home automation users want.

That is, Zigbee SE is not normally used in devices that consumer buy themselves but in devices utility companies install in your house.

Example:

https://www.rainforestautomation.com/rfa-z114-eagle-200-2/

Zigbee SE is similar but not the same as H1/P1 protocol which is accessed via wired RJ45 HAN (Home Area Network) port on meters.

https://blog.sintef.com/sintefenergy/han-port-smart-meters-ams/

https://www.kode24.no/guider/smart-meter-part-1-getting-the-meter-data/71287300

Hedda, 2021-03-19:
Oh, and just to clarify; CC1352 radio has multi-band support which it support both Sub-1 GHz and 2.4GHz frequencies at the same time

closed 2021-06-23 by tube0013