tube0013/tube_gateways #3: Fix IP address for CC2652P2 Serial Coordinator

LindsayReid, 2021-03-24:
I cannot see a way to fix the IP address of the coordinator. It has picked up an address from my DHCP server which has a block of addresses available but also has a reserved block. I wish to "fix" the coordinator to an address in the reserved block.

Under normal circumstances, Tasmota defaults to DHCP but by using the Tasmota console one can change the device to a fixed IP address, Gateway and Net Mask. How can I do the same with the CC2652P2 based Coordinator?

Many thanks

tube0013, 2021-03-24:
Are you able to assign a fixed address to the MAC address in your router?

If not a custom firmware for the sep will be needed. I can send instructions in a bit or can do this for you if you let me know the ip settings you want.

LindsayReid, 2021-03-24:
I have a number of reserved addresses, one of which I would like to use. I would like to leave the router as it is and am quite happy to modify the firmware and re-flash if necessary. I suspect that others may also appreciate the info/tools.

Many thanks for the fast response.

tube0013, 2021-03-24:
No problem I will put up some info today - the esphome folder has the current yaml config it requires the 2 additional files from the folder server_stream.h/.cpp beside it.

I'll dig into the full changes to the config shortly but it shouldn't be difficult.

Do you have esphome currently set up?

LindsayReid, 2021-03-24:
I have it in Home Assistant as I used it along with Tasmota before I moved primarily to Zigbee. I have compiled/flashed firmware for it in the past but not for some time! Many thanks again.

tube0013, 2021-03-24:
I just pushed up some docs for the esphome configuration.

It definitely helps to have some experience with it. let me know if you have any questions.

https://github.com/tube0013/tube_gateways/tree/main/esphome

LindsayReid, 2021-03-24:
Many thanks - worked a treat. Firmware now reflashed with static address in my "reserved" block of addresses.

Your speedy assistance much appreciated.

closed 2021-03-30 by tube0013

DerWe, 2023-10-28:
Hi,

I just bought one of your Zigbee2mqtt POE module as well.
As I´m currently setting Up everything to switch to this module, I wanted to set an Static IP using the Config as well. So I came across this topic. But unfortunately, the link to the Documentation is not working anymore.

I discovered the .yaml File, edited the IP and tryed to upload it with ESPHome. But it is not workink like expected.
When I do so, the log on the Device tells me the upload was successful, ESPHome thinks the IP has changed, bit it is still connected using its old IP

Here is my yaml config:
`
#Olimex ESP32-PoE Ethernet
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12
  
    # Optional manual IP
  manual_ip:
    static_ip: 192.168.20.150
    gateway: 192.168.20.254
    subnet: 255.255.255.0
 `

What files do I need to put in the Root directory?

Best Regards!

tube0013, 2023-10-28:
in this file:
https://github.com/tube0013/tube_gateways/blob/main/models/current/tubeszb-cc2652-poe-2023/firmware/esphome/tubeszb-cc2652-poe-2023.yaml

you wil want to remove the comments from Lines 35-38 only, and make sure the manual_ip: line is indented 2 spaces and the lines below that indented another 2 spaces. it you just delete the # on each line it should maintain that.

no need to pull down any files.

I'm also happy to compile a firmware binary for you I assume the IP info you want is above?

thanks,

DerWe, 2023-10-28:
I already did it like you mentioned. My mistake was the following:

1.  I tried to Upload it directly from ESPHome. This failed, because it already wanted to use the new Address, before it was changed by the Device
2. Then I tried to generate the Firmware with ESPhome. Then manually upload directly in the Webinterface. This was successful by using the Legacy Format. The other one failed with "magic bytes do not match" in the first place

Thanks a lot! - I appreciate your work. You are quick :)