tube0013/tube_gateways #4: Provide reboot option on web server

vertex-github, 2021-03-30:
When setting up the device and moving it onto the correct IP / subnet / VLAN, it would be helpful to have a reboot option in the onboard web page.

tube0013, 2021-03-30:
my first go at adding this isn't working. I will keep trying.

MattWestb, 2021-04-12:
If i remember right is:
```
switch:
  - platform: restart
    name: "Living Room Restart"
```
working and you is getting one restart switch in HA (that you can disabling) and in the web UI.

MattWestb, 2021-04-12:
In my ESP32Cam i have:
```
switch:
  - platform: restart
    name: $hostname Restart
```
and in WEB UI is looks like:
[image]

tube0013, 2021-04-12:
Yup, I've testing it.

MattWestb, 2021-04-12:
Recommending the user disabling the switch in HA is also god so not clicking on "all on" in one room and thee Zigbee coordinator is rebooting ;-))

tube0013, 2021-04-12:
The api option is needed for logging, and from what I can tell the internal flag will hide the switch from HA but also the web front end as well. Which is no good. 

I think I would actually recommend ignoring the esphome device completely in HA.

MattWestb, 2021-04-12:
If not like having some information in HA is then flagging the discovered ESPHome as ignore and its not being integrated but i think it can being problem with the watchdog in ESPHome if not having MQTT or HA APT enabled and working its rebooting (if its not being fixed now).

The "reboot sensor" is easy disabling / hiding as one entry in HA and its not problem for the user and you still having in the WEB UI.

tube0013, 2021-04-12:
I've disabled the api reboot in the config. https://github.com/tube0013/tube_gateways/blob/0271f04cfd7bef7b0245c54882c136b09107ee27/esphome/tube_zb_gw_cc2652p2.yaml#L23

I've had mutli-month uptimes with it as well.

I've had no issues with the SiLabs MGM modules. series 1 or series 2.

tube0013, 2021-04-28:
Implemented and included in most recent .bin files.

closed 2021-04-28 by tube0013