tube0013/tube_gateways #5: Missing install step in docs

vertex-github, 2021-03-30:
Attempting to flash z2mqtt coordinator firmware, when following the docs, Im running into:

```
./cc2538-bsl.py -p socket://192.168.66.8:6638 -evw CC1352P2_CC2652P_launchpad_coordinator_20210120.hex
Opening port socket://192.168.66.8:6638, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20210120.hex
Your firmware looks like an Intel Hex file
ERROR: Firmware is Intel Hex, but the IntelHex library could not be imported.
Install IntelHex in site-packages or program your device with a raw binary (.bin) file.
Please see the readme for more details.
```

The readme in the programmer specifies that I need the following three dependencies:
```
pip install pyserial
pip install intelhex
pip install python-magic
```

After installing the above three items, the flash still fails with the same message about intelhex not being available.

What step(s) is/are missing here?

tube0013, 2021-03-30:
hmm, the device is pre-flashed so should be no need to flash it.

maybe try pip3 to install the required re-reqs, as cc2537-bsl runs with python3

closed 2021-04-09 by tube0013

MiNuSMaiD, 2024-03-25:
[image]