tube0013/tube_gateways #6: Question - E180-ZG120B programing

araczkowski, 2021-04-03:
Hi @tube0013 

I'm trying to connect to Ebyte E180 via J-Link by Segger to do SWD flashing.

in the ebyte manual I found this:
[image]

unfortunately I'm not sure how to localize the PF0 and PF1 on the E180-ZG120B-TB board
ebyte|j-link
--------|------
VCC |vref 
?|SWDCLK
?| SWDIO
GND|GND

[image]

**Could you please give me a hint - how you did this?**
Thanks :+1: 

BTW
Are you from Poland like me? If yes I would like to contact with you to talk about business proposition.
[image]

tube0013, 2021-04-03:
[image]

It's the 2 pins I tried to mark in red.

My father's paternal grandparents (I think) - I am told they came to America from Eastern Poland. But I do not know much of the family history on that branch.

araczkowski, 2021-04-03:
Thanks :+1: 

I already figured out this :)

now I'm testing this, and I described step by step... a note for myself to be able to do this again... 
https://github.com/sviete/AIS-EZSP

[image]

We are working on project AI-Speaker Open Source based on Home Assistant and Zigbee2Mqtt. Your approach with zigbee gate looks very impressive and promising for me. Especially that we are on Android and by default we don't have serial in kernel. + we have some experience with Tasmota and ESP staff.

I thought that maybe you are from Poland and I would like you to join us and sell your devices to our users. But then I figured out that you are from America. Unfortunately the delivery costs, taxes and CE certificate will be to big cost for us and for our users. To give you full picture -> currently we are selling CC2531 devices for about 9$ (with support and 2 year warranty) and some our potential customers still prefer shopping in Aliexpress and waiting 1-2 months ;) So I'm not proposing you to join us and sell your devices, because I now that this will not work, sadly in Poland the market here is still very sensitive to price. 

But I will ask you directly - do you mind if I use your approach to create similar gate for our project?
I'm not sure if this will be our direction, I'm still checking all the possibilities for the more powerful Zigbee for our project.. I already tried with  Wifi zigbee bridge https://ai-speaker.discourse.group/t/zigbee-w-asystencie-domowym-inne-mozliwosci/1545 (sorry our forum is in Polish only) all works fine but wifi if no the best idea, I prefer Ethernet. 
I can join your project if possible, your solution seems to be very promising... for the future we can thing about BLE too.
 
Please let me know if my ask is not hurting you feelings, it is not my intention :peace_symbol: 
Of course you can take whatever you want from our repos https://github.com/sviete if you are interested to have HA on Android (automation platform + multimedia) I can personally guide you step by step to achieve this :)
 and if you let me join I hope I will be able to give some code back to your repo

Greetings from Poland :) 
Your project is awesome

closed 2021-04-03 by araczkowski