tube0013/tube_gateways #7: Cc2652 coordinator issues

jscherry, 2021-04-11:
Sorry if this is in the wrong area. I have a tubes cc2652 coordinator and the router. Router has been flawless. Coordinator just drops out of hom assistant using zha and I have to power cycle the coordinator to see it again. Also before having these issues I had devices connected as children through the coordinator and now I can’t get a device to connect to it? Any help is greatly appreciated

tube0013, 2021-04-11:
Can you send me a message at store  at tubeszb.com 

If it isn't working reliably I will swap it out.

jscherry, 2021-04-11:
I can’t find a email link on the store sight. That’s why I posted on GitHub.

tube0013, 2021-04-11:
It's store@tubeszb.com

jscherry, 2021-04-11:
Just emailed you

closed 2021-06-23 by tube0013