tube0013/tube_gateways #8: Support for PoE

urbozz, 2021-04-19:
Are there any plans to have a PoE option, such as using the [Olimex ESP32-POE-ISO ](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware) or the [LILYGO TTGO T-Internet-POE](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1307)?

tube0013, 2021-04-19:
No current plans, I have a LILYGO POE board, but the size of it doesn't lend itself toward a reasonable overall size in my opinion. The Olimex is to expensive for me to consider -when for 1/10 of the price you can get an external POE > Micro usb/Ethernet adapter on AliExpress. I think to make it feasible it would need to be a custom board design, and while I have looked into it I haven't gotten very far nor have the time to commit to doing it right.

Hedda, 2021-04-27:
> I think to make it feasible it would need to be a custom board design, and while I have looked into it I haven't gotten very far nor have the time to commit to doing it right.

Be sure to keep an eye on the components used by the zoe2 project from Electrolama if you ever get time to look at a custom board:

https://electrolama.com/projects/zoe2/

https://shop.electrolama.com/collections/raspberry-pi-addons/products/zoe2-cc1352p-radio-for-raspberry-pi?variant=37646452883617

https://shop.electrolama.com/collections/raspberry-pi-addons/products/zoe2-cc1352p-radio-rtc-poe-for-raspberry-pi?variant=37646454161569

https://github.com/electrolama/zoe2

It is open source hardware and while designs not posted yet it builds on top of the previous zoe design which does have plans:

https://github.com/electrolama/zoe

https://electrolama.com/projects/zoe/

Hedda, 2021-04-27:
I think that zoe2 just use the newer Silvertel Ag9900 Series module?

https://silvertel.com/9900-2/

https://silvertel.com/poe-power-over-ethernet-html/

tube0013, 2021-04-27:
To do a custom PoE Ethernet board would be a lot of work and I'd need to change everything about how I am doing this. I don't have the time to commit to that. The board design is very very important when dealing with the Ethernet and the paths are critical to get right, or else you will have fantom Ethernet issues. I'd likely need to move to a pick and place SMT service to build the boards. Current supply issues and the small orders I would make would put me at the end of the lines and cost proportionately more- just not be worth the cost for me.

Hedda, 2021-04-27:
That is understandable. Just wanted to point out Electrolama plans to sell both PoE and non-PoE version of the same zoe2 board:

zoe2-lite = CC1352P Radio for Raspberry Pi (Radio only) -> https://shop.electrolama.com/collections/raspberry-pi-addons/products/zoe2-cc1352p-radio-for-raspberry-pi?variant=37646452883617

[image]

zoe2 = CC1352P Radio + RTC + PoE for Raspberry Pi -> https://shop.electrolama.com/collections/raspberry-pi-addons/products/zoe2-cc1352p-radio-rtc-poe-for-raspberry-pi?variant=37646454161569

[image]

MattWestb, 2021-04-27:
Both version is **not** recommended using the integrated PCB antenna then is being blocked of PCB trace in  the no copper zone of the antenna !!
And  the external connection is **only** for sub band and not for 2.4 GHz that is normally used for Zigbee.

MattWestb, 2021-04-27:
5 dBm module for 2.4 GHz and not round only in one narrow lobe :-(((

Hedda, 2021-04-27:
> Both version is **not** recommended using the integrated PCB antenna then is being blocked of PCB trace in the no copper zone of the antenna !!  And the external connection is **only** for sub band and not for 2.4 GHz that is normally used for Zigbee.

yeah yeah, but the discussion here is specifically ONLY about the choice of PoE module that Electrolama is using on the zoe2 board.

tube0013, 2021-04-27:
And it's still apples to oranges as it is using the Ethernet on the pi. and using the zoe module to pull the power from it somehow.

Drizzt321, 2021-04-28:
> when for 1/10 of the price you can get an external POE > Micro usb/Ethernet adapter on AliExpress. 

I had no clue these were already a think. I should have figured.

Hedda, 2021-04-28:
> for 1/10 of the price you can get an external POE > Micro usb/Ethernet adapter on AliExpress.

Doubt you can get a cheap PoE to USB converter that follows the standard IEEE 802.3af specifications. The ones I have seen are passive and just pass along any current to set to them and as such can destroy your device if you do not input the right current.

tube0013, 2021-04-28:
I've relented a bit. I'll post details in a bit once I get parts and get the thing built and configured. But I've ordered a small batch of new pcbs and some esp32-poe-iso boards to try and test the viability.

Drizzt321, 2021-04-28:
> 
> 
> > for 1/10 of the price you can get an external POE > Micro usb/Ethernet adapter on AliExpress.
> 
> Doubt you can get a cheap PoE to USB converter that follows the standard IEEE 802.3af specifications. The ones I have seen are passive and just pass along any current to set to them and as such can destroy your device if you do not input the right current.
> 
> Better if so to buy Ubiquiti Instant 802.3af Adapter (Models: INS‑3AF‑I‑G, INS‑3AF‑O‑G, INS‑3AF‑USB, INS‑8023AF‑I, INS‑8023AF‑O)
> 
> https://www.ui.com/accessories/instant-8023af-adapters/
> 
> https://eu.store.ui.com/search?mockup=Ubiquiti+Instant+&q=Ubiquiti+Instant

I found https://smile.amazon.com/UCTRONICS-PoE-Splitter-Gigabit-Raspberry/dp/B07CNKX14C that does gigabit, 802.11af, 5v@2.4a mUSB output. Seems to have some good reviews by various people in general for it working fine.

Hedda, 2021-04-30:
> I found https://smile.amazon.com/UCTRONICS-PoE-Splitter-Gigabit-Raspberry/dp/B07CNKX14C that does gigabit, 802.11af, 5v @ 2.4a mUSB output. Seems to have some good reviews by various people in general for it working fine.

Ubiquiti Instant 802.3af to USB adapter (SKU: INS-3AF-USB) for indoor also supports Gigabit but cost about $19 without USB cable:

https://store.ui.com/collections/operator-accessories/products/instant-802-3af-to-usb-adaptor

https://eu.store.ui.com/collections/operator-accessories/products/instant-802-3af-to-usb-adaptor

Hedda, 2021-05-21:
Awesome work by tube!

https://www.tubeszb.com/product/cc2652_poe_coordinator/15

https://twitter.com/TubesZb/status/1394478007730020352

https://twitter.com/TubesZb/status/1394845581823074308

closed 2021-06-23 by tube0013

Hedda, 2021-08-17:
@tube0013 Any chance you would consider posting schematics, etc. for your PoE variant version? At least look to be missing here:

https://github.com/tube0013/tube_gateways