tube0013/tube_gateways #10: Add instructions for optional ZHA automatic discovery via Zeroconf

Hedda, 2021-04-26:
Add instructions for optional ZHA automatic discovery via Zeroconf as an alternative to manual configuration in initial config flow:

https://github.com/home-assistant/core/pull/48420

PS: Not supported in current Home Assistant 2021.4 release but should be supported in upcoming Home Assistant 2021.5 release.

Hedda, 2021-05-19:
FYI, thegroove made a Zeroconf custom component for ESPHome -> https://github.com/thegroove/esphome-zeroconf 

Also see related discussion in -> https://github.com/thegroove/esphome-zbbridge/issues/1

tube0013, 2021-05-19:
yup following. thanks.

FYI efr32/ezsp coordinator discovery config flow is broken right now, not really a priority as there is not a lot of supply of modules right now but on my radar. CC2632/ZNP works fine though.

tube0013, 2021-05-19:
exploring it.  
[image]

Hedda, 2021-05-20:
Copy from my post here https://github.com/thegroove/esphome-zeroconf/issues/4

Second part would be to extend the ZHA integration to make use of those parameter values and set them as defaults in config flow.

FYI, Home Assistant's zeroconf maintainer/dev bdraco referred to this example for using TXT record to pass values to integrations:

https://github.com/home-assistant/core/blob/d4ed65e0f53fcab991b13061b1101470f24287a6/homeassistant/components/homekit_controller/config_flow.py#L199

By the way, have also seen related requests to get more information about the ESPHome device and the service that it is is providing.

That is, it would also be nice if also pass along additional information about the service that might not be used by ZHA config flow.

Such additional information could be displayed on the ESPHom,e device and include data on the hardware device, like example:

* service_name=Tube_Zigbee_Gateway
* service_type=serial_server_bridge
* service_component=stream_server 
* gateway_manufacturer_website=www.tubeszb.com
* gateway_products_model=EFR32_Series2_Ethernet_Gateway
* gateway_connection=ethernet_and_not_wifi
* gateway_platform=Espressif_ESP32
* esp_board_type=ESP-WROVER-KIT
* esp_board_version=V4.1
* zigbee_chip_manufacturer=Silicon_Labs
* zigbee_chip_model=MGM210PA32JIA2

PS: There is also a general Home Assistant community forum thread about Zeroconf support by the ZHA integration here:

https://community.home-assistant.io/t/zha-automatic-discovery-of-zigbee-coordinator-bridges-gateways-ethernet-wifi-network-devices-that-support-zeroconf-or-ssdp/293300/

MattWestb, 2021-05-20:
I have not seen how making the basic general for adding coordinators in ZHA.

The first part is triggering ZAH with IP and port.
Next is the NCP used and com parameters.
And if like  some cosmic names for the added device.

I think for the general implantation that is being triggered and only need changing the com parameter / NNCP type.

For EZSP its also possible populating manufacture tokens and reading them if need information.
ZHA is doing it and putting the info in the log then starting.

Hedda, 2021-05-20:
> I have not seen how making the basic general for adding coordinators in ZHA.

What do you mean? 

The initial basic Zeroconf discovery of support in the ZHA integration was added by dmulcahey in -> https://github.com/home-assistant/core/pull/48420

But the config flow will need to be extended on the ZHA integration side to make use of the values that are provides via TXT records.

Again, Home Assistant's zeroconf maintainer/dev bdraco referred to this example for using TXT record values in a config flow:

https://github.com/home-assistant/core/blob/d4ed65e0f53fcab991b13061b1101470f24287a6/homeassistant/components/homekit_controller/config_flow.py#L199

PS: The discovery should preferably also be more generalized so that ZHA can discover other gateways than those from Tube.

tube0013, 2021-12-17:
components broke with newer ESPHome builds.. need someone who knows esphome/c to update them I tired.. didn't go so well.

Hedda, 2021-12-17:
FYI, I believe that @syssi got it working in his ESPHome config for ZB-GW03(?) -> https://github.com/syssi/esphome-zb-gw03

https://github.com/syssi/esphome-zb-gw03/commit/da14ed53941c76bb1b1d2889dfd2eff56e4ac14e

He uses external_components:
  - source: github://oxan/esphome-stream-server
  - source: github://syssi/esphome-zeroconf@main

tube0013, 2021-12-17:
Doesn't look like any changes to the original zeroconf component from thegroove. It compiles fine, dumps the config in the debug log as it should but never actually advertises the device on mdns in my testing last night.

syssi, 2021-12-17:
If the esphome node name begins with "tube" it gets picked up properly by the discovery. If the name doesn't match the ZHA config flow isn't triggered.

tube0013, 2021-12-17:
@syssi is the zeroconf component working for you current ESPHome release? as I stated I can compile it without issue but it fails to advertise anything over mdns. - this was tested on an olimex esp32-poe.

I'm aware how the ZHA component discovery works - and it piggy backs the esphome mdns. the goal here was to use the external_component zeroconf to have the device broadcast it's settings - then modify the zha side to use that and pull all the settings from the mdns so no user entry is required for setup. (plus the efr32 side of discovery is not currently working in ZHA)

syssi, 2021-12-17:
I must admit I never captured some traffic to get a better understanding what's happening in detail. This is what I did:

1. I flashed this configuration using ESPHome 2021.12.1

```
substitutions:
  name: tube-zigbee-gateway

esphome:
  name: ${name}
  platform: ESP32
  board: esp-wrover-kit
  on_boot:
    priority: 600
    then:
      - switch.turn_off: download_mode
      - switch.turn_on: zigbee_reset

external_components:
  - source: github://oxan/esphome-stream-server
  - source: github://syssi/esphome-zeroconf@main

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 1
  power_pin: GPIO16

<<: !include common/api.yaml
<<: !include common/ota.yaml

logger:
  level: DEBUG

uart:
  id: uart_zigbee
  rx_pin: GPIO4
  tx_pin: GPIO2
  baud_rate: 115200

stream_server:
  uart_id: uart_zigbee
  port: 6638

text_sensor:
  - !include common/text_sensors/esphome_version.yaml

binary_sensor:
  - !include common/binary_sensors/status.yaml

  - platform: gpio
    pin:
      number: GPIO34
      inverted: true
    name: "${name} button"

sensor:
  - !include common/sensors/uptime.yaml

i2c:
  - id: i2c_bus
    sda: GPIO33
    scl: GPIO32
    scan: true

status_led:
  pin:
    number: GPIO15
    inverted: true

switch:
  - platform: restart
    name: "${name} restart"

  - platform: template
    id: zigbee_reset
    name: "${name} Zigbee Reset"
    turn_on_action:
      - switch.turn_on: zigbee_rst
      - delay: 10ms
      - switch.turn_off: zigbee_rst

  # SM-011 V1.0 nRST
  - platform: gpio
    id: zigbee_rst
    pin: GPIO13
    name: "${name} Zigbee nRST"
    inverted: true
    internal: true
    restore_mode: ALWAYS_OFF

  # SM-011 V1.0 PA00
  - platform: gpio
    id: download_mode
    pin: GPIO12
    name: "${name} Zigbee Download Mode"
    inverted: true
    restore_mode: ALWAYS_OFF

zeroconf:
  - service: tube_zb_gw_efr32
    protocol: tcp
    port: 6638
    txt:
      version: 1.0
      radio_type: ezsp
      baud_rate: 115200
      data_flow_control: software
```

As you mentioned the custom components prints some logs:

```
[15:48:12][C][ota:082]: Over-The-Air Updates:
[15:48:12][C][ota:083]:   Address: tube-zigbee-gateway.local:3232
[15:48:12][C][ota:086]:   Using Password.
[15:48:12][C][api:134]: API Server:
[15:48:12][C][api:135]:   Address: tube-zigbee-gateway.local:6053
[15:48:12][C][api:139]:   Using noise encryption: NO
[15:48:12][C][streamserver:089]: Stream Server:
[15:48:12][C][streamserver:096]:   Address: 192.168.1.225:6638
[15:48:12][C][zeroconf:019]: _tube_zb_gw_efr32._tcp :
[15:48:12][C][zeroconf:020]:    port = [6638]
[15:48:12][C][zeroconf:021]:    txt = ["version=1.0" "radio_type=ezsp" "baud_rate=115200" "data_flow_control=software"]
[15:48:12][C][mdns:084]: mDNS:
[15:48:12][C][mdns:085]:   Hostname: tube-zigbee-gateway
```

2. I've started a vanilla HA core 2021.12.3 installation. In the left corner I see a notification teling me "New devices discovered". The is a "discovered" ZHA device.
3. I click "configure" at the discovered ZHA device. The endpoint (socket://IP:6338) matches the ESPHome configuration. "port speed" is populated properly. "data flow" isn't preselected. I choose "software" manually + press "submit".
4. A spinner appears and an error "Failed to connect" pops up. 

Is this the same issue as yours?

Hedda, 2021-12-17:
By the way, also suggest using a custom name like "tube_zigbee_ezsp_gateway" and submit PR for general "`*zigbee*znp*`" and "`*zigbee*ezsp*`", using  "`*`" (stars) as wildcard as prefixes and suffixes to catch more custom names, as then others esphome based zigbee gateway/bridge projects could use generic node name either like "esphome_zigbee_ezsp_bridge" or their own unqiue custom name like "syssi_zigbee_ezsp_bridge".

https://github.com/home-assistant/core/blob/dev/homeassistant/generated/zeroconf.py#L86-L94

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/manifest.json#L27-L31

tube0013, 2021-12-17:
My goal ...eventually.. is to move away from using the esphome mdns. Thegroove custom zeronconf can send all the settings needed and could go under a top level like zha or something and board name, radio,baud and flow settings could all be in the text string.

Unfortunately as I said I could not get to actually work. The actual component looks like it was just adapted from what was the older mdns component in esphome. That component has gone through some changes.

I tried last night to adapt the custom code to base it off the newer mdns but didn't get anywhere.

Hedda, 2021-12-20:
But you believe the issue to lie in ESPHome or the esphome-zeroconf component by thegroove and not in Home Assistant / ZHA?