tube0013/tube_gateways #12: Provide option to enable Zigbee Router mode and activate pairing from gateway web server

Hedda, 2021-04-27:
Might be cool for an option to enable "Zigbee Router" mode and have a button to activate pairing mode from the gateway web server so that Tube's Zigbee Gateways could alternatively be used as stand-alone Zigbee Routers instead of remote serial-to-ip Zigbee coordinator by third-party software applications.

I am not sure about Z-Stack ZNP but at least EmberZNet NCP should allow this with the same firmware as already use for EFR32.

tube0013, 2021-04-27:
I'd prefer not to do this, with out some more testing. I have flashed the router FW to the coordinator a few times by mistake, and have noticed the TX or RX pin of the ESP32 was held high constantly when this happens.

tube0013, 2021-04-27:
also the BSL button is what controls the router reset/pairing trigger and is already accessible in the web interface.

Hedda, 2021-04-27:
> I'd prefer not to do this, with out some more testing.

Cool, but might be an option that help you sell more gateway devices if the same hardware can easily serve multiple purposes 😃

tube0013, 2021-04-27:
With the cc2652 it can already be done just by flashing the router fw.

I'm still unclear on if the efr32 router solution can actually be fully triggered via serial commands so that remains an unknown.

Hedda, 2021-05-28:
FYI, @MattWestb created hardware configuration of a Zigbee router firmware made for a Silabs EmberZNet EFR32 module:

https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/issues/6

"Router firmware with pins for E1743 remote (paring / reset button = pairing button and also BLB if using my bootlooader)."

https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module/files/6559440/ZigbeeBillyRouter.zip

https://github.com/zha-ng/EZSP-Firmware/files/6560171/ZigbeeBillyRouter.zip

Hedda, 2021-12-07:
FYI, pre-configured Zigbee Router firmware for EFR32MG21 adapters has now been released by xsp1989 on GitHub. From the readme posted it sounds as if so far only tested with ITead Zigbee 3.0 USB Dongle and SM-011 based USB adapters:

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle/RouterForDongle

https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/RouterForDongle/README.md

https://github.com/xsp1989/zigbeeFirmware/issues/16

https://github.com/xsp1989/zigbeeFirmware/issues/2

closed 2021-12-17 by tube0013