tube0013/tube_gateways #16: USB-to-Serial Bridge chip is used for Tube’s CC2652P2 USB Coordinator?

Hedda, 2021-05-06:
Curious so wondering which USB-to-Serial Bridge chip exactly is used for Tube’s CC2652P2 USB Coordinator?

Also, wondering if the USB to Serial Bridge chip it is connected via RTS + CTS to CC2652P module for Hardware Flow Control?

I understand that CH340 series (example CH340C and CH340E) by WCH are popular because they are inexpensive, but I also understand that they have fewer features and are also known to have driver issues on Linux and Mac OS (though I read Linux is better now in newer Linux kernel versions).

I understand that FT231 chip by FTDI  or CP2102N by Silicon Labs / Silabs is recommended for more feature and better support:

https://ftdichip.com/products/ft231xs/
https://ftdichip.com/products/ft231xq/

https://www.silabs.com/interface/usb-bridges/classic/device.cp2102
https://www.silabs.com/interface/usb-bridges/classic/device.cp2104
https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers
https://www.silabs.com/documents/public/data-sheets/cp2102n-datasheet.pdf
https://www.silabs.com/documents/public/application-notes/an197.pdf
https://www.silabs.com/documents/public/application-notes/an978-cp210x-usb-to-uart-api-specification.pdf
https://www.silabs.com/documents/public/application-notes/AN571.pdf

FYI, I read that the upcoming ZZHP/ZZH2 adapter by Electrolama will use FT231 which has more mature device drivers and will allow them to write a customized string onto the EEPROM of the USB-to-Serial chip (such as for example "Electrolama ZZHP v1.0.0") should make it more plug-and-play friendly and easier for different home automation software to automatically discover and configure without end-user interactions. I understand that a bonus feature in FT231 of allowing users to auto-reset and put the device in bootloader mode via the DTR/RTS pins exposed will enable much easier firmware upgrades for the end-users as they will no longer need to press any buttons to enter bootloader mode, and now bootloader mode can be activated via software from the firmware flasher program software which could make the firmware upgrade procedure more user-friendly.

By the way, have you read this related discussion about hardware standardisation for pins assignments in the community:

https://github.com/Koenkk/Z-Stack-firmware/issues/210

The discussion resulted in this README.md file with a matrix for pins like BSL Trigger Pin, RF Switch Control Pins, LED(s) + Auto-BSL:

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/README.md

tube0013, 2021-05-06:
May change in v2 but right now it's a ch340E

Hedda, 2021-05-06:
Related PR -> https://github.com/Koenkk/Z-Stack-firmware/pull/287 (to update the info in mentioned README.md for Koenkk/Z-Stack-firmware).

tube0013, 2021-12-17:
v2 uses CH340B

closed 2021-12-17 by tube0013