tube0013/tube_gateways #17: Issue setting up a new EFR32 coordinator with HASS

jl-678, 2021-05-08:
I am running my new EFR32 coordinator via serial/USB and so have put the jumpers in the middle per the GitHub instructions. However, the instructions here seem to focus on Ethernet, and I am having issues with USB.

Here is what I am doing:

1. I go into integrations and add ZHA and Hass sees a device at /dev/ttyAMA0. 
2. I choose that and EZSP radio type. 
3. This brings me to the screen where it shows the path, asks for port speed and data flow. The default port speed is 57200 and I choose 115200 per GitHub. Once, I click “Submit”, I get “failed to connect.” 
4. I have tried playing with data control and port speed and always seem to get failed to connect. 
 
Any advice on what I should do?

Oh and I added those lines to configuration.yaml. TIA for any ideas!

tube0013, 2021-05-08:
Do you have access to a non HA-OS command line with bellows installed (pip3 install bellows)? if so could you try running:

```bellows -d /dev/ttyXXX -b 115200 info```

this is how I test each device before it goes out. just want to make sure it's working,

jl-678, 2021-05-08:
Python error:

`Traceback (most recent call last):
  File "/home/pi/python/hassTest/venv/bin/bellows", line 10, in <module>
    sys.exit(main())
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/bellows/cli/util.py", line 39, in inner
    loop.run_until_complete(f(*args, **kwargs))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/bellows/cli/ncp.py", line 69, in info
    s = await util.setup(ctx.obj["device"], ctx.obj["baudrate"])
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/bellows/cli/util.py", line 118, in setup
    await s.reset()
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/bellows/ezsp/__init__.py", line 94, in reset
    await self._gw.reset()
  File "/home/pi/python/hassTest/venv/lib/python3.7/site-packages/bellows/uart.py", line 223, in reset
    return await asyncio.wait_for(self._reset_future, timeout=RESET_TIMEOUT)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError`

tube0013, 2021-05-08:
Are you sure you are hitting the right device?

can you try /dev/ttyUSB0?  - I just plugged one in and that's where it is showing up

tube0013, 2021-05-08:
and also re-confirming jumpers are on the middle 2 pins in each header row.

jl-678, 2021-05-08:
HASS (running on 8GB Pi4) showed the device at /dev/ttyAMA0.  My other Pi which I am using to test (Pi3) shows the same.  Checking "ls /dev" shows nothing at TTYUSB*.

To be 100% clear, the blue jumpers are connecting the following pins: STX/zTX and SRX/zRX which is the center two pins in both cases.

tube0013, 2021-05-08:
lsusb - do you get something like 

```ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge```

Have you confirmed it works via serial over tcp/ethernet?

Couple options to make this right - can you send me your order number at store (at) tubeszb.com

jl-678, 2021-05-08:
No, I do not see it listed under lsusb just the normal root hub stuff. I have tested over Ethernet and can try that now.

This is my order number: 259961824

jl-678, 2021-05-08:
Update, it works over Ethernet.

tube0013, 2021-05-08:
interesting and frustrating! I just dropped you an email with some options.

jl-678, 2021-05-08:
Thx. I responded with a couple of emails. I think that we have a resolution.

closed 2021-12-17 by tube0013