tube0013/tube_gateways #22: Update EFR32 Series 2 modem firmware

ivanroberto, 2021-06-09:
Hey guys,

I want to upgrade firmware of my coordinator EFR32 Series 2 and do not have a MAC machine, so I can't follow the isntructions provide by this link:
https://www.youtube.com/watch?v=zKrISuWEzL4tube-

My machine is a UBUNTU, if some one already knew how to procede, your help will be very welcome.

Thanks

tube0013, 2021-06-09:
There is a script attached to the discussion linked below that was just recently made that can do it. Just make sure nothing like ha is accessing the module. And run the script it takes care putting the module in bootloader mode and all.

https://github.com/zigpy/zigpy/discussions/743#discussioncomment-771853

I will write up some docs around this but it's been tested from Linux and works. You need to pip3 install xmodem as a prerequisite

What fw are you upgrading to? I haven't posted any 6.9.2 fw binaries yet as I have been trying to fine tune performance of it and throughly test.

tube0013, 2021-06-09:
GitHub might not link to the actual post with the attachment. Let's see if this works better:

github.com/zigpy/zigpy/discussions/743#discussioncomment-771853

ivanroberto, 2021-06-09:
Thanks for your quick feedback!

I would like to try this: 
https://github.com/tube0013/tube_gateways/blob/main/tube_zb_gw_efr32/Firmware/ZigBee%20Module/Series_2/MGM210PA32JIA_ncp-uart-sw_6.9.1_2.gbl

But, I'm not sure after your comment, maybe I should wait a little longer for you to finish your tests.

My problem here is that the sonoff's port sensors still disconnect and remain disconnected even if I re-bind them.

tube0013, 2021-06-09:
that's the same fw that shipped on all the Series2 gateways, so it won't be any different. which sonoff sensors? I have a few but not sure what you mean by port sensor.  and by re-bind do you mean re-pair?

here are the release notes for 6.9.2 where they note the fixes etc and also known issues. 
https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.9.2.0.pdf

FYI - I may be rolling the max children setting back to 32 from 64 in the next fw release as it frees up some memory which I think will lead to better performance. I'm testing this now with my production network. with a few routers this should not make any difference to over all network size limits.

ivanroberto, 2021-06-09:
Ok, great! I'll be waiting. No rush.
Thanks a lot!

tube0013, 2021-06-10:
okay I have at least one of those I will test it as well. I never put them in use as I thought they reported pretty slowly/poorly. I've moved to some tuya sensors, that unlike most of their stuff is are actually spec compliant and need no quirks and have worked well.