tube0013/tube_gateways #24: [Device Configuration] Setting Static IP - EFR32 Pro

ardysusilo, 2021-06-15:
**Enhancement**
Provide guideline on how to set a static IP address on EFR32 Pro Coordinator.

**Reason**
To perform a restoration of my previous Sonoff ZBBridge configuration without repairing router and child devices.

tube0013, 2021-06-15:
to use a static ip you'd need to compile a custom build of the esphome fw for the esp32.

What I'd recommend is assigning/locking the address with your router.

You can then update your core.config_entities in the .storage folder in HA and update the IP address for the ZHA entry.

you will need to back up your ZBBridge and Restore it to the new coordinator with bellows to migrate without repairing devices. <https://github.com/zigpy/zigpy/wiki/Coordinator-Backup-and-Migration>

tube0013, 2021-06-15:
.storage/core.config_entities: make sure you back it up as any errors will cause HA to fail to start. You'd just need to change the path (and optionally the title) to match the address given to the new coordinator.

[image]

ardysusilo, 2021-06-15:
Thank you. Sorry I did not read that you already explain about setting a static IP [here](https://github.com/tube0013/tube_gateways/blob/main/esphome/README.md).

I will try to set a static IP first via ESP Home add-on. If it fails, I will use DHCP reservation to make sure the IP never change.

Looking forward for the coordinator to arrive :)

tube0013, 2021-06-15:
haha, I forgot I added that <https://github.com/tube0013/tube_gateways/tree/main/esphome>

I'll double check it's up-to-date too as things are traveling.

closed 2021-06-15 by ardysusilo