tube0013/tube_gateways #27: Add "ESP Web Tools" installing/flashing firmware upgrades (or recovery) and onboarding via web site

Hedda, 2021-06-23:
Home Assisistant announced ESP Web Tools for simplifying onboarding by enabling website flashing ESP firmware via webserial:

https://www.home-assistant.io/blog/2021/06/16/power-up-your-esp-projects/#esp-web-tools-installing-projects-on-your-microcontroller-via-the-browser

https://esphome.github.io/esp-web-tools/

https://esphome.github.io/esp-web-tools/#add-website

Home Assisistant also announced ESPHome support for "Improv" open standard for setting up Wi-Fi via Bluetooth using a smart phone.

https://www.home-assistant.io/blog/2021/06/16/power-up-your-esp-projects/#improv-wi-fi-open-standard-to-provision-wi-fi-credentials-via-bluetooth-low-energy

https://www.improv-wifi.com

Home Assisistant have posted nice videos as part of the announcement  which demos both "ESP Web Tools" and "Improv" features:

https://www.youtube.com/watch?v=du38Oir_xp8

https://www.youtube.com/watch?v=k88BS8zgWq0&t=1s&ab_channel=NabuCasa

https://www.youtube.com/watch?v=AdCsX7Ni6Jc

PS: They also announced ESPHome Dashboard has been updated with a simplified and streamlined wizard for new configurations:

https://www.home-assistant.io/blog/2021/06/16/power-up-your-esp-projects/#esphome-dashboard-simplified-and-streamlined

https://www.youtube.com/watch?v=luE1hjCf3HI

PPS: This was part of a larger article about making ESP firmware projects more user-friendly and easier to get started and work with: 

https://www.home-assistant.io/blog/2021/06/16/power-up-your-esp-projects/

tube0013, 2021-12-17:
no onboarded need with ethernet/dhcp setup. my devices don't use wifi.

Added Project and Adoption to newest esphome yamls.

closed 2021-12-17 by tube0013

Hedda, 2021-12-17:
@tube0013 FYI, Improv supports onboarding via BLE (Bluetooth Low Energy) now too -> https://www.improv-wifi.com/ble/

MattWestb, 2021-12-17:
https://esphome.io/components/esp32_improv.html

tube0013, 2021-12-17:
> @tube0013 FYI, Improv supports onboarding via Bluetooth now too -> https://www.improv-wifi.com/ble/ (BLE without WiFi)

```
The esp32_improv component in ESPHome implements the open Improv standard for configuring Wi-Fi on an ESP32 device by using Bluetooth Low Energy to receive the credentials.
```

the zigbee coordinators I sell are not sold to be used with wifi, nor do I advocate using them with wifi. therefore this is no reason to to configure wifi - or enable improv as it will start the wifi in the module. So again there is no reason for the wifi provisioning to be used in these devices.

Of the newer ESPHome features - I am adding/added support for the "Project" and "Adoption" so a user can pull in the config once the device is on their network. I will look at setting up a page so devices could be flashed through the browser but that is low priority as it's only needed for troubleshooting and not a requirement at all.