tube0013/tube_gateways #29: [Device Configuration] - Changing Zigbee Channel

ardysusilo, 2021-07-04:
**The Problem**
Unable to change coordinator radio channel to channel 25.

**Reason**
Improving the stability of zigbee radio connection. There are 7 UniFi AP AC LR located in my home, sometimes zigbee radio interference occur resulting in a loss of data (mostly from battery powered zigbee). 

Note: I have now changed the channel of my WIFI AP to only channel 1 & 6 only. 

**What I have tried**
1. Following this [guide](https://www.home-assistant.io/integrations/zha/#defining-zigbee-channel-to-use), I have added the config into my configuration.yaml. However, after viewing the logs, it is still stated as "radioChannel=15".

YAML snippet is as follows-
`zha:
  zigpy_config:
    network:
      channel: 25`

2. Using this code `sudo bellows -d socket://xxx.xxx.x.xx:6638 form -c 25`, resulting in an error of "Invalid value for "-D" /"database": file "/root/.config/bellows/app.db" does not exist". During the execution of this command, the EFR32 coordinator is completely disconnected from HA as I can execute `sudo bellows -d socket://xxx.xxx.x.xx:6638 info` successfully.

MattWestb, 2021-07-04:
You first need issuing one leave commands to the coordinator.
That is making you is loosing all your devices :-((

I think after the leave you can starting HA normal and ZHA is starting forming one new network from you the HA config parameters.

Forming one new network is also working OK with bellows i have doing it for 2 mouths ago on one tuya ZBGW.

And you have one typo in the config (or more likely one copy paste problems):
```
      channel: 25```
```

Before it was possible using ZHA custom service for changing channel without loosing the devices but its broken for the moment.

MattWestb, 2021-07-04:
You can pointing to you `zigbee.db` that is in your HA config directory and you is not getting the fault.

tube0013, 2021-07-04:
Another option is you can use the backup and migration procedure and modify the channel in the backup file and then restore it. 

https://github.com/zigpy/zigpy/wiki/Coordinator-Backup-and-Migration

If you set up the radios in zha before the desired channel is in the config that channel setting will not do anything.

tube0013, 2021-07-04:
`bellows -d socket://IP:6638 backup > /config/bellows-backup.txt`

Change the channel in bellows-backup.txt

And then `bellows -d socket://IP:6638 restore --force --backup-file /config/bellows-backup.txt`

Credit to puddy for pointing out this method

https://discord.com/channels/330944238910963714/551843655682490399/842477834592714752

ardysusilo, 2021-07-04:
Thank you @MattWestb and @tube0013. The method above this post works successfully from what I can see from `sudo bellows -d socket://IP:6638 info` where the radioChannel is 25. 

One more question, although irrelevant to the topic. Is it okay to increase the radioTxPower to, for example, 15. Currently it is set to 8. 

Many thanks :)

tube0013, 2021-07-04:
I don't know that. I think of you find the ezsp parameter you can try and match the way the other configs work. 

It may need to be set in the fw though. I'll take a look at this later when I have some time.

ardysusilo, 2021-07-06:
Many thanks, I will surely check this Github page for any coordinator updates.

closed 2021-07-06 by ardysusilo