tube0013/tube_gateways #30: EFR32 Pro Coordinator - CRC Error in frame

ardysusilo, 2021-07-06:
**The Problem**
Getting CRC error in frame from time to time
```
Logger: bellows.uart
Source: /usr/local/lib/python3.8/site-packages/bellows/uart.py:84
First occurred: 10:33:01 AM (2 occurrences)
Last logged: 10:57:57 AM

CRC error in frame b'36bea1a9a52ab7b24794f025dd5592499c4e27abedce648bffc66389fc7e3ca7ebcdde6f8fffc7dbd5d2698c4623a9ec763ba5ea758241984c13b1e070381c0e07bb26e5ca658a459a4d9e4f9ff7c3d9d46a35a251904824c5eb7e' (b'c5eb' != b'f715')
CRC error in frame b'45f9a1a9a52af6b19a94f421435792499c4e27abedce468be6c66589fe7e2aa7ebcddeff6f8fffc7dbd5d2698c4623a9ec763ba5ea758241984c2613b1e070381c0e07bbe5ca658a459a4d9e4f9ff7c3d9d46a35a251904824e5797e' (b'e579' != b'c0a6')
```

**Last known working devices**
Using Sonoff ZBBridge, I never encountered CRC error in frame (within 2+ months of usage).

**Currently Paired Devices**
3 pcs of Sonoff Zigbee (BASICZBR3)

**What I have tried**
1. Removing the devices (routers and end devices) manually
2. Removing ZHA integration
3. Removing Zigbee.db from /config
4. Readding ZHA integration and pairing the devices manually again

**Production Environment**
Hardware: HA Blue (Odroid N2+) connected via Ethernet
HA Core: core-2021.6.6
Installation type: HA OS
Supervisor: supervisor-2021.06.6
OS: Home Assistant OS 6.1
Zigbee Coordinator: EFR32 Pro by TubesZB connected via Ethernet

ardysusilo, 2021-07-06:
**Additional Test**
Removing all routers and end devices - leaving only the coordinator in the ZHA integration. However, CRC error in frame still persist in the logs.

```
Logger: bellows.uart
Source: /usr/local/lib/python3.8/site-packages/bellows/uart.py:84
First occurred: 2:02:26 PM (1 occurrences)
Last logged: 2:02:26 PM

CRC error in frame b'51ada1a9a52a35b25b944b258a5592499c4e27abedce668bfdc66389fc7e3fa7ebcdde6f8fffc7dbd5d2698c4623a9ec763ba5ea758241984c2613b1e070381c0e07bbe5caff658a459a4d9e4f9ff7c3d9d46a35a2519048246d3d7e' (b'6d3d' != b'7dc3')
```

tube0013, 2021-07-06:
Is anything working at all?

If the frame has wrong crc then it either was corrupted during transmission or it was sent incorrectly by the radio.

I am using the exact same setup and will check my logs in a bit.

ardysusilo, 2021-07-06:
@tube0013 Yes, everything is working. All devices are successfully paired and I can control them.
The error pattern is as follows-
```
2021-07-06 14:02:26 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2021-07-06 14:02:26 ERROR (bellows.thread_0) [bellows.uart] CRC error in frame b'51ada1a9a52a35b25b944b258a5592499c4e27abedce668bfdc66389fc7e3fa7ebcdde6f8fffc7dbd5d2698c4623a9ec763ba5ea758241984c2613b1e070381c0e07bbe5caff658a459a4d9e4f9ff7c3d9d46a35a2519048246d3d7e' (b'6d3d' != b'7dc3')
2021-07-06 14:02:26 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 241 (readCounters) received: b'20000200010020000000000000000000010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'
```

ardysusilo, 2021-07-06:
These are the error logs from only the coordinator in the ZHA integration-
```
Logger: bellows.uart
Source: /usr/local/lib/python3.8/site-packages/bellows/uart.py:84
First occurred: 2:02:26 PM (12 occurrences)
Last logged: 5:01:40 PM

CRC error in frame b'1202a1a9a52ac4b127944c256fff5692499c4e27abedce638bfdc66389fc7e32a7ebcdde6f8fffc7dbd5d2698c4623a9ec763ba5ea758241984c2613b1e070381c0e07bbe5ca658a459a4d9e4f9ff7c3d9d46a35a25190482436177e' (b'3617' != b'fd7f')
CRC error in frame b'120aa1a9a52a04b6de944c25ae5192499c4e27abedce638bfdc66389fc7e31ebcda7de6f8fffc7dbd5d2698c4623a9ec763ba5ea758241984c2613b1e070381c0e07bbe5ca658a459a4d9e4f9ff7c3d9d46a35a25190482425ec7e' (b'25ec' != b'0ca6')
CRC error in frame b'16eea1a9a52a9cb36b944e25dd5492499c4e27abedce658bfdc66389fc7e3aa7ebcdde6f8fffc7dbd5d2698c4623a9ec763ba5ea758241984c2613b1e070381c0e07bbe5ca658a459a4d9eff4f9ff7c3d9d46a35a25190482448277e' (b'4827' != b'f2b5')
CRC error in frame b'334ba1a9a52a62b6cd944125f25192499c4e27abedffce608bfdc66389fc7e30a7ebcdde6f8fffc7dbd5d2698c4623a9ec763ba5ea758241984c2613b1e070381c0e07bbe5ca658a459a4d9e4f9ff7c3d9d46a35a25190482476be7e' (b'76be' != b'fa07')
CRC error in frame b'5468a1a9a52a94b7e1944625c95092499c4e27abedce6f8bfdc66389fc7e2ca7ebcdde6f8fffc7dbd5d2698c46ff23a9ec763ba5ea758241984c2613b1e070381c0e07bbe5ca658a459a4d9e4f9ff7c3d9d46a35a2519048245c237e' (b'5c23' != b'9561')
```

tube0013, 2021-07-06:
Interesting. It's when counters are being read, those are just diagnostics stats from the radio. 

Are they isolated to those?

ardysusilo, 2021-07-06:
Yes, in the 12 error logs mentioned above, the CRC error in frame happen after `DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()`

Is it maybe correlated because I change the channel to 25?

tube0013, 2021-07-06:
I do t think so. I wouldn't stress to much about it as I said the counters are just stats. If devices are communicating fine. 

If you don't mind could you post a bigger chunk of log from startup of zha through the first error.

I think you can grep bellows/zigpy/zha to keep it isolated to those.

I'd like to look and get some more eyes on it.

ardysusilo, 2021-07-06:
Sure thing, here you go. From ZHA start until first error occured.

[ZHA Debug.log](https://github.com/tube0013/tube_gateways/files/6769888/ZHA.Debug.log)