tube0013/tube_gateways #31: no network connection

tmottabr, 2021-08-20:
when i plug the controller to an POE port in my switch i get a green light in the switch indicating it is powered.

the controller board i get a red and a yellow light but no connection.

connecting a serial adapter to the header i get this:

`[15:21:20]ets Jul 29 2019 12:21:46
[15:21:20]
[15:21:20]rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
[15:21:20]configsip: 0, SPIWP:0xee
[15:21:20]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:21:20]mode:DIO, clock div:2
[15:21:20]load:0x3fff0018,len:4
[15:21:20]load:0x3fff001c,len:1044
[15:21:20]load:0x40078000,len:8896
[15:21:20]load:0x40080400,len:5828
[15:21:20]entry 0x400806ac
[15:21:20][I][logger:166]: Log initialized
[15:21:20][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[15:21:20][I][app:029]: Running through setup()...
[15:21:20][C][uart_esp32:072]: Setting up UART...
[15:21:20][C][switch.gpio:011]: Setting up GPIO Switch 'zRST_gpio'...
[15:21:20][D][switch:025]: 'zRST_gpio' Turning OFF.
[15:21:20][D][switch:045]: 'zRST_gpio': Sending state ON
[15:21:20][D][switch:025]: 'zRST_gpio' Turning OFF.
[15:21:20][C][switch.gpio:011]: Setting up GPIO Switch 'zBSL'...
[15:21:20][D][switch:025]: 'zBSL' Turning OFF.
[15:21:20][D][switch:045]: 'zBSL': Sending state ON
[15:21:20][D][switch:025]: 'zBSL' Turning OFF.
[15:21:20][C][ethernet:025]: Setting up Ethernet...
[15:21:35][W][ethernet:041]: Connecting via ethernet failed! Re-connecting...
[15:21:36][D][esp-idf:000]: E (16109) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0xffff
[15:21:36]
[15:21:37][D][esp-idf:000]: E (17110) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0xffff
[15:21:37]
[15:21:37][D][esp-idf:000]: E (17112) emac: Initialise PHY device Timeout
[15:21:37]
[15:21:37][E][ethernet:125]: ETH enable error: -1
[15:21:37][E][component:092]: Component was marked as failed.`

tube0013, 2021-08-20:
I've seen this with the non PoE boards a few times and have had luck reflashing the fw bin to fix it. 

The bin file is the olimex one here: https://github.com/tube0013/tube_gateways/tree/main/esphome

You can reflash over the usb connector with the esphome flasher app. The troubleshooting guide has a guide for the non PoE board but same procedure without the serial converter.

https://github.com/tube0013/tube_gateways/blob/main/Troubleshooting.md

If that doesn't work email net at store (at) tubeszb.com and We'll figure it out.

tmottabr, 2021-08-20:
had already tried using the bin file from here: https://github.com/tube0013/tube_gateways/tree/main/tube_zb_gw_cc2652p2

but no luck.. will take a look on the other link.. thanks..

tmottabr, 2021-08-21:
that fixed.. thanks..

closed 2021-08-21 by tmottabr