tube0013/tube_gateways #32: CC2652p2 POE Coordinator ZHA connection issue

glamirand, 2021-09-04:
Hi,

I just purchased a CC2652P2 Based Zigbee to Power Over Ethernet Serial Coordinator V2 to replace a ConBeeII for my Profalux covers.

I deleted all zigbee devices via zha.remove and the ZHA integration itself.
I then re set the ZHA integration according to the instructions (ZNP, socket://ip:6638, 115200, software flow control) and joined my devices to this new gateway.

All went fine except I'm having connection issues between my HA Virtual Machine (hypervised by VMM on a Synology RS1221+) & the gateway, resulting il osme lines in the Debug Log but more embarassing : my covers are slightly moving up and down on each reconnection as I guess the gateway enter permit join after a connection?

**Gateway Log**
[image]

**HA Log**
[image]

The coordinator is powered by a Unifi 8 POE switch and is set on the same VLAN as HA (so firewall shouldn't be cocnerned).
I forced the switch port to 100mbits FDX as recommended but this didn't helped.

- Can you confirm that the coordinator should permit join after a connection (is it on the coordinator or ZHA side?), and why?
- Is it possible to disable it?
- Do you have any suggestion to improve the connection stability?

**Disconnections happens variably from every minute to every half hour.
This is quite annoying and can't stay as it.
Any help is very welcome! :)
Thanks!**

tube0013, 2021-09-04:
Hi. This is for sure not the desired behavior for the device.

Let's look at the connection issues before getting into ZHA specific issues.

Could you try to reflash the PoE-ESP32.

You can do this OTA, but I'd rather do it over the usb connection on the board, so it does an erase of the current memory completely first.

The instructions are in this doc for the non-PoE coordinator, but the PoE-esp32 has an onboard serial you only need a micro usb cable (with data).

https://github.com/tube0013/tube_gateways/blob/main/Troubleshooting.md

the FW bin is here: https://github.com/tube0013/tube_gateways/tree/main/tube_poe_coordinator

The the USB port is inside the case on the side of the board. The powered Ethernet **MUST** be removed before using or you will kill the board or your computer. 

If that helps great, if not we can try a build with a static IP - this ultimately helped another user with disconnection issues.

glamirand, 2021-09-04:
Hi again,

I've just reflashed the module with the provided binary over USB and a connection issue just happened again :(

Let's try a static IP build then :/

tube0013, 2021-09-04:
Bummer, but lets keep rolling...

instructions to compile with a static IP are here: https://github.com/tube0013/tube_gateways/tree/main/esphome

if you want to give me the ip/gateway/mask IPs I can build and send you as well. just let me know.

glamirand, 2021-09-04:
Just flashed with a static ip build, let's see in the near minutes/hours..

I've noticed the following in the logs : 
[image]
I've seen you recommend to set the switch port configuration to 100mbits/Full Duplex, is there something wrong related to the log entry?

tube0013, 2021-09-04:
I'm not aware of making any recommendations about full duplex or not. I usually do not make any changes on my switch when running the devices.

glamirand, 2021-09-04:
That was actually not you, my bad!
[image]
https://github.com/Hedda/tube_gateways/commit/e7bdd9cf2a8cfd2684759faf1cc3e6a226e0de70

Anyway, I tried both a static IP and to replace "_" by "-" in the host name to solve esphome compilatio warnings but I'm still having the same disconnection issue :/

MattWestb, 2021-09-04:
Sorry for jumping in :-(

If forcing speed / duplex in one end (of the "cable") the other must being sett to the same.
If leaving one side in auto config it can being problems or if one side is half and the next full duplex and the same for the speed 10 / 100.

So auto on both "ends" or manual and the same in both "ends".

tube0013, 2021-09-04:
are you able to set your switch back to full auto for the port and see if that helps now with a static IP?

I'm a bit perplexed as it seems the ESP is actually staying on line, but the serial connection is closing. I have not heard of anyone with this issue, so it's either the hardware, or something specific with your network config. If hardware we can work that out, but want to fully rule out the network side.

glamirand, 2021-09-04:
> So auto on both "ends" or manual and the same in both "ends".

Thanks for the point, I set the switch back to auto then but no noticeable improvement :/

> are you able to set your switch back to full auto for the port and see if that helps now with a static IP?

I tried, but no more improvement :/

> I'm a bit perplexed as it seems the ESP is actually staying on line, but the serial connection is closing

I do agree! Is there any firmware upgrade of the "real" ZigBee stuff to try?
Any way to troubleshot the zigbee network? Perhaps certain oversized data frames which may end up crashing the stream server part?

I also tried the following things with no success :/
- Changed the VM NIC type (only virtio & e1000 are working, rtl8139 doesn't give HA an IP adress)
- Tried on a physical Raspberry Pi 4 (was my "production setup as of yesterday with a conbee2) : same issue
- Tried on a fresh VM install : always the same
- Tried to lighten the ESP part to allow more cpu time for the serial stream server component by disabling the web server and HA api : no more improvement

> I'm a bit perplexed as it seems the ESP is actually staying on line, but the serial connection is closing

I do agree! Is there any firmware upgrade of the "real" ZigBee stuff to try?
Any way to troubleshot the zigbee network? Perhaps certain oversized data frames which may end up crashing the stream server part?

I also have a ZiGate I may use for sniffing the ZigBee network even if I may need guidance for this, if it can help...

tube0013, 2021-09-04:
the zigbee side really should not come into play with the serial connection. 

I'm going to reach out to you via email.

glamirand, 2021-09-05:
>connect it through another switch, even non-PoE

I actually already tried to connect it to another POE/POE+ switch with the same results

I considered EMC based interferences last evening and I tried several things : 
- Replace my ethernet patch cable by a cat6 shielded one so the gateway is now "grounded" : I had no issue for the night 👍 
- I then realized that I had a temporary cheap and not shielded USB extension (the one to plug your Logitech wireless keybord & mouse receiver) linking my UPS and my NAS so I also switched it for a shielded one, and went back to my unshielded ethernet patch cable.
I also reflashed the ESP, re-enabling HA api and webserver  : it's working fine for more than an hour now 👍 

I'll keep you updated after a day of successful function.
If not I'll try to power up the ESP with the internal USB and a non POE switch port before reporting.

Fingers crossed! :)

tube0013, 2021-09-05:
Awesome. Fingers crossed let us know.

tube0013, 2021-09-16:
Closing this as it sounds like it has been resolved

closed 2021-09-16 by tube0013

glamirand, 2021-09-23:
> 
> 
> Closing this as it sounds like it has been resolved

Hi,

Sorry for the delay.

I effectively had no disconnection issue since my last post and cable management which is really nice, thanks for helping! :)

This may be another topic but I still observe my shutters to slightly move up and down (like they do when I open the ZigBee network), once the connection is established with ZHA (basically after booting up homeassistant..)

I think it's more related th the ZHA/ZNP integration than to the router itself.
It's not really big deal but I've never had this behavioir when I was using my ConBee2.

Any thoughts?