tube0013/tube_gateways #33: Unable to update firmware

joshuaspence, 2021-09-13:
I am trying to update the CC2652 firmware (just out of curiousity) but the process seems to fail.

```
$ cc2538-bsl.py -p socket://zigbee_gateway.local:6638 -evw CC1352P2_CC2652P_launchpad_coordinator_20210708.hex
Opening port socket://zigbee_gateway.local:6638, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20210708.hex
Your firmware looks like an Intel Hex file
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:23:7B:9B:C8
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 360448 bytes starting at address 0x00000000
ERROR: No response from target on status request. (Did you disable the bootloader?)
```

joshuaspence, 2021-09-13:
I may have bricked my device oops...

```
Zigbee2MQTT:error 2021-09-13 17:01:53: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
```

joshuaspence, 2021-09-13:
Z2M was in a crash loop, creating and terminating a connection to the stream server every time. Stopping Z2M during the firmware update worked.

closed 2021-09-13 by joshuaspence

tube0013, 2021-09-13:
Yes will add to docs, you must stop z2m or zha before doing any fw updates, energy scanning, backups --- anything that would create a second serial connection to the device