tube0013/tube_gateways #34: Not able to connect to zigbee2mqtt with cc2652p2_poe

koburg, 2021-10-01:
the zigbee2mqtt keep running in circle 

`[19:02:06] INFO: Socat not enabled, marking service as down
[19:02:06] INFO: MQTT available, fetching server detail ...
[19:02:06] INFO: MQTT server settings not configured, trying to auto-discovering ...
[19:02:07] INFO: Configuring 'mqtt://core-mosquitto:1883' mqtt server
[19:02:07] INFO: Previous config file found, checking backup
[19:02:07] INFO: Creating backup config in '/config/zigbee2mqtt/.configuration.yaml.bk'
[19:02:07] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[19:02:07] INFO: Handing over control to Zigbee2mqtt Core ...
[19:03:16] INFO: Handing over control to Zigbee2mqtt Core ...`

I get the "502: Bad Gateway" on the UI

i have got the zigbee2mqtt running with a Conbee before this so the plugin is working. 

i have followed the instructions and use tcp://xxx.xxx.xxx.xxx:6638 ip address instead of mDNS

after a while the log is updated with this:

> [19:04:24] INFO: Handing over control to Zigbee2mqtt Core ...
Zigbee2MQTT:error 2021-10-01 19:05:33: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-10-01 19:05:33: Failed to start zigbee
Zigbee2MQTT:error 2021-10-01 19:05:33: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-10-01 19:05:33: Exiting...
Zigbee2MQTT:error 2021-10-01 19:05:33: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.js:66:27)
    at Controller.start (/app/lib/controller.js:110:27)
    at start (/app/index.js:95:5)
[19:05:34] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.21.1 start
> node index.js
[19:06:42] INFO: Handing over control to Zigbee2mqtt Core ...

tube0013, 2021-10-01:
try starting up with the antenna unscrewed off:  https://github.com/Koenkk/zigbee2mqtt/issues/2893

koburg, 2021-10-02:
No luck with that. 

also I can find it in ZHA but im not able to find any devices it just keep searching.
in the device for the POE log it says:

connect established 
connect disconnected 
for ever.

tube0013, 2021-10-02:
My understanding of this error is it's because of RF interference.

I'd recommend using the zigpy-znp tools and doing a energy scan to see the best channel to use.

You can then reset the nvram with the same tools and start fresh on the new channel.

https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md

The esphome logs, will show the serial connecting / disconnects as z2m / zha loops trying to start.

Can you share any errors you got with zha?

tube0013, 2021-10-02:
I also just completely ready through the issue I linked at Zigbee2mqtt. They also suggest reflashing the zigbee module as well. This is document in the readme.md of this repo. But if you need any help let me know.

Another option is to try wrapping the coordinator in a tin foil tent when starting the first time.

koburg, 2021-10-02:
[image]

after a scan I got above picture.
i was using default channel 11 in zigbee2mqtt. 

I run the nvram reset

set the channel to 22 and now it seems to work. 

i have joined the first device.

yes when I need reflashing I get back to you because im not quite sure how to do it have read the doc.
i have a Mac os. but since I don't seem to need reflashing for the moment I get back about this.

tube0013, 2021-10-02:
Okay, can you email at store @ tubesszb.com your order number so I can look it up?

🚨That energy scan does not look good and your unit likely has an issue on the zigbee module with the ipex antenna connector. (It's an ["easy" fix](https://www.youtube.com/watch?v=CQy530D5-XQ) if you have or know someone with access to a hot air solder rework station) but if not I will exchange it. A few units slipped out before I was aware of the issue. I worked with the zigbee module manufacturer to find the solution and have been more specifically testing for it before shipping to be safe. But unfortunately a few units from early August - Mid August seem to be affected.

koburg, 2021-10-02:
i will email you but I just go the unit yesterday and you send it like monday this week.

tube0013, 2021-10-02:
oh, wow, so maybe not as they were definitely tested. was that scan with the antenna off? (may explain why my thinking went straight to this antenna connector issue).

koburg, 2021-10-02:
maybe I forgot the antenna new scan with antenna 

[image]

so from this what is the best channel to choose ?
20 ?

tube0013, 2021-10-02:
Much better (whew!). yes I'd say 20 looks to be the best choice.

mmychu, 2021-10-10:
> maybe I forgot the antenna new scan with antenna
> 
> [image]
> 
> so from this what is the best channel to choose ? 20 ?

Hi @koburg Can you share your Energy Scan command?
I have the same issue with Tube's new arrived adapter

mmychu, 2021-10-10:
python -m zigpy_znp.tools.energy_scan /dev/cu.usbmodem14101

What should I use insatead "/dev/cu.usbmodem14101" ?

mmychu, 2021-10-10:
I'm trying to debug why I can't start using Tube's CC2652p2 but with no success

[image]

Any suggestions appreciated

anubisg1, 2021-12-18:
Hello,

i'm having the same issue, how did you solve this at the end?

```
zigbee2mqtt          | Zigbee2MQTT:info  2021-12-18 08:38:26: Logging to console and directory: '/app/data/log/2021-12-18.08-38-24' filename: log.txt
zigbee2mqtt          | Zigbee2MQTT:info  2021-12-18 08:38:26: Starting Zigbee2MQTT version 1.22.1 (commit #c2b5229)
zigbee2mqtt          | Zigbee2MQTT:info  2021-12-18 08:38:26: Starting zigbee-herdsman (0.13.176)
zigbee2mqtt          | Zigbee2MQTT:error 2021-12-18 08:39:34: Error while starting zigbee-herdsman
zigbee2mqtt          | Zigbee2MQTT:error 2021-12-18 08:39:34: Failed to start zigbee
zigbee2mqtt          | Zigbee2MQTT:error 2021-12-18 08:39:34: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
zigbee2mqtt          | Zigbee2MQTT:error 2021-12-18 08:39:34: Exiting...
zigbee2mqtt          | Zigbee2MQTT:error 2021-12-18 08:39:34: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
zigbee2mqtt          |     at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
zigbee2mqtt          |     at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
```

anubisg1, 2021-12-18:
my solution:

1) python -m zigpy_znp.tools.nvram_reset socket://192.168.20.19:6638
2) python -m zigpy_znp.tools.form_network socket://192.168.20.19:6638
3)  python -m zigpy_znp.tools.energy_scan socket://192.168.20.19:6638
4) wipe zigbe2mqtt clean and add the best channel listed in energy scan

badmuz, 2022-03-11:
Hi there. I've been having same issue connecting tube cc2652p2_poe to Zigbee2MQTT. Removing antenna has helped to finally install it. But now I can't connect any Ikea bulb or blinds. The Ikea remotes have paired fine.
Via ZHA everything connects and works fine.

closed 2022-06-07 by tube0013