tube0013/tube_gateways #35: Lower LQI Performance

claymen, 2021-10-02:
Using one of the POE coordinators and finding significantly lower LQI results when using this over my previous Sonoff ZHA bridge

For example with both devices next to each other, the poe coordinator on channel 11 (which I moved from 15 to test if it would improve, it made no change), and the Sonoff on 15

Sonoff
Device info
TS0014
by _TYZB01_bagt1e4o
Zigbee Coordinator
Zigbee info
IEEE: 5c:02:72:ff:fe:cf:53:e0
Nwk: 0x579f
Device Type: EndDevice
LQI: 188
RSSI: -53
Last Seen: 2021-10-02T20:29:53
Power Source: Battery or Unknown

Tubesz
Device info
TS0014
by _TYZB01_bagt1e4o
Zigbee Coordinator
Zigbee info
IEEE: 5c:02:72:ff:fe:cf:53:e0
Nwk: 0x101c
Device Type: EndDevice
LQI: 81
RSSI: Unknown
Last Seen: 2021-10-02T20:32:43
Power Source: Battery or Unknown

Seems really odd, I am also seeing a lot of strange issues recently with this device with the latest ZStack firmware, going to downgrade shortly and see if this resolves it, but since updating the switch doesn't even work at all on the tubesz coordinator.

Has anyone else come across this?

tube0013, 2021-10-02:
All issues occurred after the FW update or id they exist before as well? Many people are using the new FW now, as it's been out for a while without issues.

When did you order?

thanks.

tube0013, 2021-10-02:
I looked it up as you user name is you name.

Can you use the zigpy-znp tools energy scan? and post the results?

TLDR:

there mayne an issue on the zigbee module with the ipex antenna connector. (It's an ["easy" fix](https://www.youtube.com/watch?v=CQy530D5-XQ) if you have or know someone with access to a hot air solder rework station) but if not I will exchange it - or - try to work with you to find someone in your country region who can perform the fix. A few units slipped out before I was aware of the issue. I worked with the zigbee module manufacturer to find the solution and have been more specifically testing for it before shipping to be safe. But unfortunately a few units from early August - Mid August seem to be affected.

claymen, 2021-10-02:
LQI has been lower compared to the Sonoff from the start. But had been seeing some odd behaviour recently where all the devices just refused to talk directly to the coordinator, so was getting some long delays switching things off and on. Device shipped with 20210120 firmware, so yesterday pulled everything down, backed up both the nvram and the network config so I could full restore if needed, and then upgraded to the 20210708. All seemed to go well, everything was working that I checked but obviously missed this switch which also happens to be the only 4 gang that I have (https://www.ozsmartthings.com.au/products/zigbee-quad-light-switch-1)

Hence tonight I decided to do a full tear down, move the channel over from 15 (default) to 11. Ran nvram reset, removed everything from HA, deleted entities from the device registry and started fresh. Ran the energy check to validate that the Zigbee channel 11 would be the best option now showing the least interference. 

Channel energy (mean of 5 / 5):
------------------------------------------------
 + Lower energy is better
 + Active Zigbee networks on a channel may still cause congestion
 + TX on 26 in North America may be with lower power due to regulations
 + Zigbee channels 15, 20, 25 fall between WiFi channels 1, 6, 11
 + Some Zigbee devices only join networks on channels 15, 20, and 25
------------------------------------------------
 - 11     5.18%  #####
 - 12     8.71%  ########
 - 13     9.65%  #########
 - 14    10.12%  ##########
 - 15    35.29%  ###################################
 - 16    30.82%  ##############################
 - 17    44.71%  ############################################
 - 18    30.35%  ##############################
 - 19    32.71%  ################################
 - 20    20.47%  ####################
 - 21    25.65%  #########################
 - 22    28.94%  ############################
 - 23    26.59%  ##########################
 - 24    15.29%  ###############
 - 25     0.00%
 - 26*   16.47%  ################

Re-paired everything and still saw the weirdness on the 4 gang. So spun up a separate test VM of HA + re-connected the Sonoff and compared everything. 

I've got debug enabled and you see a lot of this
2021-10-02 20:58:04 DEBUG (MainThread) [zigpy.device] [0x4850] Extending timeout for 0x47 request
2021-10-02 20:58:04 DEBUG (MainThread) [zigpy.device] [0x4850] Extending timeout for 0x48 request
2021-10-02 20:58:04 DEBUG (MainThread) [zigpy.device] [0x4850] Extending timeout for 0x49 request
2021-10-02 20:58:04 DEBUG (MainThread) [zigpy.device] [0x4850] Extending timeout for 0x4a request

And similar when you toggle the switches.
2021-10-02 20:58:26 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x4850:1:0x0006]: command failed: 'on' args: '()' kwargs '{}' exception: 'Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'
2021-10-02 20:58:26 DEBUG (MainThread) [homeassistant.components.zha.entity] light.tyzb01_bagt1e4o_ts0014_e053cffe_on_off: turned on: {'on_off': DeliveryError('Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>')}
2021-10-02 20:58:26 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x4850:1:0x0006]: command failed: 'off' args: '()' kwargs '{}' exception: 'Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>'
2021-10-02 20:58:26 DEBUG (MainThread) [homeassistant.components.zha.entity] light.tyzb01_bagt1e4o_ts0014_e053cffe_on_off: turned off: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>

Ahhh I just finished writing this as you updated above. I watched the video, is it just a case of re-heating the connector so it's re-flowing the solder?

tube0013, 2021-10-02:
that energy scan doesn't look as bad as it could be, but it still may be a bit out of position.

Yes essentially heat it to reflow, you may need to give it a nudge too toward the shield. I know the heat gun sort of blocked the view in the video, I haven't had a chance to re-do it yet.

claymen, 2021-10-02:
Just doing some more testing with a few more devices. I think you might be on the money with the IPEX connector. Will see if I can get access to a hot air rework station and give that run through.

**Single Gang (Same room, 2m from coordinator)**

**Sonoff**
Device info
TS0011
by _TYZB01_8ppvdbpz
Zigbee Coordinator
Zigbee info
IEEE: 5c:02:72:ff:fe:cf:59:85
Nwk: 0x0b6f
Device Type: EndDevice
LQI: 240
RSSI: -40
Last Seen: 2021-10-02T21:04:52
Power Source: Battery or Unknown

**PoE**
Device info
TS0011
by _TYZB01_8ppvdbpz
Zigbee Coordinator
Zigbee info
IEEE: 5c:02:72:ff:fe:cf:59:85
Nwk: 0x9470
Device Type: EndDevice
LQI: 135
RSSI: Unknown
Last Seen: 2021-10-02T21:05:33
Power Source: Battery or Unknown

**Single Gang (Next Room)**

**Sonoff**
Device info
TS0011
by _TYZB01_8ppvdbpz
Zigbee Coordinator
Zigbee info
IEEE: 5c:02:72:ff:fe:cf:59:17
Nwk: 0xb39c
Device Type: EndDevice
LQI: 184
RSSI: -54
Last Seen: 2021-10-02T21:11:34
Power Source: Battery or Unknown

**PoE**
Device info
TS0011
by _TYZB01_8ppvdbpz
Zigbee Coordinator
Zigbee info
IEEE: 5c:02:72:ff:fe:cf:59:17
Nwk: 0xb575
Device Type: EndDevice
LQI: 30
RSSI: Unknown
Last Seen: 2021-10-02T21:10:45
Power Source: Battery or Unknown

tube0013, 2021-10-02:
okay let me know if you have any questions, I have a few contacts down your way from when I put out a ping a few weeks ago for anyone who could help with this fix for another device.

claymen, 2021-10-02:
> okay let me know if you have any questions, I have a few contacts down your way from when I put out a ping a few weeks ago for anyone who could help with this fix for another device.

Should be able to get access to some equipment, out of interest what temperature/air flow rates were you using?

tube0013, 2021-10-02:
Slowest flow, 350c

claymen, 2021-10-02:
> Slowest flow, 350c

Cheers, will see how we go tomorrow. Thank you for the help.

tube0013, 2021-10-02:
Slowest flow, 350c

claymen, 2021-10-03:
Not having a great deal of luck, have had a few goes at aligning the connector and it should be in the right spot. But still seeing a fairly big difference in signal. Same device, connected to the Sonoff, then re-paired to the POE and pulled the LQI after toggling the switch a few times.

It definitely was up to temp, was able to nudge it and move it, gave it some time after to try and make sure it all re-flowed. I've repeated this about 3 times now. I've also tested it with USB power instead of PoE just in case there was something funky there (grounding or something maybe?) and no difference.

Any other suggestions?

PoE
Device info
TS0011
by _TYZB01_8ppvdbpz
Zigbee Coordinator
Zigbee info
IEEE: 5c:02:72:ff:fe:cf:59:85
Nwk: 0xc58e
Device Type: EndDevice
LQI: 123
RSSI: Unknown
Last Seen: 2021-10-03T20:50:11
Power Source: Battery or Unknown

Sonoff
Device info
TS0011
by _TYZB01_8ppvdbpz
Zigbee Coordinator
Zigbee info
IEEE: 5c:02:72:ff:fe:cf:59:85
Nwk: 0x8746
Device Type: EndDevice
LQI: 255
RSSI: -36
Last Seen: 2021-10-03T20:53:07
Power Source: Battery or Unknown

tube0013, 2021-10-03:
did you do another energy to compare it to before and after the reflow?

LQI 123 vs 255 - Sonoff is a EFR32 - EZSP Stack radio, CC2652 is TI Z-Stack radio. they compute LQI differently. as do the devices in the mesh that report it.  Maybe it it is s super low number I'd worry a little, but you really can't get into the weeds on these numbers because due to the radios/underlying stacks, they compute it differently. The most reliable test is whether the device is working reliably.

I know you mentioned things seemed to be a little out of sorts - is anything better after the reflow?

thanks.

claymen, 2021-10-03:
The same unit that had weird issues still had them, though after restarting the coordinator it seemed to resolve itself, which does make me suspect more a firmware issue than anything else. Overall the LQI's seem about the same but will let the mesh sort itself out overnight and see where things settle.

claymen, 2021-10-05:
Been keeping an eye on things, definitely seeing improved links to the IKEA power plugs that act as routers. Previously those were much worse off.
[image]

What is strange is that none of the end devices are talking to the coordinator, instead connecting to much further away routers. For example LSW-Study is in the same room as the coordinator, but yet it wants to talk to my outlet in here, rather than the coordinator. So still a bit funky, but this seems more like a Zigbee/firmware type thing rather than something specific with the coordinator hardware.
[image]