tube0013/tube_gateways #36: Energy scan issue

mmychu, 2021-10-10:
I'm trying to debug why I can't start using Tube's CC2652p2, but with no success
I can't start the device, tried with unscrewed antenna but with no success. Have this issue  https://github.com/Koenkk/zigbee2mqtt/issues/8282 

Later, wanted to debug Energy scan but with no success too
[image]

tube0013, 2021-10-10:
Which version coordinator do you have?

If network base are you able to access the esphome web ui via a web browser and going to the ip?

mmychu, 2021-10-10:
Yes, I'm using HasOS VM (on Proxmox). I'm back to conbee2 with z2m and will try again tomorrow

closed 2022-09-02 by mmychu