tube0013/tube_gateways #37: Pre-compiled router firmware for E280-ZG120B

bennydiamond, 2021-10-18:
I know it's a long shot but would it be possible to generate a router-capable firmware for the EFR32 device on those cheap E180-ZG120B-TB dev boards? Like @tube0013 , I have too many coordinators laying around!

I'd like to try on my own but it seems access to the Scilabs ZigBee SDK is paywalled...

tube0013, 2021-10-18:
I toyed with this a while back unsuccessfully. I'll re-try. But the thing with the E180-ZG120B-TB is you need a j-link to flash over swd, you can't flash it from the factory overt usb as the bootloader doesn't allow it.

bennydiamond, 2021-10-18:
Thanks alot! I am aware of the SWD probe requirement. I do have my J-Link ready to go.

Hedda, 2021-12-07:
You could try asking nicely in issue for these Silabs Zigbee firmware repos if Adminiuga or grobasoz could compile router firmware:

https://github.com/zha-ng/EZSP-Firmware

https://github.com/grobasoz/zigbee-firmware/

FYI, pre-configured Zigbee Router firmware for EFR32MG21 adapters has now been released by xsp1989 on GitHub. From the readme posted it sounds as if so far only tested with ITead Zigbee 3.0 USB Dongle and SM-011 based USB adapters. I doubt however that xsp1989 would release firmware for anything else than ITead Zigbee 3.0 USB Dongle and SM-011 Zigbee modules since I believe he is an engineer that works for CoolKit Technologies who designed and makes ITead Zigbee 3.0 USB Dongle and  "SM-011 V1.0" Zigbee modules.

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle/RouterForDongle

https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle/RouterForDongle/README.md

https://github.com/xsp1989/zigbeeFirmware/issues/16

https://github.com/xsp1989/zigbeeFirmware/issues/2

Hedda, 2021-12-07:
Anyway, you would probably be better of buying the same Silabs MGM210P (EFR32MG21 based radio modules) that tube uses:

https://www.silabs.com/wireless/zigbee/efr32mg21-series-2-modules

Alternatively, repurpose ITead's Zigbee 3.0 USB Dongle (has TX and RX solder pads exposed). Also sold out due to chip shortage.

https://itead.cc/product/zigbee-3-0-usb-dongle/ 

> I'd like to try on my own but it seems access to the Scilabs ZigBee SDK is paywalled...

Known issue and as I understand minimum needed is to buy is the $99 dev kit and register it to get access to Silabs Zigbee SDK:

https://www.silabs.com/development-tools/wireless/efr32xg22-wireless-starter-kit

https://www.cnx-software.com/2020/06/25/efr32xg22-wireless-gecko-starter-kit-targets-bluetooth-zigbee-and-proprietary-wireless-development/

Silabs official SLWRB4180B / SLWRB4180A "EFR32xG21" radio board made for that dev kit can be purchased separately for $49

https://www.silabs.com/development-tools/wireless/slwrb4180b-efr32xg21-wireless-gecko-radio-board

https://www.silabs.com/development-tools/wireless/slwrb4180a-efr32xg21-wireless-gecko-radio-board

https://www.silabs.com/documents/public/user-guides/ug427-brd4180b-user-guide.pdf