tube0013/tube_gateways #40: Unable to update or connect to zigbee2mqtt with cc2652p2_poe.

Eagleman7, 2021-11-01:
Trying to get the cc2652p2_poe working, however I keep getting this error in zigbee2mqtt:
```
Zigbee2MQTT:error 2021-11-01 17:39:16: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.js:116:27)
    at start (/app/index.js:95:5)
```

So I decided to install firmware to see if this resolves the issue. However when I click `Prep the cc2652p2 for firmware update` nothing happens in the debug log. This is after waiting 2 minutes:
[image]
Switching browsers or incognito also doesnt work. Is the device dead somehow?

tube0013, 2021-11-01:
Hi Sorry for the trouble. If nothing is showing in the Debug log after using any of the buttons above it likely means the connection between the browser and the device has dropped. 

You can you try and re-flash the ESP32/ESPHome binary - .bin available here:  https://github.com/tube0013/tube_gateways/tree/main/tube_poe_coordinator

this is done on the Devices web page via the OTA option.

 you can also do it over usb (I like the esphome flasher app) - this method clears out the memory first. the USB port is accessible if you open the case. DO NOT keep powered ethernet plugged in if you use this - there is also a warning on the PCB about this.

Eagleman7, 2021-11-01:
Thanks, the buttons are back!
[image]
Do I still need to do the update using these files? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

tube0013, 2021-11-01:
the zigbee module comes pre-flashed with the latest fw from there, so there should be no need to update it.

when you try to connect z2m to the device, do you see anything in the debug log now?

Eagleman7, 2021-11-01:
Maybe I don't understand the toggle, I need to wait for something to show in the debug log before I can update, right?

tube0013, 2021-11-01:
There is most likely no need to update the fw on the zigbee module. it comes preflashed. the issues you are describing are with the ESP32 device. - getting connected and staying connected. 

Are you seeing any logs in the Debug Log in the browser? even when you reboot the device with the button it should show something. It should also show when a serial device connects/disconnects. if nothing is showing we need to troubleshoot the esp32's ethernet connection. maybe try a fw bin with a static ip set.

Eagleman7, 2021-11-01:
I never saw anything appearing in the debug log at all stages.

[image]

```
[root@control cc2538-bsl]# python3 cc2538-bsl.py -p socket://zigbee-gateway.corp.pootenterprise.com:6638  -evw ../CC1352P2_CC2652P_launchpad_coordinator_20210708.hex
Opening port socket://zigbee-gateway.corp.example.com:6638, baud 500000
Reading data from ../CC1352P2_CC2652P_launchpad_coordinator_20210708.hex
Firmware file: Intel Hex
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Get Status (0x23)'
```

tube0013, 2021-11-01:
the Firmware update toggle runs a script on the ESP32 that toggles on the BSL button, toggles the Reset of the zigbee module, then waits about 10 secs releases the BSL, and prints a message to use the command line update tool to flash the module. 

Again it shouldn't be necessary though, I flash all modules and test them, then add them to the boards, test again through the esp32 and then reset the NVRAM all over the ethernet connection before shipping the devices.

Eagleman7, 2021-11-01:
Lets skip the firmware through cc2538-bsl.pu for now then :). zigbee2mqtt is still having issues connecting: 
[image]

tube0013, 2021-11-01:
is Z2M pointing to an IP or a hostname? if hostname try the ip

Eagleman7, 2021-11-01:
Tried using IP, same problem :(

```
advanced:
  homeassistant_discovery_topic: homeassistant    
  homeassistant_status_topic: homeassistant/status
  last_seen: ISO_8601
  log_level: info
  log_output:
  - console
  network_key: GENERATE
  pan_id: GENERATE
device_options:
  retain: true
experimental:
  new_api: true
frontend:
  port: 8080
homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  include_device_information: true
  password: admin
  server: mqtt://mosquitto.home.svc.cluster.local
  user: admin
permit_join: true
serial:
  port: tcp://192.168.3.10:6638
```

Can I test that port from my computer instead, to see if something is running on that port?

tube0013, 2021-11-01:
yes, you can try a serial program that can connect to a raw serial port over network connection. I use a Mac App called Serial.

I suspect the ESP32 is not staying connected - this would explain nothing in the debug log window as well. 

Something you can try - use the internal USB port to power the device via a computer and use an UN-Powered ethernet cable (this is imperative)

this will allow you to use a serial app on the computer to monitor the ESPHOME console log and not have to rely on the network connection.

Be sure the ethernet connection is un-powered if using the USB - if not there will be smoke.

Eagleman7, 2021-11-01:
Can I safely tilt out the ESP board? I am cannot find an USB cable that will fit, there is only 3mm room for the connector. I also turned off the poe on the switch and the device is now turned off. Should I be able to connect using PuTTy, if so, on what IP?

tube0013, 2021-11-01:
putty should work.

the only thing really holding the board down is friction, and the SMA antenna connector - it may be best to undo the retaining nut there and just remove the board.

tube0013, 2021-11-01:
For putty you'd want to do something like this to connect over IP Serial:

[image]

Eagleman7, 2021-11-01:
No smoke (yet)!
[image]

However I cannot seem to get anything on port 6638 using raw:
[image]

Eagleman7, 2021-11-01:
Ow, I just got some input, but its not readable by putty:
[image]

tube0013, 2021-11-01:
try and reflash the ESP32 bin over USB with the esphome app

https://github.com/esphome/esphome-flasher/releases

leave the ethernet connected and this the esphome log will show in the app after the flash.

Eagleman7, 2021-11-01:
It cannot find a suitable serial port, it is connected to the PC now:
[image]

Eagleman7, 2021-11-01:
I cannot try on the Mac as well because I currently do not have an USB hub with USB A :(

tube0013, 2021-11-01:
USB cable has data, and isn't power only?  By the way thanks for your patience with this.

Eagleman7, 2021-11-01:
It seems I am missing a driver :) 
[image]

Thank you for helping me as well :)

tube0013, 2021-11-01:
Driver for windows is here:

https://www.olimex.com/Products/Breadboarding/BB-CH340T/resources/CH341SER.zip

Eagleman7, 2021-11-01:
Thanks! Got some more verbosity this time :)

From the flasher software:
```
Using 'COM5' as serial port.
Connecting................
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting..6 %)Writing at 0x000e8df6... (100 %)Wrote 904512 bytes (501591 compressed) at 0x00010000 in 12.8 seconds (effective 565.3 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[19:51:50][I][logger:166]: Log initialized
[19:51:50][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[19:51:50][I][app:029]: Running through setup()...
[19:51:50][C][uart_esp32:072]: Setting up UART...
[19:51:50][C][switch.gpio:011]: Setting up GPIO Switch 'zRST_gpio'...
[19:51:50][D][switch:025]: 'zRST_gpio' Turning OFF.
[19:51:50][D][switch:045]: 'zRST_gpio': Sending state ON
[19:51:50][D][switch:025]: 'zRST_gpio' Turning OFF.
[19:51:50][C][switch.gpio:011]: Setting up GPIO Switch 'zBSL'...
[19:51:50][D][switch:025]: 'zBSL' Turning OFF.
[19:51:50][D][switch:045]: 'zBSL': Sending state ON
[19:51:50][D][switch:025]: 'zBSL' Turning OFF.
[19:51:50][C][ethernet:025]: Setting up Ethernet...
[19:52:05][W][ethernet:041]: Connecting via ethernet failed! Re-connecting...
[19:52:13][I][ethernet:052]: Connected via Ethernet!
[19:52:13][C][ethernet:220]:   IP Address: 192.168.3.10
[19:52:13][C][ethernet:221]:   Hostname: 'tube_zb_gw_cc2652p2_poe'
[19:52:13][C][ethernet:222]:   Subnet: 255.255.255.0
[19:52:13][C][ethernet:223]:   Gateway: 192.168.3.1
[19:52:13][C][ethernet:226]:   DNS1: 8.8.8.8
[19:52:13][C][ethernet:228]:   DNS2: 8.8.4.4
[19:52:13][C][ethernet:231]:   MAC Address: 94:B9:7E:6B:00:3F
[19:52:13][C][ethernet:232]:   Is Full Duplex: YES
[19:52:13][C][ethernet:233]:   Link Up: YES
[19:52:13][C][ethernet:234]:   Link Speed: 100
[19:52:13][C][web_server:073]: Setting up web server...
[19:52:13][C][ota:029]: Over-The-Air Updates:
[19:52:13][C][ota:030]:   Address: tube_zb_gw_cc2652p2_poe.local:3232
[19:52:13][C][api:022]: Setting up Home Assistant API server...
[19:52:13][C][streamserver:027]: Setting up stream server...
[19:52:13][I][app:059]: setup() finished successfully!
[19:52:13][I][app:105]: ESPHome version 1.17.1 compiled on May 12 2021, 22:11:35
[19:52:13][C][uart_esp32:088]: UART Bus:
[19:52:13][C][uart_esp32:090]:   TX Pin: GPIO16
[19:52:13][C][uart_esp32:093]:   RX Pin: GPIO5
[19:52:13][C][uart_esp32:094]:   RX Buffer Size: 256
[19:52:13][C][uart_esp32:096]:   Baud Rate: 115200 baud
[19:52:13][C][uart_esp32:097]:   Data Bits: 8
[19:52:13][C][uart_esp32:098]:   Parity: NONE
[19:52:13][C][uart_esp32:099]:   Stop bits: 1
[19:52:13][C][switch.gpio:042]: GPIO Switch 'zRST_gpio'
[19:52:13][C][switch.gpio:042]:   Inverted: YES
[19:52:13][C][switch.gpio:043]:   Pin: GPIO33 (Mode: OUTPUT)
[19:52:13][C][switch.gpio:059]:   Restore Mode: Always OFF
[19:52:13][C][template.switch:058]: Template Switch 'zRST'
[19:52:13][C][template.switch:059]:   Restore State: NO
[19:52:13][C][template.switch:060]:   Optimistic: NO
[19:52:13][C][switch.gpio:042]: GPIO Switch 'zBSL'
[19:52:13][C][switch.gpio:042]:   Inverted: YES
[19:52:13][C][switch.gpio:043]:   Pin: GPIO32 (Mode: OUTPUT)
[19:52:13][C][switch.gpio:059]:   Restore Mode: Always OFF
[19:52:13][C][template.switch:058]: Template Switch 'Prep the cc2652p2 for firmware update'
[19:52:13][C][template.switch:059]:   Restore State: NO
[19:52:13][C][template.switch:060]:   Optimistic: NO
[19:52:13][C][logger:185]: Logger:
[19:52:13][C][logger:186]:   Level: DEBUG
[19:52:13][C][logger:187]:   Log Baud Rate: 115200
[19:52:13][C][logger:188]:   Hardware UART: UART0
[19:52:13][C][restart:021]: Restart Switch 'Restart the Gateway'
[19:52:13][C][restart:021]:   Icon: 'mdi:restart'
[19:52:13][C][ethernet:066]: Ethernet:
[19:52:13][C][ethernet:220]:   IP Address: 192.168.3.10
[19:52:13][C][ethernet:221]:   Hostname: 'tube_zb_gw_cc2652p2_poe'
[19:52:13][C][ethernet:222]:   Subnet: 255.255.255.0
[19:52:13][C][ethernet:223]:   Gateway: 192.168.3.1
[19:52:13][C][ethernet:226]:   DNS1: 8.8.8.8
[19:52:13][C][ethernet:228]:   DNS2: 8.8.4.4
[19:52:13][C][ethernet:231]:   MAC Address: 94:B9:7E:6B:00:3F
[19:52:13][C][ethernet:232]:   Is Full Duplex: YES
[19:52:13][C][ethernet:233]:   Link Up: YES
[19:52:13][C][ethernet:234]:   Link Speed: 100
[19:52:13][C][ethernet:068]:   Power Pin: GPIO12 (Mode: OUTPUT)
[19:52:13][C][ethernet:069]:   MDC Pin: 23
[19:52:13][C][ethernet:070]:   MDIO Pin: 18
[19:52:13][C][ethernet:071]:   Type: LAN8720
[19:52:13][C][web_server:136]: Web Server:
[19:52:13][C][web_server:137]:   Address: tube_zb_gw_cc2652p2_poe.local:80
[19:52:13][C][ota:029]: Over-The-Air Updates:
[19:52:13][C][ota:030]:   Address: tube_zb_gw_cc2652p2_poe.local:3232
[19:52:13][C][api:095]: API Server:
[19:52:13][C][api:096]:   Address: tube_zb_gw_cc2652p2_poe.local:6053
[19:52:13][C][streamserver:074]: Stream Server:
[19:52:13][C][streamserver:075]:   Address: tube_zb_gw_cc2652p2_poe.local:6638

```

tube0013, 2021-11-01:
try and connect z2m now.

Eagleman7, 2021-11-01:
```
Using '/data' as data directory
Zigbee2MQTT:info  2021-11-01 19:53:58: Logging to console only'
Zigbee2MQTT:info  2021-11-01 19:53:58: Starting Zigbee2MQTT version 1.21.2 (commit #4492212
)
Zigbee2MQTT:info  2021-11-01 19:53:58: Starting zigbee-herdsman (0.13.149)
Zigbee2MQTT:info  2021-11-01 19:54:03: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2021-11-01 19:54:03: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2021-11-01 19:54:03: Currently 0 devices are joined:
Zigbee2MQTT:info  2021-11-01 19:54:03: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2021-11-01 19:54:04: Connecting to MQTT server at mqtt://mosquitto.home.svc.cluster.local
Zigbee2MQTT:info  2021-11-01 19:54:04: Connected to MQTT server
Zigbee2MQTT:info  2021-11-01 19:54:04: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2021-11-01 19:54:04: Started frontend on port 0.0.0.0:8080
```

Flasher logs:
```
[19:53:59][D][streamserver:085]: New client connected from 192.168.5.20
```

tube0013, 2021-11-01:
progress!

now I'd try PoE again (with USB disconnected)

Eagleman7, 2021-11-01:
zigbee2mqtt is able to connect, but I dont think the debug log is working on the webui.

tube0013, 2021-11-01:
Good that the z2m is working now. but very odd nothing in the web debug log.. I've actually never seen that occur beyond having the webpage up when I've unpowered a device.  To be honest as long as z2m is working you should be good. the PCB has a BLS button on it to get into the module bootloader mode for FW flashing. 

Another thing you could try is pulling down the esphome .yaml for the device and adding it to your esphome folder (if you have it) then ESPHome should show the device in it's frontend, and you can us the Logs button there to connect to to the logs (uses same api connection as the web portal so may or may not work).

[image]

Eagleman7, 2021-11-04:
Thanks for your help @tube0013! I do have ESPhome running in my network, so that is worth investigating. I am not sure if I need to do this firmware update https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin ?

tube0013, 2021-11-04:
It already has the latest build installed. So really no need.

tube0013, 2021-11-29:
Copy of the yaml is here:

https://github.com/tube0013/tube_gateways/tree/main/esphome

tube0013, 2021-12-17:
how are things going @Eagleman7 okay to close this?

Eagleman7, 2021-12-23:
Hi @tube0013,

Its been working perfect for weeks now, thanks!

closed 2021-12-23 by Eagleman7