tube0013/tube_gateways #41: HomeAssistant ZHA can't find new Router/Repeater

Drizzt321, 2021-11-03:
I recently was able to order and received 2 of the Router/Repeater devices. I have the Ethernet/Serial Coordinator which I've been using successfully just fine. The 1st Router/Repeater was found right away by HA/ZHA, joined quickly and great. However the 2nd Router/Repeater I have been unable to get it to be found. I've tried hitting the BSL and RST buttons separately, while powered, but still nothing.

tube0013, 2021-11-03:
Hi, sorry for the troubles. that seams odd. can you send me a note with your name/order # at store @ tubeszb.com and I'll help get this made right for you.

Drizzt321, 2021-11-03:
@tube0013 not sure I need a replacement, I'm happy to do some troubleshooting first. I'm comfortable with CLI and serial and such, and have FTDI and such from my own projects.

tube0013, 2021-11-03:
Alright!

So the LEDs don't do anything with the current z-stack router build which is why I don't ship units with them.

For testing. use a multimeter to see if you are getting 3.3 volts on the 3.3 V pin when plugged into USB. the only real hardware issue/fix could be a bad USB connection and this will tell that for sure. I do test the connections and do a test joining of the router on all units before shipping but can't rule it out. 

Ff it has power you can try to re-flash the the module with a USB > Serial and the 4 pin header.  FW is here (launchpad variant) <https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0/bin>

Flashing tool is here: <https://github.com/JelmerT/cc2538-bsl>

I typically have the USB convert plugged into my computer and hold the BSL button while plugging in the 4 female ends of the jumper wires into the board.

then run the flashing command:
```./cc2538-bsl.py -p /dev/cu.usbserial-00F9450F -evw ../CC1352P2_CC2652P_launchpad_router_20210128.hex```
(adjust paths for you use)

closed 2021-11-03 by Drizzt321

Drizzt321, 2021-11-03:
Happy to! Would rather get something working than toss it if I can. 

Oh, and I don't know if 3D printing is the bottleneck for you for the cases, I suspect not, but many folks have 3D printers, dunno if it'd make sense to offer them without the case and just point people to the STL files here for a case. Probably not worth it though.

tube0013, 2021-11-03:
generally the 3d print's are not the bottle neck. I stockpile them during out stock outages to get ahead of the game. I look at them as a means for protection in shipping as well so that's the main reason I don't offer without.