tube0013/tube_gateways #43: Low LQI when using cc2652p2-v2 w/ Z2M 1.22.0-2

davidcoulson, 2021-11-16:
I just switched from Conbee II to cc2652p2-v2 - I wiped out my Z2M install and re-paired everything to the cc2652p2-v2. While everything seems to be working, the LQI scores for every device on my network is <70 instead of >200 like on the Conbee II. Even a Hue bulb less than 6ft from the cc2652p2-v2 has a LQI of <100, but was 255 with the Conbee II.

Not sure if this is a bug with the FW on the cc2652p2-v2, or something wrong with the device. I have tried changing the orientation of the antenna with no change. I have the Conbee disconnect and the cc2652p2-v2 is in the same spot as the stick was.

tube0013, 2021-11-16:
LQI is reported differently by different radio chipsets. So it's not uncommon to see different values with a different coordinator. The real test is of every works or not.

davidcoulson, 2021-11-16:
Yep - Everything is working great, so just a cosmetic thing with LQI values not scaling between 0-255 as expected. No big deal. Thanks!!

closed 2021-11-16 by davidcoulson