tube0013/tube_gateways #44: compile failed in latest esphome

tmottabr, 2021-11-18:
while doing some house keeping i saw that esphome was updated in my home assistant and tried to update my poe board. however it failed to compile.

i am on currently 2021.9.2, all working fine, was trying to compile 2021.11.1.

```
INFO Reading configuration /config/esphome/tube_zb_gw_cc2652p2_olimex_POE.yaml...
WARNING 'tube_zb_gw_cc2652p2_poe': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Generating C++ source...
INFO Compiling app...
Processing tube_zb_gw_cc2652p2_poe (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 1.0
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/api_connection.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/api_server.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/list_entities.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/proto.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/user_services.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/util.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/esp32/core.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/esp32/gpio_idf.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/ethernet/ethernet_component.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/json/json_util.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/logger/logger.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/md5/md5.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/network/util.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/restart/restart_switch.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/script/script.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/switch/automation.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/switch/switch.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/template/switch/template_switch.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/uart/uart.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/uart/uart_component.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/uart/uart_component_esp32_arduino.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/uart/uart_component_esp8266.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/uart/uart_component_esp_idf.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/uart/uart_debugger.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/web_server/web_server.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/web_server_base/web_server_base.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/core/application.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/core/color.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/core/component.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/core/controller.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/core/entity_base.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/core/helpers.cpp.o
[... 27 lines cut ...]
     ESP_LOGCONFIG(TAG, "  Address: %s:%u", network_get_address().c_str(), this->port_);
     ^
*** [/data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/stream_server.cpp.o] Error 1
/config/esphome/tube_zb_gw_cc2652p2_olimex_POE.yaml: In lambda function:
/config/esphome/tube_zb_gw_cc2652p2_olimex_POE.yaml:86:62: error: no matching function for call to 'StreamServerComponent::StreamServerComponent(esphome::uart::ESP32ArduinoUARTComponent*&)'
       auto stream_server = new StreamServerComponent(id(uart_bus));
                                                              ^
In file included from src/main.cpp:56:0:
src/stream_server.h:35:14: note: candidate: StreamServerComponent::StreamServerComponent(Stream*)
     explicit StreamServerComponent(Stream *stream) : stream_{stream} {}
              ^
src/stream_server.h:35:14: note:   no known conversion for argument 1 from 'esphome::uart::ESP32ArduinoUARTComponent*' to 'Stream*'
src/stream_server.h:33:7: note: candidate: StreamServerComponent::StreamServerComponent(const StreamServerComponent&)
 class StreamServerComponent : public esphome::Component {
       ^
src/stream_server.h:33:7: note:   no known conversion for argument 1 from 'esphome::uart::ESP32ArduinoUARTComponent*' to 'const StreamServerComponent&'
src/stream_server.h:33:7: note: candidate: StreamServerComponent::StreamServerComponent(StreamServerComponent&&)
src/stream_server.h:33:7: note:   no known conversion for argument 1 from 'esphome::uart::ESP32ArduinoUARTComponent*' to 'StreamServerComponent&&'
/config/esphome/tube_zb_gw_cc2652p2_olimex_POE.yaml:87:28: error: could not convert '{stream_server}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::Component*>'
       return {stream_server};
                            ^
*** [/data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/main.cpp.o] Error 1
========================= [FAILED] Took 29.45 seconds =========================

```

tube0013, 2021-11-18:
Yes the serial component isn't compatible with esphome 2021.10 and up.

There is an upstream fix, but I haven't had time to test or try to incorporate it into the version I use which is derived from it but has some additional features.

For now I just advise against updating (as you really can't). Hope to have a working fix soon.

Heydenes, 2021-11-20:
[Fix Available]
I used the older version of ESPHome on Windows.

Installed Docker on Windows.

docker pull esphome/esphome:2021.8.1

created container with the above Image - with mount as 
/C/MyData/esphome:/config   --Mount
6052:6052 --Port

Copied the .h,.cpp,.yaml files to mount location
Opened the URL - localhost:6052 
Build the file and select to install manually and the bin was downloaded automatically.

Worked like charm!

anubisg1, 2021-12-16:
this issue remains true with 2021.12.1

```
INFO Reading configuration /config/esphome/tube-zb-gw-cc2652p2-poe.yaml...
WARNING 'tube_zb_gw_cc2652p2_poe': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing tube_zb_gw_cc2652p2_poe (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 3.3.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32 @ 3.3.2 has been installed!
The platform 'platformio/espressif32 @ 3.3.2' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing platformio/toolchain-xtensa32 @ ~2.50200.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa32 @ 2.50200.97 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.10006.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32 @ 3.10006.210326 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.30100.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy @ 1.30100.210531 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40200.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%          
Tool Manager: tool-scons @ 4.40200.0 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 1.0
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/api_connection.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/api_server.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/list_entities.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/proto.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/user_services.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/api/util.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/esp32/core.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/esp32/gpio_idf.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/ethernet/ethernet_component.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/json/json_util.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/logger/logger.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/md5/md5.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/network/util.cpp.o
Compiling /data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
[... cut ...]
src/stream_server.h:32:7: note:   no known conversion for argument 1 from 'esphome::uart::ESP32ArduinoUARTComponent*' to 'StreamServerComponent&&'
/config/esphome/tube-zb-gw-cc2652p2-poe.yaml:93:28: error: could not convert '{stream_server}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::Component*>'
       return {stream_server};
                            ^
*** [/data/tube_zb_gw_cc2652p2_poe/.pioenvs/tube_zb_gw_cc2652p2_poe/src/main.cpp.o] Error 1
========================= [FAILED] Took 154.07 seconds =========================
```

tube0013, 2021-12-17:
new poe yaml which will compile and work on current esphome: https://github.com/tube0013/tube_gateways/blob/main/esphome/tube_zb_gw_cc2652p2_olimex_POE.yaml

closed 2021-12-17 by tube0013

anubisg1, 2021-12-17:
i can confirm it works. thank you

```
INFO Reading configuration /config/esphome/tube-zb-gw-cc2652p2-poe.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing tube-zb-gw-cc2652p2-poe (board: esp-wrover-kit; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 1.0
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/api/api_connection.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/api/api_server.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/api/list_entities.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/api/proto.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/api/user_services.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/api/util.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/dashboard_import/dashboard_import.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/esp32/core.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/esp32/gpio_idf.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/ethernet/ethernet_component.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/json/json_util.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/logger/logger.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/md5/md5.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/network/util.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/restart/switch/restart_switch.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/script/script.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/stream_server/stream_server.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/switch/automation.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/switch/switch.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/template/switch/template_switch.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/uart/uart.cpp.o
Compiling /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/src/esphome/components/uart/uart_component.cpp.o
[... cut ...]
Flash: [=====     ]  49.5% (used 908414 bytes from 1835008 bytes)
Building /data/tube-zb-gw-cc2652p2-poe/.pioenvs/tube-zb-gw-cc2652p2-poe/firmware.bin
======================== [SUCCESS] Took 176.60 seconds ========================

********************************************************************************
We found 98.03MB of unnecessary PlatformIO system data (temporary files, unnecessary packages, etc.).
Use `pio system prune --dry-run` to list them or `pio system prune` to save disk space.
INFO Successfully compiled program.
```

tube0013, 2022-01-06:
Notice: I have been unable to flash the cc2652 using the newer compiled ESPHome Firmware. I am unsure if the serial component changes or general esphome changes are the cause, but rolling back the firmware to the original binary allows the firmware to flash with no issues.

https://github.com/tube0013/tube_gateways/blob/main/tube_poe_coordinator/tube_zb_gw_cc2652p2_oliemex_poe.bin

anubisg1, 2022-01-08:
> 

So what's the recommendation then? The only reason I needed to rebuild the firmware was to have static ip address.

Is there a new firmware for the cc2652 worth installing? Have been happy with the poe coordinator so far, and given the history of crappy zigbee and drop outs with my old coordinator I don't want to mess up what's working

tube0013, 2022-01-08:
Leave it be then. If you ever want to upgrade the fw just flash back to the version from here and then when done flash your static build again.

I rolled back an esphome install to 2021.9.3 on a VM so happy to compile any static ip builds anyone needs as well.

I'm more of an engineer than programmer so it may be. While before this esphome is "fixed". It's a pretty nuanced issue in that just about everything else works fine it just seems to impact the fw flashing of the module.