tube0013/tube_gateways #45: tube-zb-gw-cc2652p2 compile failed in latest esphome

kylinfedora, 2021-11-25:
esphome  
esphome yaml file:
`esphome:
  name: tube-zb-gw-cc2652p2
  platform: ESP32
  board: esp-wrover-kit

external_components:
  - source: github://oxan/esphome-stream-server

#  includes:
#    - stream_server.h
#    - stream_server.cpp

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
  
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.xxx.xxx
    gateway: 192.168.xxx.x
    subnet: 255.255.255.0
    dns1: 192.168.xxx.x
    
  domain: .xxxxx

# Enable logging
logger:
  level: DEBUG
  
# Enable Home Assistant API
api:
  reboot_timeout: 0s

ota:

web_server:
  port: 80

script:
  - id: fw_update_mode
    then:
      - switch.turn_on: zBSL
      - delay: 1s
      - switch.turn_on: zRST_gpio
      - delay: 1s
      - switch.turn_off: zRST_gpio
      - logger.log: "Delaying ~10 seconds for cc2652p2 to settle"
      - delay: 11s
      - switch.turn_off: zBSL
      - logger.log: "Please try update with cc-bsl tool now"
      - logger.log: "cc-bsl usage: cc-bsl.py -p socket://ip-of-gw:6638 -evw firmware.hex"
      
switch:
  - platform: restart
    name: "Restart the Gateway"

  - platform: gpio
    pin: 33
    id: zRST_gpio
    inverted: yes
    restore_mode: ALWAYS_OFF
  - platform: template
    name: "zRST"
    id: zRST
[... 93 lines cut ...]
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/src/esphome/core/component.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/src/esphome/core/controller.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/src/esphome/core/entity_base.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/src/esphome/core/helpers.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/src/esphome/core/log.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/src/esphome/core/scheduler.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/src/esphome/core/util.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/src/main.cpp.o
Generating partitions /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/partitions.bin
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/lib528/AsyncTCP-esphome/AsyncTCP.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/lib64d/WiFi/ETH.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/lib64d/WiFi/WiFi.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/lib64d/WiFi/WiFiAP.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/lib64d/WiFi/WiFiClient.cpp.o
/config/esphome/tube-zb-gw-cc2652p2.yaml: In lambda function:
/config/esphome/tube-zb-gw-cc2652p2.yaml:99:32: error: expected type-specifier before 'StreamServerComponent'
       auto stream_server = new StreamServerComponent(id(uart_bus));
                                ^
/config/esphome/tube-zb-gw-cc2652p2.yaml:100:28: error: could not convert '{stream_server}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::Component*>'
       return {stream_server};
                            ^
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/lib64d/WiFi/WiFiGeneric.cpp.o
Compiling /data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/lib64d/WiFi/WiFiMulti.cpp.o
*** [/data/tube-zb-gw-cc2652p2/.pioenvs/tube-zb-gw-cc2652p2/src/main.cpp.o] Error 1
========================== [FAILED] Took 9.17 seconds ==========================`

tube0013, 2021-11-25:
Dupe of #44 will have fix soon

kylinfedora, 2021-11-29:
I think just config error, 
change the file like tihs, build ok, but  not test the fw
#custom_component:
#  - lambda: |-
#      auto stream_server = new StreamServerComponent(id(uart_bus));
#      return {stream_server};
stream_server:
   uart_id: uart_bus
   port: 6638

kylinfedora, 2021-12-13:
I build with this config，build is ok, who can test?
```
esphome:
  name: tube-zb-gw-cc2652p2
  platform: ESP32
  board: esp-wrover-kit

external_components:
  - source: github://oxan/esphome-stream-server

#  includes:
#    - stream_server.h
#    - stream_server.cpp

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  reboot_timeout: 0s

ota:

web_server:
  port: 80

script:
  - id: fw_update_mode
    then:
      - switch.turn_on: zBSL
      - delay: 1s
      - switch.turn_on: zRST_gpio
      - delay: 1s
      - switch.turn_off: zRST_gpio
      - logger.log: "Delaying ~10 seconds for cc2652p2 to settle"
      - delay: 11s
      - switch.turn_off: zBSL
      - logger.log: "Please try update with cc-bsl tool now"
      - logger.log: "cc-bsl usage: cc-bsl.py -p socket://ip-of-gw:6638 -evw firmware.hex"

switch:
  - platform: restart
    name: "Restart the Gateway"

  - platform: gpio
    pin: 33
    id: zRST_gpio
    inverted: yes
    restore_mode: ALWAYS_OFF
  - platform: template
    name: "zRST"
    id: zRST
    turn_on_action:
      - switch.turn_on: zRST_gpio
      - delay: 15ms
      - switch.turn_off: zRST_gpio

  - platform: gpio
    pin: 32
    name: "zBSL"
    id: zBSL
    inverted: yes
    restore_mode: ALWAYS_OFF

  - platform: template
    name: "Prep the cc2652p2 for firmware update"
    turn_on_action:
      - script.execute: fw_update_mode
    turn_off_action:
      - switch.toggle: zRST

uart:
  id: uart_bus
  rx_pin: GPIO5
  tx_pin: GPIO17
  baud_rate: 115200

#custom_component:
# - lambda: |-
#     auto stream_server = new StreamServerComponent(id(uart_bus));
#     return {stream_server};
stream_server:
   uart_id: uart_bus
   port: 6638
```

tube0013, 2021-12-17:
confirmed a similar build works (added the feature to adopt a device as well).

2652 v1 and PoE are adapted so far. v2 used a fork of the serial server with some extra features so looking as to how to get those working still.

https://github.com/tube0013/tube_gateways/blob/main/esphome

closed 2021-12-17 by tube0013

tube0013, 2022-01-06:
Notice: I have been unable to flash the cc2652 using the newer compiled ESPHome Firmware. I am unsure if the serial component changes or general esphome changes are the cause, but rolling back the firmware to the original binary allows the firmware to flash with no issues.

https://github.com/tube0013/tube_gateways/blob/main/tube_poe_coordinator/tube_zb_gw_cc2652p2_oliemex_poe.bin