tube0013/tube_gateways #46: Connect to Philips Hue Bulb

ranrinc, 2021-11-26:
Just wondering how can I link my old philips hue bulb to my new cc2652p2. I cant seems to find the bulbs on HA scan ZHA. I cant also find a way to make the bulb go into discovery mode. Anyone have any idea?

MattWestb, 2021-11-26:
ZHA dot have implanting touch link reset :-(
Using one HUE remote doing it or hue thief shall working but you need re power the light before using touch link on it if the firmware is updated

tube0013, 2021-11-26:
I find that if you have the hue hub still it's easiest to have them added to the hub, then remove them from hue at the same time as permitting joins, this puts them in pairing mode upon being removed from hue.

ranrinc, 2021-11-26:
Heemm.. I do have the hub (the old round version) and yes those bulb do able to connect to it. Now, how do I remove them and then discovered it using the zigbee. Sorry nubie here.

tube0013, 2021-11-26:
If they on hue.

(It's been awhile since I've done this)

You go into the light's settings in the hue app and there should be an option to remove/delete device. 

I would trigger ZHA to permit joins right before doing this (it is only active for 60 seconds)

ranrinc, 2021-11-26:
so far no luck. After delete ZHA seems to search and search without luck.

tube0013, 2021-11-26:
any luck adding other devices?

ranrinc, 2021-11-26:
yup.. all my aqara is now move to your device.

tube0013, 2021-11-26:
May want to try discord or forums to get a wider audience who has experience with hue. Mine is pretty limited and was several years ago.

ranrinc, 2022-01-11:
@tube0013 I got new input saying that I need to used zigbee2mqtt touchlink to factoryreset all my hue bulb. However wondering how should I do it? Since currently its connected via ZHA. 

Can I reconnected to a new RPI4 running zigbee2mqtt for only that purposes? is this possible?

https://www.zigbee2mqtt.io/guide/usage/touchlink.html

tube0013, 2022-01-11:
Using z2m like that would wipe the settings from you coordinator so if you go that route be sure to use the zigpy-znp tools and get a backup of the coord before using z2m so you can restore afterwards.

Other options as I understand it. 

If they are the latest gen hue with BLE I believe they can be reset in the app via BLE. 

If you have a hue hub it is possible to join them with the serial number and then delete them from the hue hub and they will reset.

Certain zigbee remotes support touch link and can do the reset (hue remotes for certain I think)

tube0013, 2022-01-12:
If you have the hue hub, this is how I have done it in the past.

Add the bulb to the hue hub.

After a few minutes in the hue app go to the the bulb settings - where you can delete/remove the bulb.

Now but permit joining on ZHA and after that delete the bulb from hue. 

This sequence should reset the bulb and trigger it to go into pairing mode which is what permitting joins is listening for. 

Another option is to remove power from the bulb as soon as it is deleted from hue. Then remove power from the hue bridge to get it out of the way. Then permit joins in ZHA and power up the bulb.

ranrinc, 2022-01-12:
> If you have the hue hub, this is how I have done it in the past.
>
> Add the bulb to the hue hub.
>
> After a few minutes in the hue app go to the the bulb settings - where you can delete/remove the [...]

OK will try.. I think, I have try without any success ZHA just never found the bulb.

I currently have old HUE bulb that have not being used and it pair with ZHA instantly

ranrinc, 2022-01-13:
> zigpy-znp

Let per say I used this route and not using zingpy and do backup. Will it easy to resetup ZHA? since I dont have many device running on zigbee. Once Hue connected to z2m will it be able to reconnect to ZHA?

tube0013, 2022-01-13:
If I was doing it I'd:

1. Use zigpy-znp_tools and create a network backup.

2. Use zigpy-znp_tools to reset the nvram so the device looks new to z2m.

3. Set up z2m and do the touch link. 

4. use zigpy-znp_tools and restore backup created in step 1.

The zigpy-znp-tools site has good instructions for getting it up and running.

ranrinc, 2022-01-13:
Got it... thank you again for all the assist

closed 2022-02-22 by ranrinc