tube0013/tube_gateways #50: Please add version sensor to esphome config

ptr727, 2022-01-15:
I received my tubes 2652 v2, thank you.

I'm wondering how to know what the esphome and zigbee firmware versions are?

For esphome, I run several devices myself, and I add the version sensor to show me the version info, please consider adding the version sensor to the v2 yaml:
https://esphome.io/components/text_sensor/version.html

For the zigbee firmware, it would be cool if esphome could read it from uart before connecting to TCP and set a text sensor to display, or is there a way to see the zigbee firmware in z2mqtt?

tube0013, 2022-01-15:
Esphome version is 2021.09 anything after creates issues with the stream server (serial) where the flash of the cc2652 will fail.

Zigbee firmware version is shown in the z2m about screen. It's also shown in the debug logs of zigpy/znp when it loads.

[image]

ptr727, 2022-01-15:
Ok, understood, won't try to update then.
Would still be good to see the esphome version reported on the web page.

Interesting, the coordinator firmware in about of my tubes v2 is reported as `20210708`, the paper note said it is pre-flashed with latest `20211217`. Is the version incorrectly reported, or is my device not the latest firmware?

tube0013, 2022-01-15:
It's possible I flashed the wrong one and didn't notice. apologies if that is the case.

Here's a ephome fw with the version sensor for the cc2652 ethernet. let me know if you'd like one for the efr32 one as well
[tube-zb-gw-cc2652p2-v2_2022_01_15.bin.zip](https://github.com/tube0013/tube_gateways/files/7875481/tube-zb-gw-cc2652p2-v2_2022_01_15.bin.zip)
.

ptr727, 2022-01-15:
I'm trying to update zigbee firmware now.
Do you have more detailed instructions for config and restore backup, can't get it to work, don't want to update unless I can backup and restore?

I stopped z2mqtt (else I get a only one connection allowed error), toggled update.
Not sure what the correct port is for backup and restore, but both fail, 4567 or 6638.

```
pieter@DESKTOP-2IL0UKA:/mnt/c/Users/piete$ python3 -m zigpy_znp.tools.network_backup socket://192.168.1.208:4567 -o zigbee_backup.json
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/urlhandler/protocol_socket.py", line 61, in open
    self._socket = socket.create_connection(self.from_url(self.portstr), timeout=POLL_TIMEOUT)
  File "/usr/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pieter/.local/lib/python3.8/site-packages/zigpy_znp/tools/network_backup.py", line 128, in <module>
    asyncio.run(main(sys.argv[1:]))  # pragma: no cover
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/pieter/.local/lib/python3.8/site-packages/zigpy_znp/tools/network_backup.py", line 119, in main
    await znp.connect()
  File "/home/pieter/.local/lib/python3.8/site-packages/zigpy_znp/api.py", line 545, in connect
    self._uart = await uart.connect(self._config[conf.CONF_DEVICE], self)
  File "/home/pieter/.local/lib/python3.8/site-packages/zigpy_znp/uart.py", line 182, in connect
    _, protocol = await serial_asyncio.create_serial_connection(
  File "/home/pieter/.local/lib/python3.8/site-packages/serial_asyncio/__init__.py", line 448, in create_serial_connection
    serial_instance = serial.serial_for_url(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/usr/lib/python3/dist-packages/serial/urlhandler/protocol_socket.py", line 64, in open
    raise SerialException("Could not open port {}: {}".format(self.portstr, msg))
serial.serialutil.SerialException: Could not open port socket://192.168.1.208:4567: [Errno 111] Connection refused
```

```
pieter@DESKTOP-2IL0UKA:/mnt/c/Users/piete$ python3 -m zigpy_znp.tools.network_backup socket://192.168.1.208:6638 -o zigbee_backup.json
Traceback (most recent call last):
  File "/home/pieter/.local/lib/python3.8/site-packages/zigpy_znp/api.py", line 509, in _skip_bootloader
    result = await responses.get()
  File "/usr/lib/python3.8/asyncio/queues.py", line 163, in get
    await getter
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pieter/.local/lib/python3.8/site-packages/zigpy_znp/tools/network_backup.py", line 128, in <module>
    asyncio.run(main(sys.argv[1:]))  # pragma: no cover
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/pieter/.local/lib/python3.8/site-packages/zigpy_znp/tools/network_backup.py", line 119, in main
    await znp.connect()
  File "/home/pieter/.local/lib/python3.8/site-packages/zigpy_znp/api.py", line 551, in connect
    self.capabilities = (await self._skip_bootloader()).Capabilities
  File "/home/pieter/.local/lib/python3.8/site-packages/zigpy_znp/api.py", line 509, in _skip_bootloader
    result = await responses.get()
  File "/home/pieter/.local/lib/python3.8/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/home/pieter/.local/lib/python3.8/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
```

Any pointers to a step by step for backup, update, restore, would be appreciated.

tube0013, 2022-01-15:
If using z2m it does a backup automatically.

https://community.home-assistant.io/t/tubes-zb-coordinators-and-routers-was-zigbee-router-on-steroids/280896/583

Port for communication to the zigbee module is always 6638.

ptr727, 2022-01-15:
Ok, device updated.

I'll update esphome when you publish official firmware.

May I suggest you update the firmware update instructions to mention Z2M no backup needed, and even using port 6638 I could not get `zigpy_znp.tools.network_backup` to work.

tube0013, 2022-01-15:
are you using windows? I've seen others have issues with it, so I'm wondering if that is related.

closed 2024-04-26 by ptr727