tube0013/tube_gateways #51: Zigbee2MQTT reports router is not supported

ptr727, 2022-01-15:
I got a my powered router, thank you.

It appears to work, but in z2mqtt the device is listed as unknown.
`Device '0x00124b00258f1717' with Zigbee model 'tubeszb.router' and manufacturer name 'TubesZB' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html`

This is running z2mqtt from docker using latest tag, version `1.22.2 commit: 414c51f`.

Could you possibly add a PR to z2mqtt to add your hardware config?

tube0013, 2022-01-15:
PR has been made and merged.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/3663

I'll post a external converter that you can use in the mean time next time I'm near my computer.

tube0013, 2022-01-15:
external converter here:

https://github.com/tube0013/tube_gateways/tree/main/tube_cc_router/z2m_external_converter

closed 2022-01-15 by tube0013