tube0013/tube_gateways #52: Q: what do I need to fill in for Devices: on zigbee2mqtt docker-compose.yml

ranrinc, 2022-01-16:
What should I input as devices on this scenario?

`tcp://IPADDRESS:6638:tcp://IPADDRESS:6638`

```
version: '3.8'
services:
  mqtt:
    image: eclipse-mosquitto:2.0
    restart: unless-stopped
    volumes:
      - "./mosquitto-data:/mosquitto"
    ports:
      - "1883:1883"
      - "9001:9001"
    command: "mosquitto -c /mosquitto-no-auth.conf"

  zigbee2mqtt:
    container_name: zigbee2mqtt
    restart: unless-stopped
    image: koenkk/zigbee2mqtt
    volumes:
      - ./zigbee2mqtt-data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      - 8080:8080
    environment:
      - TZ=Europe/Berlin
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0
```

tube0013, 2022-01-16:
you don't need to specify a device in docker compose as it is not physically connected (usb) to the docker host. you configure the tcp://IP:PORT in the z2m config

ranrinc, 2022-01-17:
@tube0013 Is there away for me to contact you directly? Currently after testing and running zigbe2mqtt which is fail misrably since I have no clue what I'm doing and know trying to get back to my ZHA which now unable to connect even though its being discover. Just failed to connect. Help

tube0013, 2022-01-17:
Here or I'm on discord. 

Others can likely help as well if you post in a place with more traffic like discord or community forums.

ranrinc, 2022-01-18:
> Here or I'm on discord.
> 
> Others can likely help as well if you post in a place with more traffic like discord or community forums.

which discord channel? thanks..

tube0013, 2022-01-18:
Homeassistant #zigbee or pm me there... I'm on eastern US time though so it's late now and I won't be on again until morning