tube0013/tube_gateways #53: Q: Add a relay module to CC2652P2 based Zigbee to Ethernet/USB Serial Coordinator

vampywiz17, 2022-01-18:
Hello there,

i would like to add a one channel relay module to your coordonator. Like this 

(https://www.banggood.com/2Pcs-5V-1-Channel-Level-Trigger-Optocoupler-Relay-Module-p-1366337.html?imageAb=1&p=MA240439985285201910&akmClientCountry=HU&cur_warehouse=CN)

Possible to use the ethernet connection pins to connect it?

tube0013, 2022-01-18:
Should work. The used pins are defined in the esphome yanks posted here.

Just be aware the custom serial server component has issues on esphome after 2021.09.x. It will not be able to flash the cc2652 module over the network connection.

vampywiz17, 2022-01-18:
@tube0013 

My plan that i install Tasmota and use that control zigbee TRV-s. Create a rule, that any of TRV is open state >=10% the relay is On. If all of lover than 10%, relay Off.

I thinking a bit,