tube0013/tube_gateways #54: Updated ESPHome on my tube-zb-gw-cc2652p2-v2 and now it doesn't work :)

davidcoulson, 2022-01-18:
So I think I messed up my coordinator, but not really sure how. It is the tube-zb-gw-cc2652p2-v2 model :)

I updated ESPHome to the tube-zb-gw-cc2652p2-v2_2022_01_15.bin that was posted on GitHub, that worked fine, but then I could not connect to the serial port on 6638. I rolled it back to tube-zb-gw-cc2652p2-v2_2021_09_17.bin which was what I wasn on previously, but it still did not work. I attempted to flash new zstack code onto it, but that just fails - I was able to update it to this version of zstack a few days ago without problems.
```

./cc2538-bsl.py -p socket://10.2.3.70:6638  -evw ../CC1352P2_CC2652P_launchpad_coordinator_20220103.hex
Opening port socket://10.2.3.70:6638, baud 500000
Reading data from ../CC1352P2_CC2652P_launchpad_coordinator_20220103.hex
Firmware file: Intel Hex
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

And yes - lesson learned. If it ain't broke... :)

tube0013, 2022-01-18:
definitely stick to tube-zb-gw-cc2652p2-v2_2021_09_17.bin

It may be worth flashing the esp32 from serial so it does a full erase first.
you can do this over USB on the V2 you just need to move the jumpers into the flashing mode per https://github.com/tube0013/tube_gateways/tree/main/V2_tube_zb_gw_cc2652p2 Hold the boot button when plugging in to get it into bootloader mode.

the esphome flasher works well for me.

while you have it near a computer switch the jumpers to the middle position so the serial is going to the cc2652 module and try the flash the cc2652 over serial (hold the BSL button down when plugging in)  This will take a few seconds vs minute when doing it over the network.

davidcoulson, 2022-01-18:
Thanks - Using USB I was able to flash both ESPHome and the cc2652 module and it's all back up and running. Thanks for the quick feedback!

closed 2022-01-19 by davidcoulson

hvddrift, 2022-02-09:
Well, I made exactly the same mistake. However, I cannot even get mine into bootloader mode. 

Any tips on how you did it?