tube0013/tube_gateways #55: Updated ESPHome on my tube-zb-gw-cc2652p2-v2 and now it doesn't work :)

hvddrift, 2022-02-09:
I did exactly the same as:

> So I think I messed up my coordinator, but not really sure how. It is the tube-zb-gw-cc2652p2-v2 model :)
> 
> I updated ESPHome to the tube-zb-gw-cc2652p2-v2_2022_01_15.bin that was posted on GitHub, that worked fine, but then I could not connect to the serial port on 6638. I rolled it back to tube-zb-gw-cc2652p2-v2_2021_09_17.bin which was what I wasn on previously, but it still did not work. I attempted to flash new zstack code onto it, but that just fails - I was able to update it to this version of zstack a few days ago without problems.

But I am unable to get the device into boot mode.

I have the jumpers in this postion

[image]

I hold down boot, plug in the device, but my windows PC does not see it.

Any suggestions?

tube0013, 2022-02-09:
Windows needs the ch340 serial driver. Does device show in device manager?

hvddrift, 2022-02-09:
No, but I did add the driver

hvddrift, 2022-02-09:
Just want to make sure I am doing it correctly. 

Photo of jumpers and boot button I am pressing.

I hold the boot down, plug into the USB and release. I get no sound of a USB device being plugged in like other devices.

[image]

tube0013, 2022-02-09:
that's all good, but you don't have the ESP32 module on the board so there is nothing to put into boot mode or talk to over serial.

hvddrift, 2022-02-09:
Ok. Old age mistake there. I could not read the PCB so I was took off what I thought was just the ethernet connector,

Let's see if I can get it to work.

P.S I feel both embarrassed and stuip

tube0013, 2022-02-09:
no worries. and not discounting something could be wrong.  

Also I saw yo post on twitter, and there is no compatible PoE module with this device. The PoE model I sell is has a different esp32 which has a different pin out, and is much larger. If you want poe I'd advise getting a PoE > micro usb + ethernet splitter.

hvddrift, 2022-02-09:
Ok... Still no luck. I have an adapter
[image]

I could try to connect with tx,rx, gnd and 3.3

tube0013, 2022-02-09:
yes, you can give that a try. is the board getting power over usb? (lights on the esp32 light up at all?)

hvddrift, 2022-02-09:
Just plugged into a USB power board to check. No lights there either.

tube0013, 2022-02-09:
nothing n the esp32? there will be no lit leds on the tubeszb pcb, but should see some on the esp32 when powered up.

It is possible the usb connector has come loose. and this could be causing the issues - no power or serial.

can you try to power the esp32 module separately with the adaptor you showed? 

[image]

hvddrift, 2022-02-09:
[image]

Ok, we can power the esp32 module

hvddrift, 2022-02-09:
And a close up of the USB connector
[image]

tube0013, 2022-02-09:
okay thanks. 

that's a better picture than I could have gotten ;) 

do you have a multimeter by chance?

hvddrift, 2022-02-09:
yes

tube0013, 2022-02-09:
could try and update the FW on the router, new build I put out ups the tx strengh a little bit.

hvddrift, 2022-02-09:
And that's using this method correct?

https://github.com/tube0013/tube_gateways#zigbee-module-firmware-updates

hvddrift, 2022-02-09:
And that's using this method, right?

https://github.com/tube0013/tube_gateways#zigbee-module-firmware-updates

But with a serial adapter. Is there a page with instructions for this?

tube0013, 2022-02-09:
I actually just add the router specific flashing guide here:

https://github.com/tube0013/tube_gateways/tree/main/tube_cc_router#flashing-the-router

hvddrift, 2022-02-09:
Where you able to test if the 3.3 v is ok for the short term?

tube0013, 2022-02-09:
no still in my office. can check in a few hours.. I have to build some devices as well before I can confirm as I don't have any on hand.

tube0013, 2022-02-10:
confirmed 3.3v from a usb serial adapter powers the board up including the esp32 when using the 3.3v and GND pins on the PCB.

hvddrift, 2022-02-10:
Ok, will test now. 

However, I wanted to update the routers first (as I wont have my serial adapter) and I have tested with both a Linux and Windows machine and I get this error on the flash attempt.

```
Opening port COM3, baud 500000
Reading data from tubeszb_router_20220111.hex
Your firmware looks like an Intel Hex file
ERROR: Hex file contains invalid record at line 7
```

tube0013, 2022-02-10:
my bad, I should have zipped the .hex file. I just replaced it with a zipped version,  pull this down and unzip and use it to flash I think that will solve this.

https://github.com/tube0013/tube_gateways/raw/main/tube_cc_router/firmware/tubeszb_router_20220111.hex.zip

hvddrift, 2022-02-10:
Getting closer.... 

```
Opening port COM3, baud 500000
Reading data from tubeszb_router_20220111.hex.zip
Cannot auto-detect firmware filetype: Assuming .bin
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:23:7D:FC:D5
    Performing mass erase
Erasing all main bank flash sectors
    Erase done
Writing 193671 bytes starting at address 0x00000000
ERROR: Invalid data size: 193671. Size must be a multiple of 4.
```

hvddrift, 2022-02-10:
Nope, had to unzip.... Flashing now

tube0013, 2022-02-10:
looks good. you'll need this essentially factory resets them as well, so you'll need to rejoin them once coordinator is back up.

hvddrift, 2022-02-10:
And I think I will need to solder a header on the board first. This is where I connect the 3.3 and GND to correct?

[image]

tube0013, 2022-02-10:
correct.

hvddrift, 2022-02-10:
OK... All seems to be up. I was not able to keep my devices, But I thought I might as well start of right. 

From a tweet I posted. Would you change the channel?
[image]
[image]

hvddrift, 2022-02-10:
And the first device I added wad the freshly flashed router. Its 4 feet way.  But RSSI is Unknown.

Just not sure if this is a symptom of bigger issues. 

[image]

hvddrift, 2022-02-13:
Made a big change and switched to zigbee2mqtt.

At this point I could not be happier. I find it so much better and more robust that ZHA.

Just seeing if or how we can rectify the usb tracing issue. Can I get a replacement board?

tube0013, 2022-02-13:
can you send me an email at my store email - store @ tubeszb.com and we can figure something out.

thansk!

closed 2022-02-14 by hvddrift