tube0013/tube_gateways #56: PoE Coordinator unresponsive

scruloose, 2022-02-10:
Hi there! I bought one of your PoE Zigbee Coordinators recently (Order No. 729000642), and I'm (somewhat belatedly) trying to get it up and running.

I'm a fairly competent Linux admin, but *completely* new to Zigbee and home automation, so it's entirely possible I'm missing something really basic here, but I can't seem to get past step 0 with this device. I don't have mDNS, but I have dnsmasq all set up on my router and I'm very familiar with setting up static DHCP leases. I was expecting the coordinator to request a DHCP lease from the router as soon as it was plugged in to PoE, so I could grab the MAC address from the router's logs and give it a static entry in my dhcp-hosts file just like any other client on the LAN.

What's actually happening is nothing. I've tried connecting it directly to an Ethernet port on my Edgerouter and turning on PoE to that port, and I see the PoE indicator light come on, but the link indicator just stays dark. I've also tried connecting it to an Ethernet switch elsewhere in the house via a PoE injector, with similar results: the PoE injector's power LED comes on, but the link indicator LED on the switch stays dark. In the name of thoroughness, I've tried three different Ethernet cables. Looking at the logs for dnsmasq, no activity at all shows up when I plug in the co-ordinator. I've tried leaving it plugged in for ~30 minutes and nothing changes.

Is there something I'm missing, or some obvious troubleshooting I should try? For reference, I have near zero expertise on the hardware side of things.

scruloose, 2022-02-10:
Confirmed. Popping off the cover while it's connected to PoE, there are no lights lit up.

scruloose, 2022-02-10:
Ah, okay. I see I'm going to have to dismantle the antenna connector and take the board out of the case to access the microUSB port. I'll try that next and let you know what happens, but it won't be today.

tube0013, 2022-02-10:
Okay. Just so you know where this is going. I am assuming it will work over usb.

We can try re-flashing the ESP32 over usb and see if that sorts it out. If not, I'll do and exchange. I've had a few PoE modules exhibit this, but it has been usually straightaway, and they haven't made it out the door as I test all over PoE before shipping. 

thanks for your patience.

scruloose, 2022-02-11:
As you expected, when connected over USB, the lights come on and /dev/ttyUSB0 appears on the host machine. When connected to both USB and  (regular, non-PoE) Ethernet, it grabs a DHCP lease and I can reach the ESPHome web interface.

Now should I follow the *Zigbee Module Firmware Updates* instructions from [here](https://github.com/tube0013/tube_gateways)? Or do you have a different process in mind?

tube0013, 2022-02-11:
Pretty much

Esphome firmware for the PoE unit is here: https://github.com/tube0013/tube_gateways/tree/main/tube_poe_coordinator (bin file)

Recommend the standalone esphome flasher if you don't have something else you are use.

This shouldn't take long and if it doesn't work will get the exchange going.

tube0013, 2022-02-11:
Edited the above a few times 🤦‍♂️

scruloose, 2022-02-11:
Haha, okay.  Just to be sure I'm not about to do something stupid, let me confirm:
* When you say "the standalone esphome flasher", you mean the web interface on the device at `http://tube-zb-gw-cc2652p2-poe/`, yes?
* And if so, do I need to do the two-step process of "prep for firmware update" and then run `cc2538-bsl programmer`? Or can I just click the OTA upgrade button and browse to the bin file I got from the link you just posted?

I definitely don't have a firmware flashing tool I'm used to using, no. ;-)

tube0013, 2022-02-11:
Sorry for the confusion.

I'm mean this one:

https://github.com/esphome/esphome-flasher

It will do a full erase vs the one in the web hi on the device.

The toggles on the web ui relate to doing Zigbee module fw updates.

The coordinators are made of 2 components both with their own firmware so it can be confusing.

Here we are dealing with the esp32 which is using esphome.

scruloose, 2022-02-11:
That made no difference, I'm afraid.

For whatever reason, it took several tries to get a successful flash… but it did ultimately report success then reboot and display the whole bootup log in the console.

The symptoms remain exactly the same: when connected to PoE, there's no activity, no LED indicators or anything.

tube0013, 2022-02-11:
Okay. Send me an email at store @ tubeszb.com. So I can find your order and I'll get a replacement out today.

Thanks for going through the testing steps.

scruloose, 2022-02-11:
Done!

And no problem. It would have been a win for me too if we'd been able to sort it out without having to wait for a replacement to get here. It was worth a try!

scruloose, 2022-03-01:
Update: I received the replacement coordinator and plugged it into the PoE injector, and got no response (no DHCP request, no LEDs light up; nothing).

I haven't yet popped this one out of its case to try powering it from microUSB, but I thought I should let you know the preliminary result anyway.

Now I'm wondering whether there are incompatible PoE voltages or something. My only PoE gear is a Ubiquiti AP connected to a Ubiquiti router, so it was just plug and play, but... what are the odds that two PoE modules in a row spontaneously died in transit after you tested them, right? It strikes me as more likely that whatever the PoE module in the coordinator is, it's not playing well with whatever PoE supply Ubiquiti uses. I'll let you know what happens when I try the new coordinator on USB power, and I guess I'll look up the specs on the PoE output for the Ubiquiti stuff in case that's helpful.

tube0013, 2022-03-01:
Wow, yeah, that doesn't seem like a coincidence. 

the Olimex-ESP32-POE:

ESP32-PoE is an IoT WIFI/BLE/Ethernet development board with Power-Over-Ethernet feature. The PoE is handled by TPS2375PW chip (Si3402-B in older revisions) that is IEEE 802.3-compliant, including pre-standard (legacy) PoE support. The PoE powering requires at least 37V DC to operate successfully. The board takes power from the Ethernet cable and can be expanded with sensors and actuators. Perfect solution for Internet-of-Things projects.

Not really sure why an ubiquity poe switch would be any issue. I have one that I can check when next home. - I don't use it generally when testing as it's powering some APs and Cams.

scruloose, 2022-03-01:
From the manual for my Edgerouter:

Feature | Specification
--- | ---
Max. Total PoE Output | 50W @ 24V
PoE Output | Passive 24V (Pins 4, 5+; 7, 8‑)

So I guess that explains it. There's probably nothing wrong with either of the coordinators, though I'll have to get my hands on a proper standards-compliant PoE injector to find out for sure, and you might want to add a note to the product listing. Ubiquiti's pretty popular, and they're not the only ones using [nonstandard implementations](https://en.wikipedia.org/wiki/Power_over_Ethernet#Non-standard_implementations) of PoE, as it turns out.

tube0013, 2022-03-01:
interesting.  I have an injector that came with one of the Unifi AP's that I've used before. also have a a pair these TP-Link switches I use for testing them.

https://www.amazon.com/gp/product/B076PRM2C5

scruloose, 2022-03-06:
Yeah, it looks like Ubiquiti has used "Passive 24V" in some of their products and proper 802.3 PoE in others. I've been mostly very happy with Ubiquiti, and I'd like to think all their new PoE gear will be 802.3, but who knows.

With all the supply issues going on at the moment, it was ridiculously hard to find *any* consumer-priced switch with 802.3 PoE and tagged VLAN capability available in Canada, but I finally managed to order this [Netgear switch](https://www.netgear.com/ca-en/business/wired/switches/plus/gs108pe/). In theory, that'll be here in a few days, and then I'll be able to verify the coordinators and send one back to you.

tube0013, 2022-03-06:
Good to know for sure, I updated the little paper I send with the units to note the requirement for at least 37V DC. Will update the product listing as well shortly.

let me know how the new switch works out and we can sort out hopefully the return of the second one.

scruloose, 2022-03-16:
Sure enough, once I got that switch set up and tried the coordinators on its PoE ports, I was able to verify that both of them power up and grab a DHCP lease within a few seconds, as expected.

I haven't yet set up an actual Zigbee network — there's a bit of a learning curve to get everything set up before I reach that point. The coordinators seem to be fine, though, so I'll try to get the second one mailed in the next day or two.

tube0013, 2022-03-16:
Awesome, good to hear!

closed 2022-06-07 by tube0013