tube0013/tube_gateways #58: Need some documentation for migrating from ZHA to Zigbee2Mqtt

ranrinc, 2022-02-26:
Is there a documentation on how I can migrate from ZHA to Z2M? Its seems that many of my Zigbee devices did not compatible with ZHA which causing problems. Many suggest to move to Z2M. So planning to do that.

Currently the plan is to add z2m on supervised HA version. 

TIA...

tube0013, 2022-02-26:
A bit out sore of the scope here.

You could maybe look at the debug startup logs from zha or create a network backup and then replicate those settings in z2m otherwise wipe the coordinator and start up z2m fresh.

ranrinc, 2022-02-26:
the plan is to reset everything, however not sure on how to reset the coordinator. Just to be sure it will start fresh for all the device. I have only 10 device, so it wont be to painful. 

I guess removing it from ZHA will reset it?