tube0013/tube_gateways #59: Connectivity issue with CC2652P2 Ethernet/UDB Coordinator after some minutes

didles, 2022-03-09:
Hello,

   I ordered a CC2652P2 based Zigbee to Ethernet/USB Serial Coordinator beginning of Feb 2022. 
   It was working fine until last week when I lost ethernet connection regularly. Need a power on/off to have it up again during a few hours or minutes.
  Then I removed the WT32_ETH01 board. Without this board, the zigbee part is working fine connected via USB (several days without issues).

   If I power the WT32_ETH01 separately and connect to ethernet, then pinging stops after some minutes.

   So seems an issue with either the WT32_ETH01 board or with the ESP software.
   
   Any idea to investigate more this issue ?

   Regards
  
   Didier

tube0013, 2022-03-09:
try a serial reflash of the esp32. have seen them get into a funky state and and a reflash helps. 

stock esphome fw image is here: https://github.com/tube0013/tube_gateways/tree/main/V2_tube_zb_gw_cc2652p2/esphome

Flashing mode:

https://github.com/tube0013/tube_gateways/tree/main/V2_tube_zb_gw_cc2652p2#esp32-flashing-mode---esp32-serial-connection-via-usb

Hold the BOOT button while plugging into USB

troubleshooting doc has a flashing section and links to the esphome flasher tool:
https://github.com/tube0013/tube_gateways/blob/main/Troubleshooting.md

didles, 2022-03-09:
Thanks, I will try, I need first to buy the USB to serial adaptor.

Note that the green led on ethernet jack continues to blink when I have the issue, so the software is not completely dead.

tube0013, 2022-03-09:
Nope not with the model that works via usb or network like you described

you can just change the jumpers as shown so onboard serial talks to the esp32.

https://github.com/tube0013/tube_gateways/tree/main/V2_tube_zb_gw_cc2652p2#esp32-flashing-mode---esp32-serial-connection-via-usb

you do have to trigger the boot mode by holding the boot button on the main PCB when plugging in the USB cable.

didles, 2022-03-09:
Same issue after reflashing.
It is working for some minutes ( ping is ok), then it stops and I get in debug console:

[W][ethernet:115]: Connection via Ethernet lost! Re-connecting...
[W][ethernet:106]: Connecting via ethernet failed! Re-connecting...

tube0013, 2022-03-09:
can you send me a note at store@tubeszb.com so I can look up your order. I can send you another WT32-ETH01 module as it seems this one has issues. Or you can look to pick one direct from Aliexpress.

logichousepcb, 2022-03-09:
> can you send me a note at ***@***.*** so I can look up your order. I can send you another WT32-ETH01 module as it seems this one has issues. Or you can look to pick one direct from Aliexpress.

I have used about 50 of those ETH01 boards. I can share they definitely need a bit of power for Ethernet. I have experienced connection loss and brownouts when not using at least 2amp of power. 

Sent from Mike's iPhone

didles, 2022-03-10:
> I have used about 50 of those ETH01 boards. I can share they definitely need a bit of power for Ethernet. I have experienced connection loss and brownouts when not using at least 2amp of power. Sent from Mike's iPhone

Hello Michael, I connected the ETH01 to a Lab PSU ( can give up to 3A ). I get 120 mA max at 5V, and still have same issue after a while. So seems not to be a power supply issue.