tube0013/tube_gateways #62: Trying to update firmware OTA but getting Update Failed: Wrong Magic Byte

Ming-A, 2022-04-08:
I have the Tubes ZB CC2652P2 ethernet cordinator and im trying to update it OTA but im getting 'Update Failed: Wrong Magic Byte'

[image]

tube0013, 2022-04-08:
Hello, the ota in the web front end is for the esp32/esphome.

To update the zigbee module please follow the guide in the repo's main readme.

Ming-A, 2022-04-08:
> Hello, the ota in the web front end is for the esp32/esphome.
> 
> To update the zigbee module please follow the guide in the repo's main readme.

currently, I'm reading how to update the firmware and seems like i should backup first since it would erase the previous files. im quite lost on where to start or proceed with backing up, do i run the python commands in HA or on windows cmd?

(https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md#backup-and-restore)

tube0013, 2022-04-08:
do you use z2m or ZHA? if z2m check your config directory for a current backup.

Ming-A, 2022-04-08:
> do you use z2m or ZHA? if z2m check your config directory for a current backup.

im currently using ZHA

tube0013, 2022-04-08:
Okay, then follow the instructions at the top of the page you linked, it has instructions on how to get the tool up and running on most platforms. https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md#installation

another option - which will do a backup and restore of the full nvram is the zigstar multitool: https://zig-star.com/radio-docs/zigstar-multi-tool/

Ming-A, 2022-04-08:
> Okay, then follow the instructions at the top of the page you linked, it has instructions on how to get the tool up and running on most platforms. [...]

while following the steps i got this error, i have also toggled 'Prep the cc2652p2 for firmware update' previously but i also toggled reboot since it was not connecting to zha on ha

[image]

Ming-A, 2022-04-08:
i turned on zbsl since i think that is what prevented it to connect and now have this 

`bash-5.1# python -m zigpy_znp.tools.network_backup socket://10.69.40.5:6638
2022-04-09 00:48:43.714 homeassistant zigpy_znp.znp.security WARNING Skipping hashed link key 8d:bb:56:78:a5:55:e8:4f:9b:c4:4f:5f:f0:91:74:b8 (tx: 165, rx: 0) for unknown device a4:c1:38:d7:28:c4:22:f6
2022-04-09 00:48:43.747 homeassistant zigpy_znp.znp.security WARNING Skipping hashed link key b5:ee:80:37:05:55:e8:4f:a3:91:99:10:50:91:74:b8 (tx: 165, rx: 0) for unknown device a4:c1:38:77:67:12:77:ce
{
    "metadata": {
        "version": 1,
        "format": "zigpy/open-coordinator-backup",
        "source": "zigpy-znp@0.7.0",
        "internal": {
            "creation_time": "2022-04-09T00:48:45+08:00",
            "zstack": {
                "version": 3.3
            }
        }
    },
    "coordinator_ieee": "00124b00237dfc74",
    "pan_id": "f5a9",
    "extended_pan_id": "b701dac7806e1d04",
    "nwk_update_id": 0,
    "security_level": 5,
    "channel": 15,
    "channel_mask": [
        15
    ],
    "network_key": {
        "key": "b31a6e82ff867c1370200bb44dd1a51c",
        "sequence_number": 0,
        "frame_counter": 1500677
    },
    "devices": [
        {
            "ieee_address": "04cf8cdf3c8dd0f1",
            "nwk_address": "a767",
            "is_child": false,
            "link_key": {
                "key": "6842dd4eb2a52469175bfa1b7639d37f",
                "tx_counter": 11,
                "rx_counter": 0
            }
        },
        {
            "ieee_address": "14b457fffedf19a1",
            "nwk_address": "57eb",
            "is_child": false
        },
        {
            "ieee_address": "27fffea0dcc41946",
            "nwk_address": "8471",
            "is_child": true
        },
        {
            "ieee_address": "2c1165fffe969ff1",
            "nwk_address": "3388",
            "is_child": true
        },
        {
            "ieee_address": "44100008c31744ff",
            "nwk_address": "54ef",
            "is_child": true
        },
        {
            "ieee_address": "54ef44100008c317",
            "nwk_address": "44ff",
            "is_child": false,
            "link_key": {
                "key": "47b16529fb2909df60e4a1b50c3f76c6",
                "tx_counter": 77,
                "rx_counter": 0
            }
        },
        {
            "ieee_address": "54ef44100008cf05",
            "nwk_address": "996a",
            "is_child": false,
            "link_key": {
                "key": "8eb82fa9a55894cd979f7a6d39af82b2",
                "tx_counter": 77,
                "rx_counter": 0
            }
        },
        {
            "ieee_address": "57fffedf19a157eb",
            "nwk_address": "14b4",
            "is_child": true
        },
        {
            "ieee_address": "65fffe969ff13388",
            "nwk_address": "2c11",
            "is_child": true
        },
        {
            "ieee_address": "847127fffea0dcc4",
            "nwk_address": "1946",
            "is_child": false
        },
        {
            "ieee_address": "8cdf3c8dd0f119f4",
            "nwk_address": "04cf",
            "is_child": true
        },
        {
            "ieee_address": "8d000155dda5661c",
            "nwk_address": "0015",
            "is_child": true
        },
        {
            "ieee_address": "a4c1389ff208e6b5",
            "nwk_address": "a6aa",
            "is_child": false,
            "link_key": {
                "key": "924fbdeee4a153f4c78b2119f2de4ad3",
                "tx_counter": 165,
                "rx_counter": 0
            }
        },
        {
            "ieee_address": "a4c138fa7442e29f",
            "nwk_address": "f8fe",
            "is_child": false,
            "link_key": {
                "key": "79693553538ddddf0670120697112ac9",
                "tx_counter": 165,
                "rx_counter": 0
            }
        }
    ],
    "stack_specific": {
        "zstack": {
            "tclk_seed": "50726d29eb6de68b7727a9b51c7b9992"
        }
    }
}bash-5.1#`

tube0013, 2022-04-08:
Disable the zha integration.

In web front end hit the toggle for fw update switch.

Then run the zigpy-znp tools backup.

tube0013, 2022-04-08:
Okay that worked, you just need to put a   '''> backup.json''' 

At the end to write it out to a file vs the std out.

tube0013, 2022-04-08:
```python -m zigpy_znp.tools.network_backup socket://10.69.40.5:6638 -o backup.json```

Ming-A, 2022-04-08:
> python -m zigpy_znp.tools.network_backup socket://10.69.40.5:6638 -o backup.json

seems like theres some issue connecting with the cordinator now? It shows that my HA server (10.69.40.4) has connected but disconnected after the error pops up

[image]
[image]

tube0013, 2022-04-08:
Not sure there. Restart it to be safe?

Ming-A, 2022-04-08:
> Not sure there. Restart it to be safe?

ever since i toggled 'Prep the cc2652p2 for firmware update' theres issues sometimes to connect to HA which prevents me to connect, was able to get it by toggling zBSL or zRST earlier but now im back to the same issue

Ming-A, 2022-04-08:
tried toggling both zBSL and zRST and managed to get a different popup, how do i check if the backup is sucessful?

 `bash-5.1# python -m zigpy_znp.tools.network_backup socket://10.69.40.5:6638 -o backup.json
2022-04-09 01:11:30.016 homeassistant zigpy_znp.znp.security WARNING Skipping hashed link key 8d:bb:56:78:a5:55:e8:4f:9b:c4:4f:5f:f0:91:74:b8 (tx: 176, rx: 0) for unknown device a4:c1:38:d7:28:c4:22:f6
2022-04-09 01:11:30.048 homeassistant zigpy_znp.znp.security WARNING Skipping hashed link key b5:ee:80:37:05:55:e8:4f:a3:91:99:10:50:91:74:b8 (tx: 176, rx: 0) for unknown device a4:c1:38:77:67:12:77:ce
bash-5.1#
`

tube0013, 2022-04-08:
The prep button runs a script that turns on bsl, resets the coordinator then waits 10 seconds or so and should disable the bsl. After a fw write the device will reset but if no write it will be in boot loader mode and need a reset or reboot to get out of that mode.

tube0013, 2022-04-08:
> tried toggling both zBSL and zRST and managed to get a different popup, how do i check if the backup is sucessful?
> 
> 

Check the backup.json should contain a bunch of ids and keys.

Ming-A, 2022-04-08:
> > tried toggling both zBSL and zRST and managed to get a different popup, how do i check if the backup is sucessful?
> 
> Check the backup.json should contain a bunch of ids and keys.

i took a look at backup.json and heres what it contains, does it seem right?

`{
    "metadata": {
        "version": 1,
        "format": "zigpy/open-coordinator-backup",
        "source": "zigpy-znp@0.7.0",
        "internal": {
            "creation_time": "2022-04-09T01:15:24+08:00",
            "zstack": {
                "version": 3.3
            }
        }
    },
    "coordinator_ieee": "00124b00237dfc74",
    "pan_id": "f5a9",
    "extended_pan_id": "b701dac7806e1d04",
    "nwk_update_id": 0,
    "security_level": 5,
    "channel": 15,
    "channel_mask": [
        15
    ],
    "network_key": {
        "key": "b31a6e82ff867c1370200bb44dd1a51c",
        "sequence_number": 0,
        "frame_counter": 1500677
    },
    "devices": [
        {
            "ieee_address": "04cf8cdf3c8dd0f1",
            "nwk_address": "a767",
            "is_child": false,
            "link_key": {
                "key": "6842dd4eb2a52469175bfa1b7639d37f",
                "tx_counter": 22,
                "rx_counter": 0
            }
        },
        {
            "ieee_address": "14b457fffedf19a1",
            "nwk_address": "57eb",
            "is_child": false
        },
        {
            "ieee_address": "27fffea0dcc41946",
            "nwk_address": "8471",
            "is_child": true
        },
        {
            "ieee_address": "2c1165fffe969ff1",
            "nwk_address": "3388",
            "is_child": true
        },
        {
            "ieee_address": "44100008c31744ff",
            "nwk_address": "54ef",
            "is_child": true
        },
        {
            "ieee_address": "54ef44100008c317",
            "nwk_address": "44ff",
            "is_child": false,
            "link_key": {
                "key": "47b16529fb2909df60e4a1b50c3f76c6",
                "tx_counter": 88,
                "rx_counter": 0
            }
        },
        {
            "ieee_address": "54ef44100008cf05",
            "nwk_address": "996a",
            "is_child": false,
            "link_key": {
                "key": "8eb82fa9a55894cd979f7a6d39af82b2",
                "tx_counter": 88,
                "rx_counter": 0
            }
        },
        {
            "ieee_address": "57fffedf19a157eb",
            "nwk_address": "14b4",
            "is_child": true
        },
        {
            "ieee_address": "65fffe969ff13388",
            "nwk_address": "2c11",
            "is_child": true
        },
        {
            "ieee_address": "847127fffea0dcc4",
            "nwk_address": "1946",
            "is_child": false
        },
        {
            "ieee_address": "8cdf3c8dd0f119f4",
            "nwk_address": "04cf",
            "is_child": true
        },
        {
            "ieee_address": "8d000155dda5661c",
            "nwk_address": "0015",
            "is_child": true
        },
        {
            "ieee_address": "a4c1389ff208e6b5",
            "nwk_address": "a6aa",
            "is_child": false,
            "link_key": {
                "key": "924fbdeee4a153f4c78b2119f2de4ad3",
                "tx_counter": 176,
                "rx_counter": 0
            }
        },
        {
            "ieee_address": "a4c138fa7442e29f",
            "nwk_address": "f8fe",
            "is_child": false,
            "link_key": {
                "key": "79693553538ddddf0670120697112ac9",
                "tx_counter": 176,
                "rx_counter": 0
            }
        }
    ],
    "stack_specific": {
        "zstack": {
            "tclk_seed": "50726d29eb6de68b7727a9b51c7b9992"
        }
    }
}`

Ming-A, 2022-04-08:
successfully updated the firmware and currently trying the restore command with same errors as before. The difference now is i get `[D][api.connection:604]: Client 'Home Assistant 2022.4.1 (10.69.40.4)' connected successfully!` as the first line after a reboot without it saying it got disconnected (i still have zha disabled). 

Whenever i ran the command before on ssh i would see it connected and disconnected in the debug, now im not getting any msg at all.

tube0013, 2022-04-08:
Can you post the backup.json? Here?

Ming-A, 2022-04-08:
> Can you post the backup.json? Here?

https://pastebin.com/hMbnzAKv

tube0013, 2022-04-08:
Can you be certain the backup.json has the json you posted in it and that is the same file you are attempting to restore?

The error is showing the file is empty when trying to restore.

(I went to the source and had puddly who wrote these tools / library take a look at this to confirm).

Ming-A, 2022-04-08:
> Can you be certain the backup.json has the json you posted in it and that is the same file you are attempting to restore?
> 
> The error is showing the file is empty when trying to restore.
> 
> (I went to the source and had puddly who wrote these tools / library take a look at this to confirm).

i found the backup.json is in the config folder after the backup so it should be the same but im not too sure 

[image]

Ming-A, 2022-04-08:
> > python -m zigpy_znp.tools.network_backup socket://10.69.40.5:6638 -o backup.json
> 
> seems like theres some issue connecting with the cordinator now? It shows that my HA server (10.69.40.4) has connected but disconnected after the error pops up.
> 
> I have toggled restart a few times too
> 
> [image]
> 
> [image]

the error I think is similar to the backup error previously until I somehow got it to work by toggling zBSL and zRS, but sadly it dosent seem to work now

tube0013, 2022-04-08:
the backup.json is 0k

open it and paste in what you put on pastebin. resave and try again.

Ming-A, 2022-04-08:
> the backup.json is 0k - it's empty.
> 
> open it and paste in what you put on pastebin. resave and try again.

ohh so that's what's been causing the issue, I wondered how it ended up becoming blank. I think it should be alright now, thank you so much for helping out!

ill do some test and see if my devices are functioning after a restart of HA

[image]

closed 2022-04-08 by Ming-A