tube0013/tube_gateways #66: Coordinator unresponsive

kinselp, 2022-05-31:
Hi, have had my Poe coordinator running for months and in an attempt to fix an issue with Z2M I tried to flash it thru the webserver OTA update using the bin file. 

Guessing I haven't showered myself in glory, as even though the update said successful it now doesn't seem to get an IP (have red and orange leds on, but no green). 

What's the best way to check what's going on?  I'm guessing I've either flashed the wrong image or its failed mid flash - hoping a USB flash will get me back on track. 

Appreciate any assistance.

tube0013, 2022-05-31:
DO you have the round PoE version? if so you'll want this bin file:

https://github.com/tube0013/tube_gateways/raw/main/tube_poe_coordinator/tube_zb_gw_cc2652p2_oliemex_poe.bin

I suggest flashing over USB, so remove PoE power, pop the coordinator lid, and you will likely need to undo the antenna retaining nut to full remove the PCB from the case. once removed you can use the micro USB connection to flash the bin. 

ESPhome Flasher works well.

kinselp, 2022-06-04:
Coordinator back up - thanks @tube0013.  now to get Z2M working again!

closed 2022-06-04 by kinselp

tube0013, 2022-06-04:
Good to hear 👍