tube0013/tube_gateways #68: Setting up zha integration in home assistant fails

drewthor, 2022-06-07:
I recently received the CC2652P2 based Zigbee to Ethernet/USB Serial Coordinator.

Whenever the zha integration in homeassistant tries to load (after the integration successfully connects which is a bit sporadic), I get the following error:

```
Error setting up entry socket://192.168.0.172:6638 for zha
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 888, in request_callback_rsp
    return await callback_rsp
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 282, in write_network_info
    commissioning_rsp = await self.request_callback_rsp(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 888, in request_callback_rsp
    return await callback_rsp
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.9/site-packages/async_timeout/__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 339, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 102, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 173, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 69, in new
    await app.startup(auto_form)
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 152, in startup
    return await self._startup(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 207, in _startup
    await self.form_network()
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 393, in form_network
    await self.write_network_info(network_info=network_info, node_info=node_info)
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/zigbee/application.py", line 1149, in write_network_info
    return await self._znp.write_network_info(
  File "/usr/local/lib/python3.9/site-packages/zigpy_znp/api.py", line 311, in write_network_info
    raise RuntimeError(
RuntimeError: Network formation refused, RF environment is likely too noisy. Temporarily unscrew the antenna or shield the coordinator with metal until a network is formed.

```

I tried looking up this error in the home assistant community forums and elsewhere but the proposed solutions do not work.

Things I have tried:
1) removing the zha integration, deleting zigbee.db, restarting home assistant
2) removing the antenna temporarily and trying to re-setup
3) with the antenna removed covering the controller with cast iron and cast steel pans
4) rebooting the controller

Any advice on how I can get this integration to successfully be set up?

tube0013, 2022-06-08:
Hi,

I'd recommend you use the zigpy-znp tools and run an energy scan.

https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md

From the scan, pick a channel with a low to no bar. ideally it will be 15,20 or 25 ZHA default is 15 though so hopefully 20 or 25 are clearer if not next best is 11 and then go from there to the others.

once that is done if you have already set up the device you'll want to wipe it with a NVRAM reset (also in the tools).

Before adding ZHA again, create an ZHA entry in the configuration.yaml for HA and define the channel you want to use based off the info from the scan (you don't need the mask part):

https://www.home-assistant.io/integrations/zha/#defining-zigbee-channel-to-use

Hopefully that gets you up and running.  The tools link above has great instructions for setting it up in any OS/Environment. It is likely easier to run on another machine vs your HA machine. Just be sure ZHA is not enabled so only one thing is using the coordinator at a time. all the command can be address to the network coordinator by using the socket://IP_ADDRESS:6638 address in place of the /dev/serial/by-id/......  

Hopefully that gets you up and running.

drewthor, 2022-06-08:
@tube0013 thanks for the reply! Trying to work through the tools you linked and steps right now. The forming of the network is the part that keeps on failing. Running a scan on my phone It appears that the 2.4GHz Wi-Fi channels that are the strongest around me (apartment) are channels 5,2,10,4 in that order. My own 2.4 GHz network is run on channel 1 (since I actually know about networks and want to be a good neighbor and cause the least overlap).

I think zigbee channel 25 would be the best place for me to start and see if it works.

tube0013, 2022-06-08:
I'd seriously try running the energy scan from the coordinator where you'd like it to be run from. it will give you a really good picture like this of all the nosie on the different zigbee channels:

[image]

drewthor, 2022-06-08:
When I try and run that I get this error in the output:
```
2022-06-07 23:04:37.341 thorserver __main__ INFO Starting up zigpy-znp
2022-06-07 23:04:37.933 thorserver __main__ ERROR Could not start application: Cannot start application, network is not formed
2022-06-07 23:04:37.933 thorserver __main__ ERROR Form a network with `python -m zigpy_znp.tools.form_network
```

Which of course form_network fails since the RF environment is too noisy.

tube0013, 2022-06-08:
Ah, forgot about that - I run it for ax but have everything in one command line 🤦‍♂️

Things to try.. no antenna and wrap the coord in tin foil (dead serious)

Another option - Pop the case lid off and disconnect the antenna cable from the module - you will need to temporarily remove the esp32 module as it is over top of the zigbee module. 

You could try the channel you saw with your Wi-Fi scan in the zha settings that will create to network on that channel if successful. You can leave it or disable zha and run the scan then to confirm. 

Some of the zigbee modules vary in sensitivity, you might have a sensitive one and also have a good bit of interference in play creating a perfect storm.

tube0013, 2022-06-08:
It's late here so signing off for tonight. Good luck can help some more tomorrow if needed.

tube0013, 2022-06-08:
super odd. can you let me know you order No.? Maybe we can swap it out for a less another unit that maybe is less sensitive.

drewthor, 2022-06-09:
Order no: 1926149398

I tried just now wrapping the whole thing in many feet of aluminum foil and still it says the RF environment is too noisy (the antenna is still disconnected). I can try a long Ethernet cable to the far corner of my apartment as a last test (with it wrapped in aluminum foil and antenna removed)

tube0013, 2022-06-09:
let Let me know how that last one goes, if still an issue I will prepare a replacement for you.

drewthor, 2022-06-14:
PoE replacement came and worked instantly. Was able to run an energy scan as well and get it set up on the lowest energy channel (25). Thanks for the help and awesome product!

closed 2022-06-14 by drewthor