tube0013/tube_gateways #71: Converting Zigbee to IP

lacojim, 2022-06-14:
Not related to anything necessarily here, but maybe something that can be built by Tube? I have a need for connecting Zigbee devices to a remote site through a VPN (Wireguard/OpenVPN), while keeping only one coordinator locally. My thought is a zibee transceiver which converts and carries zigbee signals over TCP/IP, then once it transcends to the other side, it's converted back to zigbee radio waves. Has something like this been done, or can it even be done for that matter?

tube0013, 2022-06-15:
Thread's model with border routers is more logical for this. Zigbee doesn't really have a mechanism to do this... more the tcp back to zigbee. Not saying it's impossible just now the spec was designed.

lacojim, 2022-06-15:
I assumed zigbee would not have the built in ability to do this. It would certainly be something custom to intercept the zigbee signals, Put them into some kind of a wrapper (or compress/wrapper), and send them off to a decoder on the other side. For that matter, it would have to be bi-directional. The s/w and h/w would certainly be custom, but should be identical on each side for simplicity sake. Latency MIGHT be a problem too. Just a thought. Maybe get some wheels turning.