tube0013/tube_gateways #72: ZHA showing 2 coordinators??

kksligh, 2022-06-16:
I recent purchased and installed the CC2652P2 based Zigbee to Ethernet/USB Serial Coordinator to ZHA on my home assistant instance. For some reason, its showing 2 coordinators after the set up. The info on each device are as follows:

1) "Silicon Labs EZSP"
2) "Zigbee Coordinator" - ZNP - Texas Instruments Z-Stack ZNP protocol: CC253x, CC26x2, CC13x2

Is this normal or did I do something wrong upon installation?

tube0013, 2022-06-16:
Hi,

Did you migrate from another coordinator?

kksligh, 2022-06-16:
Yes - previously I had a HUBZ coordinator but removed that entire instance...and started over from scratch. At least I thought...

tube0013, 2022-06-16:
Did you delete the zigbee.db file in the config folder?

If not this shouldn't be a big deal, and is just cosmetic. I think you can use a SQLite editor to remove any reference of it (by its ieee address) in the zigbee.db file and I think that would take care of it. (If you do this be sure to backup the db before the edit just in case anything goes wrong)

kksligh, 2022-06-16:
Dumb question, but such one do I delete?

Thanks,

Kale Sligh

VP, Marketing
o 615.744.1206
m 615.917.1441

kksligh, 2022-06-16:
What's very odd is my zigbee map shows devices connected to both

Thanks,

Kale Sligh

VP, Marketing
o 615.744.1206
m 615.917.1441

tube0013, 2022-06-16:
Look for the ieee address of the old coordinator.

Then go through the tables in the zigbee.db and delete any records that have that address I believe most tables will have ieee as a column. (It's been a while since I looked inside one)

Seeing lines in the map may just be old cached records.

kksligh, 2022-06-16:
I just deleted the file and started fresh.  Cleared it up!

closed 2022-06-16 by kksligh