tube0013/tube_gateways #74: Flashed router but firmware didn't change in Zigbee2MQTT

drjjr2, 2022-07-12:
I am thinking it's something I've done wrong. I have it connected and the cc2538-bsl.py script acts like it completes and verifies, but the version in Zigbee2MQTT still reads "20210128" I have removed and readded the device to the Zigbee network.

I flashed the tubeszb_router_20220223.hex file and also tried CC1352P2_CC2652P_launchpad_router_20220125.hex. Same result.

Any ideas on what to try?

drjjr2, 2022-07-12:
Following up:
Doing a read of the basic cluster shows the date code of the firmware I flashed:
Read result of 'genBasic': {"swBuildId":"20220125"}

tube0013, 2022-07-12:
It's in the sw_build_id attribute

[image]

drjjr2, 2022-07-12:
Well, patience is a virtue. Just needed to wait some amount of time. Just checked and it's 20220125 in the firmware version of the device info.

closed 2022-07-12 by drjjr2

tube0013, 2022-07-12:
I posted screenshot from one of mine not on latest 😂