tube0013/tube_gateways #75: Error starting docker container with tcp coordinator

larrym, 2022-07-21:
I am attempting to run a Smartlight SLZB-05 tcp coordinator with z2m in a docker container but I am getting a herdsman/serial error. I have tried many variations in the docker-compose.yaml file, but nothing seems to work. It constantly restarts.

There is no /dev/ttyACM0 on this machine.

Any advice?

**OS**
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04 LTS
Release:	22.04
Codename:	jammy

**Coordinator**
Zigbee LAN CC2652P Coordinator SLZB-05

**log**
```
Using '/app/data' as data directory
Zigbee2MQTT:info  2022-07-21 07:22:13: Logging to console and directory: '/app/data/log/2022-07-21.07-22-13' filename: log.txt
Zigbee2MQTT:info  2022-07-21 07:22:13: Starting Zigbee2MQTT version 1.26.0 (commit #cd23aea)
Zigbee2MQTT:info  2022-07-21 07:22:13: Starting zigbee-herdsman (0.14.40)
Zigbee2MQTT:error 2022-07-21 07:22:14: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-07-21 07:22:14: Failed to start zigbee
Zigbee2MQTT:error 2022-07-21 07:22:14: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-07-21 07:22:14: Exiting...
Zigbee2MQTT:error 2022-07-21 07:22:14: Error: Error while opening serialport 'Error: Error: Not a tty setting custom baud rate of 115200'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
```

**partial docker-compose.yaml:**
```
  mosquitto:
    image: eclipse-mosquitto
    container_name: mosquitto
    volumes:
      - ./mosquitto/config:/mosquitto/config/
      - ./mosquitto/data:/mosquitto/data/
      - ./mosquitto/log:/mosquitto/log/
    ports:
      - 1883:1883
      - 9001:9001

  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    restart: always
    volumes:
      - /opt/docker/zigbee2mqtt/data:/app/data
    #  - /run/udev:/run/udev:ro
    ports:
      # Frontend port
      - 8091:8080
    environment:
      - TZ=[myTz]
    network_mode: host
    #devices:
      #- /dev/tty9:/dev/ttyACM0
    #  - /dev/ttyS0:/dev/ttyACM0
```

**zigbee2mqtt configuration.yaml:**
```
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://172.17.0.1'
  # MQTT server authentication, uncomment if required:
   user: larry
   password: qiqltoq

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: 'tcp://192.168.10.183:6638'

advanced:
  log_level: debug
```

tube0013, 2022-07-21:
I think that all looks good. No need to pass any devices in the docker-compose for a network based device. maybe try adding baud rate and other settings from:

https://www.zigbee2mqtt.io/guide/configuration/adapter-settings.html

larrym, 2022-07-21:
I changed the z2m config to the following, but the error is the same.

```
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://172.17.0.1'
  # MQTT server authentication, uncomment if required:
   user: larry
   password: qiqltoq

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: tcp://192.168.10.183:6638
  disable_led: false
  # Optional: adapter type, not needed unless you are experiencing problems (default: shown below, options: zstack, deconz, ezsp)
  adapter: null
  # Optional: Baud rate speed for serial port, this can be anything firmware support but default is 115200 for Z-Stack and EZSP, 38400 for Deconz, however note that some EZSP firmware need 57600.
  baudrate: 115200
  # Optional: RTS / CTS Hardware Flow Control for serial port (default: false)
  rtscts: false

advanced:
  log_level: debug
```

tube0013, 2022-07-21:
may reach out on HA or Z2M discord. I'm not really sure why it's throwing this error.

larrym, 2022-07-21:
Do you know the discord channel name?

tube0013, 2022-07-21:
#zigbee in ha discord. z2m should have a link to theirs on their site.

closed 2022-07-21 by tube0013