tube0013/tube_gateways #76: CC2652P flashing fails

nerdinand, 2022-07-25:
Today I tried connecting the shiny new PoE-Coordinator to my Home Assistant instance. It got auto-discovered and I tried scanning for devices but couldn't find any. So I figured, let's do a firmware update (probably a dumb idea)...

I toggled "Prep the cc2652p2 for firmware update" and then ran:
```
$ python cc2538-bsl.py -p socket://tubeszb-cc2652-poe-2022.local:6638  -evw ~/Downloads/CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
```
This seemed to do the right thing, until it didn't. After a while It failed with `ERROR: Timeout waiting for ACK/NACK`.

I have now tried numerous times to flash, everytime it fails with a similar error after a couple of seconds.
I also tried flashing via USB, with similar misfortune:

```
$ python cc2538-bsl.py -p /dev/tty.usbserial-1110 -evw ~/Downloads/CC1352P2_CC2652P_launchpad_coordinator_20220219.hex

Opening port /dev/tty.usbserial-1110, baud 500000
Reading data from /Users/ferdi/Downloads/CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
Your firmware looks like an Intel Hex file
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

Any ideas?

tube0013, 2022-07-25:
The Coordinator was flashed with the latest build of the Z-Stack FW when shipped. 

as far as flashing, sometime you have to try multiple times. another option is to trigger the boot loader manually, to do this, pop the top with the device unpowered, hold down the BSL button while plugging in PoE, it's important to hold it for several seconds as the ESP32 will trigger a reset of the zigbee module after it boots.

To flash over USB you will need a USB Serial converter. the Onboard USB on the PoE module only talks to the ESP32 and is only good for flashing/debugging that.

Typically when devices can't join, it's due to interference. I'd suggest running a energy scan using the [zigpy-znp tools](https://github.com/zigpy/zigpy-znp/blob/dev/TOOLS.md) to see which channel has the least noise. After that you can do a NVRAM reset and then define the desired channel in HA/ZH configuration.yaml (see [zha docs](https://www.home-assistant.io/integrations/zha/#configuration---yaml))  and when zha starts up on a fresh radio it will form a new network on the new specified channel.

nerdinand, 2022-07-26:
Thanks for your input!

I finally managed to get the flashing through by moving my laptop and the coordinator closer together on the network (only a switch apart, no wifi involved).

However, now when I try to convince Home Assistant to connect to it, I get a "Failed to connect" everytime:

[image]

The tube_zb_gw_cc2652p2_poe Web Server is showing this:

```
[D][streamserver:085]: New client connected from 192.168.88.228
[D][streamserver:050]: Client 192.168.88.228 disconnected
```

Maybe related question: Which one of these is the right firmware for the Olimex board? 

tube_zb_gw_cc2652p2_oliemex_poe.bin
tube_zb_gw_cc2652p2_(2021_04_08).bin
tube-zb-gw-cc2652p2-poe_olimex_round_latest.bin

Could it be that i flashed the wrong firmware on the Olimex board?

nerdinand, 2022-07-26:
Ah, I see. I did a successful OTA update with that one just now. However I still get the "Failed to connect" and sometimes "Unknown error" and the autodiscovery of home assistant doesn't discover it either... :(

[image]

I'll update my home assistant to the latest release now, but I don't think that'll help...

tube0013, 2022-07-26:
do you see any errors in the HA log? can you post them?

you can also try to manually set up the integrations and type in socket://ip:6638

nerdinand, 2022-07-26:
Finally got it to work! Not sure, what exactly changed this time around. I could imagine the reboot of Home Assistant because of the update did something. I was able to connect to 2 of my Hue devices with no problems and I'm able to control them! Thanks for your prompt help! Here's to no more problems... 🍻

tube0013, 2022-07-26:
Awesome. glad to hear!

closed 2022-07-26 by tube0013