tube0013/tube_gateways #80: Unable to upgrade cc2652p2 firmware

samip5, 2022-08-11:
I have tried to upgrade firmware of the module, but without any luck, any ideas?

I have the model of the poe gateway that doesn't include a jumper nor a proper USB port that I could use to flash the module.

```
python3 ./cc2538-bsl.py -V -p socket://192.168.2.112:6638 -evw CC1352P2_CC2652P_launchpad_coordinator_20220219.hex
Traceback (most recent call last):
  File "/home/sky/Downloads/cc2538-bsl-2.1/venv/lib/python3.10/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: 'socket://192.168.2.112:6638'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/sky/Downloads/cc2538-bsl-2.1/./cc2538-bsl.py", line 1042, in <module>
    cmd.open(conf['port'], conf['baud'])
  File "/home/sky/Downloads/cc2538-bsl-2.1/./cc2538-bsl.py", line 195, in open
    self.sp = serial.Serial(
  File "/home/sky/Downloads/cc2538-bsl-2.1/venv/lib/python3.10/site-packages/serial/serialutil.py", line 244, in __init__
    self.open()
  File "/home/sky/Downloads/cc2538-bsl-2.1/venv/lib/python3.10/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port socket://192.168.2.112:6638: [Errno 2] No such file or directory: 'socket://192.168.2.112:6638'
ERROR: [Errno 2] could not open port socket://192.168.2.112:6638: [Errno 2] No such file or directory: 'socket://192.168.2.112:6638'
```

tube0013, 2022-08-11:
be sure nothing else is connected to the device on port 6638, that message is saying it can't even connect to it.

samip5, 2022-08-11:
To me that error sounds like it's not even trying to connect and I cannot even see it trying to connect to it in Wireshark.

tube0013, 2022-08-11:
have you been able to connect ZHA or Z2M and set up a network?

samip5, 2022-08-11:
Yes, I have.

samip5, 2022-08-11:
zigbee2mqtt is currently scaled down.

tube0013, 2022-08-11:
check that you have pyserial installed it's a dependecy for cc-2538-bsl and is what creates the serial connection.

samip5, 2022-08-11:
It would yell if I didin't have, thus why I'm confused.

I wonder which version of python is that supposed to work with?

tube0013, 2022-08-11:
pyserial's last release was several years ago, so it should work with any version since then. 

maybe try this - it's a one line docker command:

https://github.com/git-developer/ti-cc-tool

samip5, 2022-08-11:
Huh, okey so that did the trick but it didn't even try to connect on host but via Docker it worked.

closed 2022-08-11 by samip5

tube0013, 2022-08-11:
awesome.