tube0013/tube_gateways #82: Questions

rayzilt, 2022-08-17:
Hi,

Not sure how to contact you and thought a issue should be the way to go.

I recently received the router and have some questions about it.
- There a two led lights, green and red. I guess green is powered up. Does the red one has any function? If not, can this be turned off? It's enlighten my hallway in the night, even in the white printed box.
- I've paired the router in zigbee2mqtt. In z2m I'm using the ping function. Does the router supports this? z2m is unable to ping the router. If the router supports this, then I guess the router is out of range. Running z2m map function shows multiple connections with the router.

Thanks

tube0013, 2022-08-17:
Hi, 

only the green led should light, this is controlled in the router FW as a way to know the router fw has booted and the router is powered on. At current the red led is not configured in the fw and should be off. I'm not sure why it would be lit up to be honest.

As far as the ping function, I'm not familiar with that, and assume it's just an attempt to read a cluster. But again I don't know why it wouldn't work. The fw on the router is virtually the same as the one made by the z2m author (it's based on it).

If you have a USB to serial adapter you may try reflashing the router. <https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-router#flashing-the-router>

rayzilt, 2022-08-17:
Thanks for the quick response!

I've reflashed firmware 20220223 and also tried 20220111. Still the red led is shining bright :-)

I noticed that reflashing to 20220111 and pairing with z2m again still gives firmware 20220223 as version in the About screen. (Force remove done).
Guess some caching on the z2m side? I noticed a different CRC32 calculation, so definitely a different firmware as seen in the screenshot:

[image]

The red led is always on, even before the green led goes on or in flash mode.

[image]

Any more idea's?

tube0013, 2022-08-17:
Thanks for the picutre!

my head is scrambled, you got a version of the board that included an extra power led, I transitioned away from this after not to many units.

so to kill this LED. just cut the trace coming to it from R5.

rayzilt, 2022-08-17:
Ohhh, I got the special edition! :-)

I'll try to cut the trace. Thanks

closed 2022-08-17 by rayzilt

tube0013, 2022-08-17:
> Ohhh, I got the special edition! :-)

you could say that!

rayzilt, 2022-08-18:
To be sure, I've to cut the trace between the red lines?

[image]