tube0013/tube_gateways #85: More than one coordinator???

ripvega, 2022-09-15:
I have both a coordinator and 5 routers, however, the distance is too great for the routers to pair.

Can I use 2 ethernet connected coordinators on one HA instance, or better still, convert an ethernet coordinator to a router?

tube0013, 2022-09-15:
You can run ZHA and z2m concurrently or multiple instances on z2m.

Yes the coordinator can be flashed with router fw.

ripvega, 2022-09-15:
Great, so if I buy 5 more coordinators I can flash them with router firmware and ZHA will lick them up over ethernet?

tube0013, 2022-09-15:
No, zigbee doesn't work like that. There is one coordinator per network.

A zigbee router/repeater is a wireless device in the mesh. 

For example: The PoE coordinator I sell can be flashed with router fw. At that point the PoE is only providing power.

ripvega, 2022-09-15:
Thank you. 

So, my best option then is to have ZHA in one part of the house and Z2MQTT in my best option then is to have space ZHA another. 

Is that correct?

tube0013, 2022-09-15:
2 networks yes.

the coordinators come flashed with z-stack firmware compatible with ZHA and Zigbee2MQTT.  you'd need to set up the zigbee2mqtt instance yourself. if you are running a version of HA that supports Addons you can do this with the addons. there are a number of youtube videos that show how to get this running. the difference with a network coordinator, is for the port you enter tcp://ip_address:6638 instead of the path like /dev/serial/......

ripvega, 2022-09-15:
OK, so just the same as adding ZHA really - no flashing required!

I'll order two more.

Thank you.

ripvega, 2022-09-24:
My coordinators arrived today, thank you.

I'm not having much success with adding Zigbee2MQTT and would appreciate some help.

The config is:
`data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  serial: port:'tcp://10.0.1.2:6638'`

The log shows:
`[22:19:03] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:error 2022-09-24 22:19:05: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-24 22:19:05: Failed to start zigbee
Zigbee2MQTT:error 2022-09-24 22:19:05: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-24 22:19:05: Exiting...
Zigbee2MQTT:error 2022-09-24 22:19:05: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_64241bb1f012ec11bdd823c7bd930c07-if00-port0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
[22:19:06] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:error 2022-09-24 22:19:08: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-24 22:19:08: Failed to start zigbee
Zigbee2MQTT:error 2022-09-24 22:19:08: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-24 22:19:08: Exiting...
Zigbee2MQTT:error 2022-09-24 22:19:08: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_64241bb1f012ec11bdd823c7bd930c07-if00-port0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
[22:19:08] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:error 2022-09-24 22:19:10: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-24 22:19:10: Failed to start zigbee
Zigbee2MQTT:error 2022-09-24 22:19:10: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-24 22:19:10: Exiting...
Zigbee2MQTT:error 2022-09-24 22:19:10: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_64241bb1f012ec11bdd823c7bd930c07-if00-port0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
[22:19:11] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:error 2022-09-24 22:19:13: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-24 22:19:13: Failed to start zigbee
Zigbee2MQTT:error 2022-09-24 22:19:13: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-24 22:19:13: Exiting...
Zigbee2MQTT:error 2022-09-24 22:19:13: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_64241bb1f012ec11bdd823c7bd930c07-if00-port0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
[22:19:13] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:error 2022-09-24 22:19:15: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-24 22:19:15: Failed to start zigbee
Zigbee2MQTT:error 2022-09-24 22:19:15: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-24 22:19:15: Exiting...
Zigbee2MQTT:error 2022-09-24 22:19:15: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_64241bb1f012ec11bdd823c7bd930c07-if00-port0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
[22:19:16] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:error 2022-09-24 22:19:18: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-24 22:19:18: Failed to start zigbee
Zigbee2MQTT:error 2022-09-24 22:19:18: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-24 22:19:18: Exiting...
Zigbee2MQTT:error 2022-09-24 22:19:18: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_64241bb1f012ec11bdd823c [... cut ...]
Zigbee2MQTT:error 2022-09-24 22:19:33: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_64241bb1f012ec11bdd823c7bd930c07-if00-port0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
[22:19:34] INFO: Handing over control to Zigbee2MQTT Core ...`

Add-on CPU load is 42%, pushing my total load to 97%!

Any help is appreciated.

tube0013, 2022-09-24:
Check the zigbee2mqtt config file has the correct device defined? Per the log it's trying to open a serial port for a sonoff. 

Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_64241bb1f012ec11bdd823c7bd930c07-if00-port0' at SerialPort.<anonymous>

ripvega, 2022-09-24:
Thank you. Some progress.

Log is now:
`s6-rc: info: service z2m: starting
s6-rc: info: service socat: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service socat successfully started
s6-rc: info: service z2m successfully started
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[22:46:30] INFO: Socat not enabled
[22:46:30] INFO: Handing over control to Zigbee2MQTT Core ...`

config file is:
`external_converters: []
devices:
  - devices.yaml
groups:
  - groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  user: addons
  password: Evaij7uameex6xaeDei4jeegu4Aighah2keijuut8eich5quooYo7ohmaeSheiz4
  server: mqtt://core-mosquitto:1883
serial:
  port: "tcp://10.0.1.2:6638"
advanced:
  log_level: warn
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental: {}
availability: true
`

CPU load is now 8% but I only see `502: Bad gateway` when trying to access.

Any idea what to do next?

tube0013, 2022-09-24:
Any more in the log? Hit refresh, that's just the addon startup

ripvega, 2022-09-24:
Not, that's it. Refresh just gives the same.

ripvega, 2022-09-24:
I stopped the add-on and then restarted it.

The log is now:

`    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
[22:57:10] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
[22:57:11] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
[22:57:12] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
[22:57:13] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
[22:57:14] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
[22:57:16] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
[22:57:17] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
[22:57:18] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
[22:57:19] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
[22:57:20] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)`

High CPU is back.

tube0013, 2022-09-24:
Try starting from scratch.

Delete z2m addon

Delete z2m config folder

Re add z2m addon

This is my z2m addon config for my test z2m network:

[image]

tube0013, 2022-09-24:
If switching to yaml view (which I may have done when I set it up):

[image]

ripvega, 2022-09-24:
I've done that.
Deleted the z2m folder and uninstalled it.
Restarted HA.
Reinstalled Z2m.

Log is now:
`s6-rc: info: service z2m: starting
s6-rc: info: service socat: starting
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service socat successfully started
s6-rc: info: service z2m successfully started
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
[23:13:42] INFO: Socat not enabled
[23:13:43] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:info  2022-09-24 23:13:45: Logging to console and directory: '/config/zigbee2mqtt/log/2022-09-24.23-13-45' filename: log.txt
Zigbee2MQTT:info  2022-09-24 23:13:45: Starting Zigbee2MQTT version 1.27.2 (commit #unknown)
Zigbee2MQTT:info  2022-09-24 23:13:45: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:info  2022-09-24 23:14:28: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-09-24 23:14:28: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20220219,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2022-09-24 23:14:28: Currently 0 devices are joined:
Zigbee2MQTT:info  2022-09-24 23:14:28: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-09-24 23:14:28: Connecting to MQTT server at mqtt://core-mosquitto
Zigbee2MQTT:error 2022-09-24 23:14:28: MQTT failed to connect: Connection refused: Not authorized
Zigbee2MQTT:error 2022-09-24 23:14:28: Exiting...
Zigbee2MQTT:info  2022-09-24 23:14:28: Stopping zigbee-herdsman...`

Config is:
`data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto
serial:
  port: tcp://10.0.1.2:6638`

tube0013, 2022-09-24:
Progress!

Now it's the MQTT.. so you have the MQTT addon installed and configure in HA?

ripvega, 2022-09-25:
Yes, it's installed and running. It's used for my RF bridges.
It's configured as below:
`logins: []
require_certificate: false
certfile: fullchain.pem
keyfile: privkey.pem
customize:
  active: false
  folder: mosquitto`

It's at 10.0.1.105:1883

Could it be the login that's the problem?

ripvega, 2022-09-25:
Few!
After much installing, uninstalling, and config fiddling it's finally working.

My config is now as below:

data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: tcp://10.0.1.2:6638

Thank you for your help.

ripvega, 2022-09-26:
I'm now installing my second coordinator on an identical HA instance and am coming up against a different set of issues.

I'm seeing 'Error while starting zigbee-herdsman'

Is there a way to test my coordinator?

It's showing as connected to the network in UniFi and the the config is:

data_path: /config/zigbee2mqtt
socat:
enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: "-d -d"
log: false
mqtt: {}
serial:
port: tcp://10.0.1.102:6638

I've uninstalled Z2M deleted the folder and the reinstalled and reconfigured MQTT.

As far as I can see it is identical to my other HA instance in set up.

tube0013, 2022-09-26:
what is the full error?

you may want to set it up on a different zigbee channel too.

ripvega, 2022-09-26:
Thank you.

This is the full error:

[17:39:12] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:info  2022-09-26 17:39:14: Logging to console and directory: '/config/zigbee2mqtt/log/2022-09-26.17-39-14' filename: log.txt
Zigbee2MQTT:info  2022-09-26 17:39:14: Starting Zigbee2MQTT version 1.27.2 (commit #unknown)
Zigbee2MQTT:info  2022-09-26 17:39:14: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:error 2022-09-26 17:39:33: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-26 17:39:33: Failed to start zigbee
Zigbee2MQTT:error 2022-09-26 17:39:33: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-26 17:39:33: Exiting...
Zigbee2MQTT:error 2022-09-26 17:39:33: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)

The HA instances are located in different properties so the channel shouldn't be an issue.

tube0013, 2022-09-26:
okay.

I hate that error.

double check that ZHA is not enabled hitting the coordinator.

unplug it and wait a few minutes, maybe unscrew the antenna and then try again with antenna unscrewed.

last resort reflash the esp32 on the coordinator. - Do you have PoE or non PoE?

ripvega, 2022-09-26:
Non-PoE

tube0013, 2022-09-26:
okay if you get to that last reflash step, here are some instrucitons: https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-eth_usb#esp32-flashing

if trying to set up the device with z2m with zha running and accessing the same device it will fail with this error as only one service can use the coordinator.

ripvega, 2022-09-26:
I'm using a different coordinator for ZHA and Z2M.

tube0013, 2022-09-26:
Okay, just making sure. if the brief power off and no antenna doesn't get z2m starting try the re-flash of the esp32.

ripvega, 2022-09-26:
I'm doing this remotely as I'm not at the property but I just disabled ZHA and then Z2M started fine.

What needs to change to have both running like on my other machine?

They are all your coordinators, if that makes a difference, and all non-POE.

ripvega, 2022-09-26:
There's clearly a conflict here somewhere as I've reenabled ZHA and now ZHA shows 'Error while loading page device'.

tube0013, 2022-09-26:
They seem to be trying to connect to the same device.

tube0013, 2022-09-26:
you have 2 coordinators at the remote property. be sure lock down their IP addresses in the router. and besure ZHA is pointing to the original device IP and then configure z2m for the 2nd IP.  If it would help to have static IPs I can compile FW for you with desired IPs.

ripvega, 2022-09-26:
ZHA shows it as using device 00:12:4b:00:25:90:13:ba
Z2M shows it as using 0c:b8:15:28:b0:9f

Both have reserved IPs. 10.0.1.101 and 10.0.1.102.

tube0013, 2022-09-26:
try flashing the coordinator you want to use with z2m with the bin in this zipfile. you can do this via the coordinator's web frontend just point to the file and hit Update. It will give the device a name with z2m suffixed at the end.

[tube-zb-gw-cc2652p2-v2-z2m.bin.zip](https://github.com/tube0013/tube_gateways/files/9648612/tube-zb-gw-cc2652p2-v2-z2m.bin.zip)

ripvega, 2022-09-26:
Before I do that, I just disabled Z2M and restarted.
ZHA then started up fine and Z2M now fails again, so you are correct, there is clearly a conflict somewhere.

I'll try the flashing but can't see where to do that in the Z2M web front end.

tube0013, 2022-09-26:
you do it from the coordinator webpage, you'd browse to http://10.0.1.102 (assuming that's the send z2m coord)

ripvega, 2022-09-26:
OK. I was hoping to do it remotely. I'm connected to HA via NabuCasa and can't access the device from here.

It'll have to be tomorrow now.

Have a good evening. I'll let you know how I get on.

ripvega, 2022-09-28:
I flashed it this morning, which completed fine. However, it then shows up on my network with a self-assigned IP address (I think). It should be 10.0.1.*** as per the other one.
[image]
Could this be an indicator to the problem?

tube0013, 2022-09-28:
Apologies I think that is my fault and the coordinator I for got to comment out a static ip in the config before compiling that binary. I’ll send a corrected one later today.

ripvega, 2022-09-28:
OK. No problem.

Can’t I flash it using USB?

tube0013, 2022-09-28:
I was able to do this quickly now. you can flash it over usb as well byt moving some jumpers on the board. but the FW would be the same

I don't currently recommend the current esphome release as it has shown exhibit frequent serial disconnects. (It's improving and I'm wokring with someone to get it working better so you could just pull down a esphome yaml and compile yourself).
[tube-zb-gw-cc2652p2-v2-z2m_dhcp.bin.zip](https://github.com/tube0013/tube_gateways/files/9664156/tube-zb-gw-cc2652p2-v2-z2m_dhcp.bin.zip)

tube0013, 2022-09-28:
Oh yes , if you aren't able to reach it due to the static ip it has on your network, then yes you will likely need to flash it over USB.

https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-eth_usb#esp32-flashing

ripvega, 2022-09-28:
So, if I can’t access it over the network, USB is my only option.

Is there a tutorial for moving the jumpers?

tube0013, 2022-09-28:
Yes scroll up on the flashing instruction link posted above, the board also has markings to note the positions - you'll want to move them to the pins marked USB 2 ESP.

Basically move both so they cover the other 2 pins in the header.

ripvega, 2022-09-28:
> you could just pull down a esphome yaml and compile yourself)

Where do I get that yaml?

tube0013, 2022-09-28:
"I don't currently recommend the current esphome release as it has shown exhibit frequent serial disconnects"

I recommend esphome 2021.09.x or earlier. these can be pulled from docker, can't easily pull a specific version of the addon though. which is why I try to help out and post binaries for people.

https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-eth_usb/firmware/esphome/source

ripvega, 2022-09-28:
OK, thank you.

In that case, would you mind compiling me another binary with a ZHA suffix to make things easily distinguishable, like you added Z2M?

tube0013, 2022-09-28:
[tube-zb-gw-cc2652p2-v2-zha.bin.zip](https://github.com/tube0013/tube_gateways/files/9667135/tube-zb-gw-cc2652p2-v2-zha.bin.zip)

No problem.

ripvega, 2022-09-29:
Thank you.

I have reflashed them both and they are now connected to the network again.

I just need to get both ZHA and z2M working at the same time. I know that this is achievable as I already have one setup working well.

Thank you for your help.

Would it be possible for you to let me have a copy of the lower part of the 3D case file? I'd like to add some tabs to it so that it can be wall mounted.

I use Fusion 360 for 3D modelling.

tube0013, 2022-09-29:
I would focus on getting one up at a time, like get zha up and locked on to it's coordinator, then work toward getting z2m up. If you set both up to point to their specific IP addresses I think that should work. (skip the discovered ZHA device in HA to set up manually by IP)

I only have stl files of the cases for these devices I didn't move to fusion 360 until later. <https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-eth_usb/case>

ripvega, 2022-09-29:
Thank you.

ZHA seems to be working fine.

I'm starting the Z2M fault-finding now.

tube0013, 2022-09-29:
awesome.

just be sure you put the network into pairing/permit mode before powering them up, they like to join straight away.

ripvega, 2022-10-11:
My previous was was a little optimistic.

I still cannot get the ZHA integration to connect. I'm sure I'm missing something obvious.

Please can you take a look at the images below? It all seems like it should connect.
[image]

[image]

[image]

As you can see, I can can access the Web UI and it shows up fine in UniFi. It just fails to connect.

Any pointers will be much appreciated.

tube0013, 2022-10-11:
What version on HA?

ripvega, 2022-10-11:
2022.10.2

tube0013, 2022-10-11:
Can you pull some debug logging per the zha docs for when you try to add the coordinator.

ripvega, 2022-10-11:
> See here:
>
> https://www.home-assistant.io/integrations/zha/#debug-logging
>
> You'll want the zigpy,zigpy-znp and homeassistant.components.zha lines

Thank you.

Here's everything from the log after restart followed by me trying to add
the ZHA integration.
Hope this helps.
2022-10-11 16:09:19.182 DEBUG (MainThread) [zigpy.appdb] SQLite version for
<module 'sqlite3' from '/usr/local/lib/python3.10/sqlite3/__init__.py'>: 3.
38.5
2022-10-11 16:09:20.213 INFO (MainThread) [custom_components.hacs] Loading
known repositories
2022-10-11 16:09:20.770 INFO (MainThread) [custom_components.hacs] Starting
recurring background task for downloaded repositories
2022-10-11 16:09:20.771 INFO (MainThread) [custom_components.hacs] Stage
changed: HacsStage.RUNNING
2022-10-11 16:09:20.806 INFO (MainThread) [
homeassistant.components.mobile_app.notify] mobile_app push notification
rate limits for ash_iphone: 28 sent, 500 allowed, 0 errors, resets in
8:50:39
2022-10-11 16:10:00.001 DEBUG (MainThread) [zigpy_znp.uart] Connecting to
socket://10.0.1.101:6638 at 115200 baud
2022-10-11 16:10:00.020 DEBUG (MainThread) [zigpy_znp.uart] Opened socket://
10.0.1.101:6638 serial port
2022-10-11 16:10:00.021 DEBUG (MainThread) [zigpy_znp.uart] Connected to
socket://10.0.1.101:6638 at 115200 baud
2022-10-11 16:10:00.022 DEBUG (MainThread) [zigpy_znp.api] Toggling RTS/DTR
pins to skip bootloader or reset chip
2022-10-11 16:10:00.475 DEBUG (MainThread) [zigpy_znp.api] Sending request:
SYS.Ping.Req()
2022-10-11 16:10:00.976 DEBUG (MainThread) [zigpy_znp.api] Sending CC253x
bootloader skip bytes
2022-10-11 16:10:03.478 DEBUG (MainThread) [zigpy_znp.api] Sending request:
SYS.Ping.Req()
2022-10-11 16:10:04.479 DEBUG (MainThread) [zigpy_znp.api] Sending request:
SYS.Ping.Req()
2022-10-11 16:10:05.026 DEBUG (MainThread) [zigpy_znp.api] Connection to
socket://10.0.1.101:6638 failed, cleaning up
2022-10-11 16:10:05.026 DEBUG (MainThread) [zigpy_znp.uart] Closing serial
port

tube0013, 2022-10-11:
hmm, seems to connect but not be able to talk to the zigbee module. are the  jumpers in the correct location for network connected Zigbee?

https://github.com/tube0013/tube_gateways/tree/main/models/current/tubeszb-cc2652-eth_usb#zigbee-connection-over-ethernet-serial-over-tcp

If yes, can you try and do a zRST toggle from the Web UI before trying ZHA and see if it starts.

ripvega, 2022-10-11:
Thank you.

I'll go and check on the jumpers tomorrow (it's not at my current location).

I appreciate your help.

ripvega, 2022-10-12:
> You even can have 2 Z2M running on different channels,panid .This what i
> have in current server.

Really? How is that done?

MattWestb, 2022-10-12:
@mercenaruss You also **must** have unique E-PAN-ID **and** newtwork key and the channels is not one must but recommended or you can getting strange problem then device is jumping around like the Philips HUE motion sensor normal dos (for finding the worse router it can use in your Zigbee 3 mesh).

ripvega, 2022-10-13:
> hmm, seems to connect but not be able to talk to the zigbee module. are the jumpers in the correct location for network connected Zigbee? [...]

The zRST seems to have done the trick. Many thanks for your help.